merbanan/rtl_433 #1811: Protocol decoding and sample rate issues

nav-star, 2021-09-21:
Hello,

First I would like to thank the author and the community for this great piece of software. It spared me lot's of hours of reverse engineering attempts.

In a recent project of mine I'm trying to read the data sent by three types of sensors -  protocols 79, 142 and 145.

Using the default settings I can only decode  protocols 79 and 142

# rtl_433   -R 142 -R 145 -R 79
rtl_433 version 21.05-7-g7833050 branch master at 202106031747 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 3 out of 186 device decoding protocols [ 142 145 79 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-21 13:34:47
model     : Fineoffset-WH51                        ID        : 00e30c
Battery level: 0.889     Battery   : 1500 mV       Moisture  : 25 %          Transmission boost: 0     AD raw    : 159           Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-21 13:34:47
model     : Fineoffset-WH51                        ID        : 00e30c
Battery level: 0.889     Battery   : 1500 mV       Moisture  : 25 %          Transmission boost: 0     AD raw    : 159           Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-21 13:35:06
model     : Fineoffset-WH0530                      ID        : 253
Battery   : 1            Temperature: 19.0 C       Rain      : 132.6 mm      Integrity : CRC

Later on I discovered that by setting the sampling rate to 1000k   suddenly the device using protocol 145 becomes readable.  But for my  surprise protocol 142 became undecodable.

root@Konst:~# rtl_433 -s 1000k   -R 142 -R 145 -R 79
rtl_433 version 21.05-7-g7833050 branch master at 202106031747 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 3 out of 186 device decoding protocols [ 142 145 79 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 100000 Hz, 10.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-21 13:41:24
model     : WS2032       StationID : EFDE
Battery   : 1            Temperature: 16.9 C       Humidity  : 49 %          Wind Direction: 315.0     Wind avg speed: 0.0       Wind gust : 6.2           Rain tips : 256
Flags     : 00           Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-21 13:41:30
model     : Fineoffset-WH0530                      ID        : 253
Battery   : 1            Temperature: 18.7 C       Rain      : 132.6 mm      Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-21 13:42:12
model     : WS2032       StationID : EFDE
Battery   : 1            Temperature: 16.9 C       Humidity  : 49 %          Wind Direction: 0.0       Wind avg speed: 4.6       Wind gust : 9.3           Rain tips : 256
Flags     : 00           Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

I don't really understand how the higher sampling rate affects the reading of protocol which was properly read on lower sampling rates. I tried finding out some "goldilocks" sampling rate, but  all sampling rates in the range 500k-1000k failed with warning:

WARNING: Failed to set sample rate.

I'll appreciate your advice.

Thank you!

zuckschwerdt, 2021-09-21:
The signal might be out of the frequency window for 250k. Try both rates and `-S known` to grab some sample. View them on https://triq.org/pdv/ to see the exact frequency. Pick a center frequency in somewhere in the middle but not exactly on one of the signals.
Also `-Y autolevel -Y squelch -M level` are recommended "level" will give you an idea where the frequencies are.
With `-s 1M` you might also want `-Y minmax`

nav-star, 2021-09-21:
Thanks for the lightning fast response,  zuckschwerdt!
I'll give those a try.

Best regards!

nav-star, 2021-09-21:
Hello,

Indeed the three signals occupied a frequency range from 433.76 up to 433.92 MHz. Setting the  tuner  frequency around the middle seems to have fixed the issue for me. Even with the default sampling rate of 250kS. 

Still it is pretty obscure to me how did the increase of sampling rate make the signal with the greatest freq. offset to become decodable  and the other signal with the smaller offset which was pretty well decodable with sampling rate of 250kS to become undecodable with sampling rate of 1MS.

At the end of the day - the important thing is the setup  is doing the job.

Thanks for the help provided!

closed 2021-09-21 by nav-star

zuckschwerdt, 2021-09-22:
Glad it worked! Higher sample rate means wider frequency bandwidth.
The problem with 1M could have been filters that favor the center or increased chance to read spurious pulses.

You can grab the signal at 1M with `-S unknown` and compare (same rough freq, same "shape") to the working sample from the grab at 250k. If you manage to get those please upload here as zip so we can diagnose and improve things :)

nav-star, 2021-09-24:
Hello,

As suggested, I'm putting here captures of the same signal captured with default sampling rate of 250k and again with 1MS/s.   rtl_433 manages to properly decode the signal when the lower sampling rate is used. With the higher one in general the signal remains unrecognized, but on rare occasions it gets decoded. 

Best regards
[signal.zip](https://github.com/merbanan/rtl_433/files/7225455/signal.zip)
!

zuckschwerdt, 2021-09-25:
This is an interesting test case. The additional sample rate gets more noise in, and we would need tighter filters to smooth the FSK:
250k:
[image]

1000k:
[image]

Filters are shown for 250k and 1000k as:
`baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us`
`baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us`
If we force e.g. a 20 µs wide filter (`-Y filter=20`), the FM is cleared up. But we can't find that setting automatically. Also the demod still won't detect this.

Maybe @merbanan has some idea about this. It's a good example to optimize our FSK further in any case.

merbanan, 2021-09-25:
Looking at the AM signal there are some spikes in the amplitude. I am not sure what causes them but these spikes distort the FM signal. The filters used need to be strong enough to get rid of the spikes but still pass through the FSK signal. We could try a set of filter limits but the current pipeline does not support that kind of processing. That is the only solution I can come up with.

klohner, 2024-04-02:
FWIW, manually specifying a level anywhere from -3 to -9 will pick out the signal in the referenced 1000k sample file.

`rtl_433 -r g003_433.83M_1000k.cu8 -Y level=-5`
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.069544s
model     : Fineoffset-WH51                        ID        : 00e30c
Battery level: 0.889     Battery   : 1500 mV       Moisture  : 33 %          Transmission boost: 0     AD raw    : 189           Integrity : CRC
[Auto Level] Estimated noise level is -18.3 dB, adjusting minimum detection level to -15.3 dB
```