merbanan/rtl_433 #1813: Proove PIR sensor, won't even trigger debug output.

pacc, 2021-09-22:
I got some trouble with a Proove PIR sensor art. 00166 which a nexa/proove switch could learn.

rtl_433 won't give any outputs with different debug settings i.e -A 4 
 - while my 433 remote and various sensors shows up fine nothing seems to be recorded when I trigger PIR commands.

My telldus device reacted to it so I installed the ancient telldus-core and tdtool-improved 
and given the houseid in the remote commands I pick up I'd expect that at least arctech protocol is supported by rtl_433
see telldus logs below

Any ideas?  I'd like to sample SDR data while triggering the PIR but rtl_433 won't react and tools like gqrx failed to start for me.

For reference the  remote that rtl_433 can detect shows:
1632337948: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:13895893;unit:1;group:0;method:turnoff;
  controllerId: 1
1632337948: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:1111110101;method:turnoff;
  controllerId: 1
1632337948: RawDeviceEvent: class:command;protocol:everflourish;model:selflearning;house:282;unit:3;method:turnoff;
  controllerId: 1
1632337948: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:27791786;unit:1;group:0;method:turnoff;
  controllerId: 1
1632337948: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:1111111010;method:turnon;

The PIR device has very similar outputs and protocols  (PIR detect, Learn ON, Learn OFF)
1632338588: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnon;
  controllerId: 1
1632338588: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110011011;method:turnon;
  controllerId: 1
1632338589: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnon;
  controllerId: 1
1632338589: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110011011;method:turnon;
  controllerId: 1
1632338589: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:279608;unit:7;group:1;method:turnoff;
  controllerId: 1
1632338589: RawDeviceEvent: class:command;protocol:waveman;model:codeswitch;house:G;unit:3;method:turnon;
  controllerId: 1

1632338604: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnon;
  controllerId: 1
1632338604: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110011011;method:turnon;
  controllerId: 1
1632338605: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnon;
  controllerId: 1
1632338605: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110011011;method:turnon;
  controllerId: 1
1632338606: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnon;
  controllerId: 1
1632338606: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110011011;method:turnon;
  controllerId: 1

1632338611: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnoff;
  controllerId: 1
1632338611: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110111011;method:turnon;
  controllerId: 1
1632338612: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnoff;
  controllerId: 1
1632338612: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110111011;method:turnon;
  controllerId: 1
1632338613: RawDeviceEvent: class:command;protocol:arctech;model:selflearning;house:1118434;unit:10;group:0;method:turnoff;
  controllerId: 1
1632338613: RawDeviceEvent: class:command;protocol:sartano;model:codeswitch;code:0110111011;method:turnon;
  controllerId: 1

zuckschwerdt, 2021-09-23:
If you need a quick way to get a view on the spectrum or grab some samples try:
`rtl_433 -w mytest_433.92M_250k.cu8 -T 20`
this will record 20 seconds. Use the file on https://triq.org/pdv/ to get an overview of signals.
Get multiple samples (maybe shorter time to keep file size down) verify the signal is the one you want, zip and post here or on some one-click hoster if the size exceeds GH limits.

pacc, 2021-09-23:
It seems that the frequency was off at 433.362, 
if it can be decoded, is there also a way to make it work together with the standard protocols?

[PIR_433.362M_250k.zip](https://github.com/merbanan/rtl_433/files/7215246/PIR_433.362M_250k.zip)
[PIR_433.4M_250k.zip](https://github.com/merbanan/rtl_433/files/7215249/PIR_433.4M_250k.zip)

zuckschwerdt, 2021-09-23:
Uh, what's that tone at 433.315M? That will be interfering with reception.
Since your device is at 433.36M it's best to put that at the edge of the band and not receive 433.315M.
E.g. `-f 433.47M` at 250k will cover (+/-125k) 433.345M to 433.595M.
If you want to receive at 433.92M also you can try 1M sample rate:
`-f 433.85M -s 1M` at 1M will cover (+/-500k) 433.35M to 434.35M.

pacc, 2021-09-23:
Thank you for your help,
using -s 1M I saw that it picks up the protocol OK and your input about frequency coverage will make this useful in practice.

I have no idea what is sending on 433.315M but I debugged a bit and caught this:
[unknown_432.75M_1024k.cu8.zip](https://github.com/merbanan/rtl_433/files/7215800/unknown_432.75M_1024k.cu8.zip)
from a small sample I caught the 433.315 signal cuts off when these OOK is sent so I tried to isolate it.

Have a look if you are curious and feel free to close this issue.
I have a clear view to the horizon so I don't expect it to be anything I can turn off easily :)

zuckschwerdt, 2021-09-23:
Likely the tone is just shadowed by the strong signal of that nearby OOK.
The tone is narrow and stable, so it's likely a pilot tone or empty carrier from some HAM radio station relay.

closed 2021-09-23 by zuckschwerdt