merbanan/rtl_433 #1815: Lutron Caseta signal being detected as single pulses

synthead, 2021-09-25:
I'm looking to add support for Lutron Caseta devices that operate at 433 MHz.  Here's an example signal from GQRX:

[image]

I believe that the system sends multiple identical signals for robustness, so to the best of my knowledge, the first emission before the pause should contain all the data to retrieve information about the signal.  The rest are repeats.

With `rtl_433` with the `-A` flag, it seems to recognize the entire signal as single pulses:

```
rtl_433 -A

Detected OOK package	2021-09-25 02:40:53
Analyzing pulses...
Total count:    1,  width: 0.05 ms		(   12 S)
Pulse width distribution:
 [ 0] count:    1,  width:   48 us [48;48]	(  12 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width:   48 us [48;48]	(  12 S)
 [ 1] count:    1,  width: 10004 us [10004;10004]	(2501 S)
Level estimates [high, low]:   1989,    254
RSSI: -9.2 dB SNR: 8.9 dB Noise: -18.1 dB
Frequency offsets [F1, F2]:    -652,      0	(-2.5 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB102003027148155
```

I'm fairly new to SDR and DSP in general, and could use a little guidance on how I can parse this signal into data.  Thanks in advance!

zuckschwerdt, 2021-09-25:
You are seeing the (bright) "mass" of the signal left and right in the spectrum but not really in the center. This is FSK, viewed as ASK (OOK) this is just a single pulse. Seems our default FSK demodulation can't pick that signal up.
Try `-Y minmax`, otherwise grab a signal (`-S unknown`) and upload the .cu8 here as zip.

synthead, 2021-09-25:
`-Y minmax` doesn't seem to make it work.

Here's a cu8! [g003_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/7229645/g003_433.92M_250k.cu8.zip)

zuckschwerdt, 2021-09-25:
Drop that .cu8 on https://triq.org/pdv/ to view. The file is empty ;)
I just realize you want to receive at around 433.6MHz -- that's out of band for 433.92 M at 250k. Try `-f 433.65M`

synthead, 2021-09-25:
Oh interesting!  I used `-f 433.65M` as you suggested, and viewed the output in https://triq.org/pdv.  This is what I see:

[image]

It seems like there isn't enough clarity in the waveform somehow, and with this online viewer, it does look like single pulses from the recording.  Maybe some settings need to be tweaked?

Here's the cu8 file from the screenshot above: [g003_433.65M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/7230426/g003_433.65M_250k.cu8.zip)

If it helps, here is an IQ file of the signal (an SDRplay RSP1A over a network with the rsp_tcp daemon): [gqrx_20210925_222452_433436000_625000_fc.raw.zip](https://github.com/merbanan/rtl_433/files/7230445/gqrx_20210925_222452_433436000_625000_fc.raw.zip)

synthead, 2021-09-25:
Here is a waterfall of the same signal being recorded with an RTL2832U SDR:

[image]

It seems like rtl-sdr dongles should be able to decode this signal without issues :slightly_smiling_face:

zuckschwerdt, 2021-09-25:
Rename the .raw file to .cf32, then https://triq.org/pdv/ can read it.
There is a pronounced DC spike in both recordings. Not sure why the RTL-SDR sample show those wide shadows. Not good.
Your SDRplay is far superior in all regards. Use rtl_433 with `-d "" -s 1M -Y autolevel -Y minmax -M noise -M level`
That might already be better, otherwise try 1M or 1024k sample rate on the RTL-SDR and try to get a smoother reception. Is the antenna the same one?

synthead, 2021-09-26:
DC spike, like a notch in the recording?  I think that is just an artifact of the SDR.  It seems to have notches right at the tuned frequency and at the USB frequency harmonics.  It hasn't caused any issues for me, yet.

I seem to be getting closer with these options and a RTL2832U SDR:

```shell
rtl_433 -f 433.65M -X 'n=name,m=FSK_PCM,s=11,l=10,r=1000' -g 50 -Y minmax -Y autolevel
```

However, even though I'm receiving chunks of data now, the data seems to be somewhat random.  I'm sorta just guessing about all the options to `-X` with some success :eyes:

zuckschwerdt, 2021-09-26:
Yes, the DC spike is not something we usually see unless there are conversion steps that introduce it. (USB as in upper side band?)
The signal should look like your gqrx recording but without the notch and louder.
Here is an example, you can see the signal clearly, with FSK data.
https://triq.org/iqs/#/archos-tbh/01/tbh_temp_001_433.901M_250k.cu8

If `-S unknown` won't give you a snapshot reliably, you can try to record time-based with e.g. `-w mytest_433.65M_250k.cu8 -T 30`

gdt, 2023-10-04:
What's the status and path forward?  Is anyone working on it?  Is there partial work to capture?