merbanan/rtl_433 #1816: Old protocol list

exeljb, 2021-09-25:
So this may be a dumb question but how do we update the protocol list? 
My current list only goes to 186.

I appreciate any pointers!

zuckschwerdt, 2021-09-25:
Use `rtl_433 -V` to get the version installed. This might not work if the version is really old.
Current version is `rtl_433 version 21.05` -- you might want to upgrade to that.

closed 2021-09-25 by zuckschwerdt

exeljb, 2021-09-25:
that command shows version 21.05. I must be missing something else?

zuckschwerdt, 2021-09-25:
Oh, sorry, I didn't check when 186 was current. Indeed the last stable version, 21.05 has protocols up to 186. Six more protocols have been added in the mean time. You need to compile yourself from github sources for that. I new stable version is due, maybe in a month or so.