merbanan/rtl_433 #1817: Warning from "emonTx OpenEnergyMonitor"

gvanem, 2021-09-25:
Running my modified `rtl_433_tests/bin/run_test.py` script, I've noticed a lot of:
```
WARNING: Undeclared field "mic" in [45] "emonTx OpenEnergyMonitor"
```

Or manually with this command:
 `rtl_433.exe -c NUL -M newmodel -s 2500000 -F json -r rtl_433_tests\tests\emontx\02\gfile073.cu8`

I think the error is in the `output_fields[]`. Since this patch squelches this warning:
```diff
--- a/src/devices/emontx.c 2021-06-30 18:34:47
+++ b/src/devices/emontx.c 2021-09-25 15:36:11
@@ -149,6 +149,7 @@
         "temp5_C",
         "temp6_C",
         "pulse",
+        "mic",
         NULL,
 };
```

closed 2021-09-25 by zuckschwerdt

gvanem, 2021-09-25:
There are more warnings like this:
```
WARNING: Undeclared field "power0" in [44] "CurrentCost Current Sensor"
WARNING: Undeclared field "battery_ok" in [35] "Esperanza EWS"
```

zuckschwerdt, 2021-09-25:
Fixed!

gvanem, 2021-09-25:
Found some more.
```
WARNING: Undeclared field "id" in [154] "Standard Consumption Message Plus (SCMplus)"
WARNING: Undeclared field "id" in [178] "Security+ (Keyfob)"
```

zuckschwerdt, 2021-09-25:
Fixed! :)

gvanem, 2021-09-25:
Good!

BTW, here is my modified `run_test.py` script: https://gist.github.com/gvanem/1184e11fab3fea4511122094fa7e1f3a

I shining colours using colorama. E.g.:
[image]

zuckschwerdt, 2021-09-25:
Complete modernized rewrite ;) We might want to update the script with some of those ideas.

commit c191c184 (2021-09-25): minor: Fix missing mic key in emonTx (closes #1817)