merbanan/rtl_433 #1823: Built in MQTT has more delay than using mosquitto_pub

micahvandeusen, 2021-10-04:
There seems to be a good 1-2 second minimum time difference between the built in MQTT capability and piping json into mosquitto_pub.

zuckschwerdt, 2021-10-04:
What configuration (arguments to rtl_433) are you using? Is there much data to send?
The network processing is sending at the same instant as the printed outputs. But the network could be buffered.
If you are running at 250k sample rate, can you test 1M sample rate and measure if there is a change?

zuckschwerdt, 2021-10-04:
Using `-M time:usec` and `mosquitto_sub -v -t '#' | ts "%.T"` (ts from moreutils) I see a delay of about 1.5s.
Half that using `-s 1M`. Note that the time reported is the start of the radio frame, multiple frames might be processed and then reported after the next frame interval (250ms for 250k, 100ms for 1M).
So this does not seem too much delay. Do you see much more delay?
I'll add more timing diagnostics and checks. Fastest possible reports is a priority for rtl_433.

zuckschwerdt, 2021-10-04:
Actual delay from output to MQTT receive for me is 525ms at 250k, 130ms at 1M.
That looks consistent with network processing bound to the frame rate.

We do have some experimental code to run the SDR input on another thread and thus keep the main thread free to run a faster loop. It will take some more time to get that ready though.

closed 2021-11-18 by micahvandeusen