merbanan/rtl_433 #1824: Adding Support for Acurite Grill/Meat Thermometer 01185M

exeljb, 2021-10-04:
This Acurite device features a dual temperature indication for both the ambient grill temp and a meat temp.  There's a local battery powered transmitter unit that only displays the temperatures and sensor error and battery messages.  The Remote unit receives all that information as well and displays it.  Each unit has its own switch for C and F, it doesn't appear that having the switch set to C on the transmitter affects what the receiver displays.  When a sensor is unplugged, both displays show an E1 error.  More details and images can be found where I[ bought it](https://www.amazon.com/gp/product/B07BRNB1ND).

[acurite_01185M_uc8_files.zip](https://github.com/merbanan/rtl_433/files/7280714/acurite_01185M_uc8_files.zip)
This zip file contains 5 .cu8 files with signals recorded under certain scenarios.  
Data captured include:
     * Different steady temps to identify which temp is the ambient/meat readings
     * Individual sensor unplugged conditions while the other sensor is at 150F
     * The above readings were captured with a full battery
     * Low battery indication received by the receiver while both sensors held at 200F

zuckschwerdt, 2021-10-05:
Very good description and clean signals. Easy to demod:
Use a flex decoder with -X 'n=name,m=OOK PWM,s=840,l=2070,y=6600,g=5000,r=6000'

You get the [data in this BitBench.](https://triq.net/bitbench?c=430f6f5927e54b%20%5Bt1%20150f%20%20t2%20unplugged%20batt100%202%5D&c=ca3e2fe52f45a0%20%5Bt1%20200f%20%20t2%20200f%20batt%20lo%202%5D&c=430f5f830fbdf8%20%5Bt1%2044f%20%20t2%20300f%20batt100%202%5D&c=430f5fe16f5913%20%5Bt1%2050f%20%20t2%20150f%20batt100%202%5D&c=430f27e56f594b%20%5Bt1%20unplugged%20%20t2%20150f%20batt100%202%5D&f=%5E8h%20&cl=56&cw=8) Since the bits look reversed I reflected them (`^8h`).
If we then use that reflected data and a constant of `5` we see the [checksum match in this BitBench](https://triq.net/bitbench?c=05%20c2f0f69ae4a7d2%5Bt1%20150f%20t2%20unplugged%20batt100%202%5D&c=05%20537cf4a7f4a205%5Bt1%20200f%20t2%20200f%20batt%20lo%202%5D&c=05%20c2f0fac1f0bd1f%5Bt1%2044f%20t2%20300f%20batt100%202%5D&c=05%20c2f0fa87f69ac8%5Bt1%2050f%20t2%20150f%20batt100%202%5D&c=05%20c2f0e4a7f69ad2%5Bt1%20unplugged%20t2%20150f%20batt100%202%5D&f=8x%208h8h%20~16d%20~16d%20CHK%3A8h&cf=ADD&cl=56&cw=8).

Also the battery flag shows and the values somewhat work (increasing/decreasing and a fixed value for unplugged), not sure how yet.

zuckschwerdt, 2021-10-05:
Next step would be to run the flex decoder, capture the output codes while running a slow ramp on each temperature input.
The codes then should show which bits do what (LSB/MSB, width, ...)

exeljb, 2021-10-05:
Here's some of the data I've received using that flex decoder string.  I'm curious if this is what you're looking for and if it makes any sense to you. 

```
{"time" : "2021-10-05 18:06:04.946106", "model" : "name", "count" : 16, "num_rows" : 16, "rows" : [{"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}, {"len" : 56, "data" : "b80f9f7e9f5637"}], "codes" : ["{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637", "{56}b80f9f7e9f5637"], "mod" : "ASK", "freq" : 433.910, "rssi" : -9.761, "snr" : 29.373, "noise" : -39.134}
{"time" : "2021-10-05 18:06:22.434534", "model" : "name", "count" : 8, "num_rows" : 8, "rows" : [{"len" : 56, "data" : "b80f9f819f76cf"}, {"len" : 56, "data" : "b80f9f819f76cf"}, {"len" : 56, "data" : "b80f9f819f76cf"}, {"len" : 56, "data" : "b80f9f819f76cf"}, {"len" : 56, "data" : "b80f9f819f76cf"}, {"len" : 56, "data" : "b80f9f819f76cf"}, {"len" : 56, "data" : "b80f9f819f76cf"}, {"len" : 56, "data" : "b80f9f819f76cf"}], "codes" : ["{56}b80f9f819f76cf", "{56}b80f9f819f76cf", "{56}b80f9f819f76cf", "{56}b80f9f819f76cf", "{56}b80f9f819f76cf", "{56}b80f9f819f76cf", "{56}b80f9f819f76cf", "{56}b80f9f819f76cf"], "mod" : "ASK", "freq" : 433.909, "rssi" : -10.783, "snr" : 25.340, "noise" : -36.124}
{"time" : "2021-10-05 18:06:38.434296", "model" : "name", "count" : 8, "num_rows" : 8, "rows" : [{"len" : 56, "data" : "b80f9f219fcedf"}, {"len" : 56, "data" : "b80f9f219fcedf"}, {"len" : 56, "data" : "b80f9f219fcedf"}, {"len" : 56, "data" : "b80f9f219f4e5f"}, {"len" : 56, "data" : "b80f9f219f4e5f"}, {"len" : 56, "data" : "b80f9f219f4e5f"}, {"len" : 56, "data" : "b80f9f219f4e5f"}, {"len" : 56, "data" : "b80f9f219f4e5f"}], "codes" : ["{56}b80f9f219fcedf", "{56}b80f9f219fcedf", "{56}b80f9f219fcedf", "{56}b80f9f219f4e5f", "{56}b80f9f219f4e5f", "{56}b80f9f219f4e5f", "{56}b80f9f219f4e5f", "{56}b80f9f219f4e5f"], "mod" : "ASK", "freq" : 433.909, "rssi" : -10.965, "snr" : 26.408, "noise" : -37.373}
```

Is the area of interest the data `b80f9f7e9f5637` or the codes `{56}b80f9f219fcedf`?

I'll post more if this is what you are looking for.

exeljb, 2021-10-05:
Not sure if this is helpful or not but I tossed the following lines into BitBench...
```
56 b80f9fb99fd93c
56 b80f9f399fd9dc
56 b80f9f399f595c
56 b80f9f985fb2d6
56 b80f9fd85f3236
56 b80f9f385f52d6
56 b80f9ff85f6256
56 b80f9ff85fa296
56 b80f9f845f520e
56 b80f9f445f3ac1
56 b80f9f445f8611
56 b80f9f445f4e99
56 b80f9f245f7ee5
56 b80f9fe45fc5f3
56 b80f9fd45f9d97
```
and it spit out...
[image]

zuckschwerdt, 2021-10-06:
Just a plain list of codes (`{56}b80f9f219fcedf`) is most useful.
Note that the `{56}` ist the length of the code. You can copy that verbatim to BitBench or strip the length off when it's a multiple of 8 anyway.

*But in that BitBench you need to prefix `a0` to get a working checksum. It's a hack, sorry for the confusion.*

Can you find a formula to match the temperatures with those numbers? Something like F or C divided by x plus y?

exeljb, 2021-10-06:
Nope you are right, I changed it to 8b and it does show bits changing. Here’s what changes:
[image]

As for the channel setting or power cycling ID changes…there is no channel switch on either the transmitter or receiver. I am unsure if there’s an ID reset when batteries are changed.  I’ll try messing around with that later tonight. The manual makes no mention of this though.

zuckschwerdt, 2021-10-06:
Could also be that the battery-low flag is `c2` changing to `53` right? But it's unclear what `1d` is still...

exeljb, 2021-10-08:
[This BitBench](https://triq.net/bitbench?c=a0%20430f6f5927e54b%20%5Bt1%20150F%2065C%20%20t2%20unplugged%20batt100%202%5D&c=a0%20ca3e2fe52f45a0%20%5Bt1%20200F%2093C%20%20t2%20200F%20batt%20lo%202%5D&c=a0%20430f5f830fbdf8%20%5Bt1%2044F%207C%20%20t2%20300F%20159C%20batt100%202%5D&c=a0%20430f5fe16f5913%20%5Bt1%2050F%2010C%20%20t2%20150F%2065C%20batt100%202%5D&c=a0%20430f27e56f594b%20%5Bt1%20unplugged%20%20t2%20150F%2065C%20batt100%202%5D&c=a0%2098ce9f859f193d%20%5Blow%20bat%5D&c=a0%2098cf9f799fa96c%20%5Bincrease%20bat%5D&c=a0%2098ce9fd99f490d%20%5Blow%20bat%5D&c=a0%2098cf9fd99f89f4%20%5Binc%20bat%5D&c=a0%20e7cf9fe99f091f%20%5Bpower%20cycle%5D&c=a0%20f69f9fe99f89e1%20%5Bpower%20cycle%5D&c=a0%20249f9f999f05b2%20%5Bpower%20cycle%5D&c=a0%20fdcf9fb99fa5d7%20%5Bpower%20cycle%5D&c=a0%20e00f9fb99f658c%20%5Bpower%20cycle%5D&c=a0%20d49f9fb99fa5ba%20%5Bpower%20cycle%5D&c=a0%20d49f9f799fe506%20%5Bpower%20button%20on%2Foff%5D&c=a0%205a3f9f799f15c9%20%5Bpower%20cycle%5D&f=8x%208d%201b%203b%204d%20~16d%20~16d%20CHK%3A8h&r=true&cf=ADD&cl=56&cw=8) might make a little more sense in the formatting I'm thinking is correct so far.  Those 3 bits between the battery flag bit and the channel bits are what I'm gonna try to figure out next.

exeljb, 2021-10-08:
It appears random.  There are no switches like on other acurite devices and the manual doesn't state anything about channels either.  Actually, the manual doesn't state anything about having to re-pair after changing the batteries in the tx.

zuckschwerdt, 2021-10-09:
I'm just writing the decoder, to be clear the E2 error is a value of 0x0200 = 512, i.e. -38.8 F?

exeljb, 2021-10-09:
Yes it’s the latter so the 200 in decimal which equates to the -70.0F.

commit d257d6ce (2021-10-09): Add support for Acurite-01185M (closes #1824)