merbanan/rtl_433 #1837: Help decoding Auriol LD5661 (LIDL)

Gollam, 2021-10-17:
Hi,
I've been trying to figure out how to decode these samples for a while now. The device reports outside temperature, amount of rain and I guess low battery signal. Once I think I saw some hex data indicating some decoded bits, but I could never produce them again. Probably I'm not doing the analyzing part right. Can someone please help me on how to start working with these samples?
[samples_ld5661_1.zip](https://github.com/merbanan/rtl_433/files/7359755/samples_ld5661_1.zip)

zuckschwerdt, 2021-10-17:
Try the files on https://triq.org/pdv/ to get a spectrogram. The second file is empty, the other three files only have a tail portion of a signal. I.e. the frame is picked up to late. The last file is picked up as `Acurite-01185M` strangely.

Try to add `-Y autolevel -Y minmax -Y magest -M level -M noise` if that doesn't help to get good sample (with `-S unknown`) try to record fixed time files (`-w myname_433.92M_250k.cu8 -T 10`) and pick one containing the full signal.

It looks like PWM with 1000/2000 µs. Ideally this should work: `-X 'n=name,m=OOK_PWM,s=1000,l=2000,r=3000'`

Gollam, 2021-10-18:
Thanks, it went much better this time. Attaching some samples, in which I am able to see the temperature bits with the following parameters: `rtl_433 -r *.cu8 -X 'n=name,m=OOK_PPM,s=960,l=1936,g=1952,r=1952'`

d1, d2, d4 is the temperature, but I couldn't able to figure out the rain measurement. It increments by 0.3, in the last sample, it shall be up to 3.9mm.
[samples_ld5661_2.zip](https://github.com/merbanan/rtl_433/files/7362620/samples_ld5661_2.zip)

Gollam, 2021-10-18:
I know about 1 other person who bought it, so we will know if the sync-word is common later. 

I can try writing the decoder of course!

4-LD5661-2 is written both on the console and on the rain sensor, I'll have another look tomorrow.

elenril2, 2021-11-21:
Hello,
RTL suddenly stopped collecting data from my Auriol LD5661. I think there is a problem with decoding. I've capture some data without further analyze [rtl_samples.zip](https://github.com/merbanan/rtl_433/files/7576212/rtl_samples.zip)
Could you help me with this case or give me params to make correct analyze?

zuckschwerdt, 2021-11-21:
Drop the files on https://triq.org/pdv/ and have a look. Most are just empty (just noise). Frequency drift might have put your sender outside of the band. Try https://triq.org/rtl_433/ANALYZE.html#verify-a-transmission

elenril2, 2021-11-21:
Thanks for advice, I'll try to analyze it. I don't think that sender is outside of the band because weather station collect this data.

elenril2, 2021-11-21:
My RTL only collects a dozen samples - is this normal behavior? Maybe it's broken?
They looks the same as the first one, maybe there is a source of noise somewhere near my antenna which is interfering the proper reception.

zuckschwerdt, 2021-11-21:
Your weather station head unit and the RTL-SDR receiver are not set to the same effective band. S. https://triq.org/rtl_433/PRIMER.html 2nd question.

elenril2, 2021-11-21:
I'm confused - I thought my RTL was tuned to correct band, cause earlier device was collecting data. However when I analyze my HA entity log I noticed that there was some gaps in collecting even for couple of hours.
By the way I manually measured my rain gauge and I think that 0.3 mm per tipping bucket is not correct quantity. It is about three times larger - closer to 1 mm.

zuckschwerdt, 2021-11-21:
"correct band" is subjective to each user. You really need to locate your signal, then set *your* band.

Gollam, 2021-11-21:
How did you manually measure the amount of rain? 1mm of rain equals to 1 liter per square metre, and the surface of the collecting are of this sensor is about 0.007386065 m^2.

elenril2, 2021-11-21:
@zuckschwerdt Thanks for advice, I was not aware of that. Unfortunately, I tried some bands from 433 to 434 MHz but I didn't catch decoded signal. I think that if it was drifted then not far away form default 433,92 MHz. Maybe there is another way to determine right band? I was simply trying ```rtl_433 -R 200 -f 433000000``` and then increase frequency by 200000 Hz.
@Gollam Diameter of the rain gauge has approximately 10cm, and this gives a 78.54 cm^2. When I poured water to this gauge tipping bucket tilt at 8ml. Earlier I compare Auriol measurment with the manual rain gauge and the difference was about 2,5-3 times.

Gollam, 2021-11-21:
Just FYI, I have 2 sensors here, both were successfully received simultaneously with just `rtl_433 -R 200`.

zuckschwerdt, 2021-11-21:
Stepping through bands with increasing `-f` is ok. If it does not work follow https://triq.org/rtl_433/ANALYZE.html#verify-a-transmission -- i.e. try to visualize the band and find your signal.

elenril2, 2021-11-21:
Strange, now I catched two reports on 433,90 MHz. When it was stable sensor sent reports once in a minute.
@Gollam And can you verify that measurements with the step of 0,3mm are correct? I've noticed that temperature measurements are also bit inaccurate - especially at dusk sensor lowers readings and on a sunny day measurement is too high.

elenril2, 2021-11-21:
Now the signal appeared at 433.885 MHz - but I don't know why it is so unstable, and RTL stops catching reports (or device stops sending it) once in a while.

Gollam, 2022-06-08:
@elenril2 now I was able to confirm. 4mm of rain fell, measured with my analog rain meter. And the device measured 1.8mm. So it really seems inaccurate.

elenril2, 2022-06-08:
I think the manufacturer knows this problem too. I have a second newer unit and measurement step is 0,9 mm, but once upon a time (maybe in 2-3 steps) is 0,6 mm.

Gollam, 2022-06-08:
@elenril2 @zuckschwerdt so the question is, how shall we proceed, the simplest way is to just adjust the multiplier in the code, the question is with what value, 2.5?

elenril2, 2022-06-08:
I'm doing this calculation in my Home Assisant enviroment. I convert 0.3mm to 1mm. I think the built-in rtl_433 solution will not be perfect, cause as I posted earlier newer units have this conversion hardcoded, but still inaccurate.

Gollam, 2022-06-08:
But, what do you mean new units have this hardcoded? The display unit does the multiplying, the rain sensor only sends a counter value (0, 1, 2, etc.), which can be multiplied.

merbanan, 2022-06-08:
The solution is to report tips and/or several mm measures.

Gollam, 2022-06-08:
Ok, of course I can report the tip count, but the question from my side is that in rtl_433, we currently multiply the tip count with 0.3, which is clearly wrong, but it can be changed to a more accurate value, should we adjust it to something between 0.7-1.0 mm? In that case, the display unit won't show the same mm value like what we report, likely not with the old or new units either. But it will be the truly accurate value.

elenril2, 2022-06-08:
> But, what do you mean new units have this hardcoded? The display unit does the multiplying, the rain sensor only sends a counter value (0, 1, 2, etc.), which can be multiplied.

I don't know exactly how it is done, but rtl_433 gather multiplied value (steps are 0.9mm or 0.6mm) and the display too (but in this case I'm not 100% sure, cause the range of device is very weak in my home). If I remember correctly the rain sensor sends a total amount of rain since battery replacement and not the counter value.

Gollam, 2022-06-08:
I implemented the decoder for this sensor, and I'm 100% sure that only the counter value is being sent by the sensor, and I hardcoded 0.3 as a multiplier in the decoder source :) So we can change that to anything.

elenril2, 2022-06-08:
Ok :) Maybe my new unit is broken - but it send 3 sometimes 2 tips at the same time, never one.

merbanan, 2022-06-08:
We should match the display unit. Then for people how know better we could also supply other/better values and then the user can pick, it is also possible to inject a multiplier. I'm unsure what the best thing to is, but I guess the best way is to actually calibrate the device by pouring 1l of water in the collector cup and based on that supply the multiplier from the command line.

Or we change the multiplier to a better value and if the user needs another/better value one can be provided via the command line. It seems like these devices are nonlinear and can use some help to get better measurements out of them.

Gollam, 2022-06-08:
@elenril2 The question is, how many physical tips were happening when the sensor sent 2 or 3. I had a broken unit, which for 1 tip always reported multiple, and other times for many tips it did not increment the counter at all. I tested it by manually tilting the sensor and then waiting for the new report. I had that one changed under warranty by Lidl. So you must bring the sensor in, tip it once at a time, then check the new report (it is sent in every 55 seconds).

@merbanan ok, I think I will do the experiment and try to accurately measure it. If we change it to anything other than 0.3, then the display unit won't show the same value.

elenril2, 2022-06-08:
And as for multiplication for one step after approximate measurements of the rain gauge I assumed the one step is 1mm. I've based calculation on diameter of the gauge and the amount of water I poured in it. My Auriol Gauge is installed on the roof and is exposed to wind. I think the most appropriate method of determining the exact capacity would be to place a classic rain gauge next to it and take a few measurements.

Gollam, 2022-06-08:
That's what I did, I have my analog rain meter next to it, which I commented about this morning. That one measured 4mm, and the auriol measured 1.8. I guess we can believe that the analog one is accurate.

merbanan, 2022-06-08:
If the display unit uses 0.3 then that is the "correct" value. But we can also report more keys, like steps and custom values based on command line options. Then a user can supply its own multiplier and select another key. But presenting the same value as the display unit is sort of the rule we follow. All devices deviate but we can only do so much to correct it.

zuckschwerdt, 2022-06-08:
We don't want conversion factors that are fine-tuned too much. Those might be only valid for that sensor measured.
Instead we try to find out what the intent of the protocol was, or how the head unit does the conversion.

If however the conversion is obviously broken (off by a factor of 2.2 here, right?) we might opt to fix that in the decoder with a comment explainig the situation. But that should be the exception, we would try to match the "official" decoding normally.

Other cases where we don't like to follow the official receiver presentation is e.g. a channel setting shown as A, B, C -- we match the majority of devices there and show 1, 2, 3.

Your measured 4mm at 6 tips (1.8 shown) is curious. Did the sensor reliably miss every second tip? Is that a mechanical (stuck) or electrical (bounce) fault? I guess that judging if the cup is 0.3 or 0.66 sized is hard to get wrong, right?

elenril2, 2022-06-08:
Earlier I've done measurment in my device -  diameter of the rain gauge has approximately 10cm, and this gives a 78.54 cm^2. When I poured water to this gauge tipping bucket tilt at 8ml.

Gollam, 2022-06-08:
@zuckschwerdt this instance of the unit is not faulty, every tip is counted once, I examined it earlier. So there's nothing wrong with the counting, apparently they really seem to have failed correctly sizing the tip (as you said, 0.66 instead of 0.3). I'll fill my manual rain meter with different amounts of water, then I'll test the sensor by pouring it in and see how many times it tilts.

Gollam, 2022-06-08:
@zuckschwerdt yes, of course, the diameter of both has to be taken into account.

zuckschwerdt, 2022-06-08:
That seems to solve @elenril2 's observations: 1mm rain per tip, and 1 tip produces 3 counts. One count at 0.3mm (or 1/3mm maybe?) works for the display unit. Did I get that right?

elenril2, 2022-06-08:
In my case only one of the two devices I have behaves like this (2 or 3 counts per tip) - I think it is faulty, but I check this more precisly like Gollam suggested.
In my calculation there might be some error cause I have poured such a little amount of water to observe a tilt.

elenril2, 2022-06-13:
> @elenril2 The question is, how many physical tips were happening when the sensor sent 2 or 3. I had a broken unit, which for 1 tip always reported multiple, and other times for many tips it did not [...]

I've checked once again and the first tilt send 3 tips and the second one 2 tips. Or maybe the other way round - recently I have a problem with collecting data at RTL from this device (central unit next to my HA server with RTL dongle collects this data more often but not always). I've also double checked capacity of the tipping bucket - I confirm that for my two rain gauges it is aprox 8ml.
But overall I'm very dissapointed with this device. Couple of days ago there was a heavy rain in my location. For the first period there was 10 tip counts (in device that not incorretly reports multiple tips) for 14mm of rain measured by traditional rain gauge. 10 tips is about 10mm with my measurment, that is less than real readings. However, the most frustrating was that when, after a several hours break , there was another rainfall both of my Auriol rain gauges did not notice any rain (in fact, another 14mm rains fell).
Auriol Rain Gauge is highly inaccurate and unreliable. I've ordered weather station with wifi connection and I hope my problems will be over.

Gollam, 2022-07-26:
Just an FYI, I had some time, so I made a very accurate calculation this time. The collecting area is 96mm wide, and as @elenril2 correctly measured exactly 8ml makes the gauge tip. 8 ml / 72.3822947 cm^2 = 1.10524266 millimeters. So we can conclude that the display shipped with the unit is incorrectly displaying the fallen rain, as it is multiplying the tips with 0.3 mm instead of this value. 

@zuckschwerdt, shall I make an extra field in the decoder, something like "corrected_rain_mm" for example? And in a comment, I can explain it. Or if you don't want this, then the tip count which is read from the sensor, could be provided as a field, so I can multiply with my own, corrected value when I process the tip count in my system. Or I can add both fields of course. Which one shall it be?

zuckschwerdt, 2022-07-26:
Remind me, the 0.3 mm factor is just to match the display, which is wrong? Or are there other units for which 0.3 mm per tip is the correct factor?

As a rule of thumb we don't want to replicate receivers/displays but standard values, so I'd say replace 0.3 with 1.1 and comment on the findings and mention this issue id. Unless 0.3 is the working factor for some other device?

sorny, 2023-11-08:
Stumbled across a set of consisting of two temperature/humidity sensors and one temperature/rain sensor at Aldi, brand is 'Sempre', model name is 4-AH0423-4, looks like the white version of the Auriol

Did some digging into the awesome work that has been done in here for the Lidl Auriol 4-LD6313 and looks like Auriol and Sempre can be decoded in nearly the same way :)
The first byte is `9f ` for the Aldi Sempre rain meter tho.
The two TH sensors were picked up by the `Nexus, FreeTec NC-7345... temperature/humidity sensor` protocol.

Based on that I copied the flags part from `nexus.c` implementation since that was preventing the Auriol 4-LD6313 protocol from picking up the Aldi rain meter...
->
https://triq.net/bitbench#c=1a807ef0000ff&c=1a8081f000100&c=1a8088f000145&c=c680bff00000e&c=c6806bf00000f&c=c680bcf00001b&c=1a00f7f000000%20%5Bbatt%20low%5D&c=1a00faf000001%20%5Bbatt%20low%5D&c=9fb0bcf0000ef%20%5BAldi%20Sempre%20Rainmeter%5D&f=ID%3Ahh%20BATTERY%3Ab%20FLAG%3Ab%20CHANNEL%3A2d%20TEMP_C%3A12d%20h%20RAIN%3A%2024d&cw=4

sorny, 2023-11-08:
Just checked, Sempre 4-AH0423-4 has no dedicated channel selector or any pinout on the pcb for channel selection, so i guess its always `1` for the 2 channel bits for the Sempre, and always `0` for Auriol. 
That bit between battery and channel seems to be consistently `0`, no explanation over at `nexus.c` either for what that bit would do