merbanan/rtl_433 #1838: Hopping on successful event for Freq1 and wait time for Freq2

bobhavlin, 2021-10-20:
Feature request of an index to control whether a hop occurs after a successful event or should wait time.  
i.e. `rtl_433 -f 433M -f  912M -H 15 -E hop -E wait` ( where "wait" stays on the 912M freq for the full -H 15 regardless of events).

I have a situation where the 433M freq has numerous signals I want to read for the -H interval, but the 912M has only one - so one and done.  

Great work here!!  Thanks!

zuckschwerdt, 2021-11-11:
We could add `-E` to the list of things that index with current freq. A more general idea would be to put all hop related args into a struct and switch all of those.
We need an command arg structure more suited to that though. Something like "every -f starts a new set of args", e.g.
`rtl_433 -f 433M -s 250k -Y classic -H 15 -f 912M -s 1024k -Y minmax -E hop`
but randomly mixing -f -s -H will give strange results, mostly incompatible with what we provide now :/

Ian-Zz, 2021-12-29:
I think I would love this feature.
I would like to give Bob's cheesy hack a try but I am running Home Assistant OS on a Pi4 and using this HA addon pbkhrv/rtl_433-hass-addons.
My problem not exactly sure how I would go about changing/adding/editing to Bob's rtl_433.c file locally on my machine. And after that if that new parameter would even be available to use in the rtl_433 HA add on after that. Gonna pose this question over there too.

Btw... GREAT software this is. I LOVE IT!!!

zuckschwerdt, 2021-12-29:
I think I'll look at adding this all-options-per-hop idea in the weeks to come. But like noted it will break the way the arguments work right now, so likely this will sit in a branch for a while.

Ian-Zz, 2022-01-07:
fyi...
Having trouble with the home assistant add-on aspect of Bob's hop2 repo (https://github.com/bobhavlin/rtl_433) but as a stand alone install on a separate Pi3 his code does just what I needed. My meter readings have not skipped an hour yet.

zuckschwerdt, 2022-01-07:
The `hop2` idea looks good enough until we can get proper support in rtl_433.