merbanan/rtl_433 #1839: TPMS T6

bondiblueos9, 2021-10-20:
I purchased a TPMS from amazon here: https://www.amazon.com/gp/product/B09287DLVG

It is decoded with the tpms_jansite decoder but incorrectly.

I was able to make it work by changing the preamble_pattern to {0xaa, 0xaa, 0xa9, 0x66} in the callback function (along with changing 24 to 32 in the calls to bitbuffer_search and to the decode).  Then in the decode function changing the id_str from 7+1 to 8+1, the code_str from 7*2+1 to 9*2+1, and the 56 to 72 in the manchester_decode call and packet_bits condition.

For the actual data then (9 bytes total):
the id (as displayed on the TPMS) is bytes 3, 2, 1, and 0
byte 4 is maybe the sensor wheel position (mine are 00, 01, 02, and 03)
byte 5 might be some sort of flags
byte 6 is the pressure in kPa (no need to multiply by 1.7)
byte 7 is the temperature in C (no need to subtract 50)
byte 8 may be a checksum

Hopefully, this can help someone else.  If there was a way to do the manchester decode with a flex decoder on the command line figuring this out would have been easier.  If there is and it's in the docs I missed it!

bondiblueos9, 2022-02-28:
My TPMS seems to be decoded properly by protocol 201 now (which did not exist when I opened this issue).

zuckschwerdt, 2022-02-28:
For reference: that TPMS is from #1893