merbanan/rtl_433 #1844: Toyota PMV-C210 sensor not reporting at regular intervals

ezhik1, 2021-10-30:
Hi, I've completed a version of this project and can successfully plot readings from 5 sensors!

I am able to excite the sensors by applying ~5 psi and releasing pressure rapidly; they start to transmit every couple seconds for a few transmissions (10 or so.)

The issue is, these sensors stop emitting signals after resting for a bit, and only pressure differences wake them up.

Since they are new sensors, and will not be used in a car with TPMS (1996 jeep in my case) I've been looking for activation procedures. Does anyone know how to activate these sensors beyond their apparent factory sleep mode ?

Knowns:

PMV-C210 sensors from some alibaba brand 'BAIXINDE' https://www.amazon.com/gp/product/B09CMMRYHS
Sensor all report correctly when provoked (see above)
Scan tool Autel MaxiTPMS TS508 successfully triggers and reports ID, Temp, Pressure of all mounted sensors
No report from sensors, stationary or movement, only when pressure changes
Unknowns:

Do the sensors report at regular intervals when stationary/movement without a host car? Thinking maybe a host signal triggers sensors to report

If in fact there is a factory sleep mode, how do I get the sensors out of it?

From one source, I've been told to use this tool for initial sensor activation:

ATEQ VT37: https://www.amazon.com/gp/product/B08H8TKSW9

It's in the post, so I suppose I'll find out soon enough 😃 Meanwhile, perhaps someone can shed light on the topic!

zuckschwerdt, 2021-11-14:
The most common input to wake the sensors is rotation, also rapid pressure loss/change. It's rather uncommon to see periodic updates when in sleep mode, but there have been some sensors that do that.

gdt, 2023-10-14:
This is a question and not about rtl_433 but adjacent.  Feel free to use discussions or the list.

closed 2023-10-14 by gdt