merbanan/rtl_433 #1845: Random dropped reception of thermostats after upgrade of the hardware

louisnichols, 2021-11-02:
Hi!

I am running rtl_433 using Docker on a Raspberry Pi 4, using [IOTstack](https://sensorsiot.github.io/IOTstack/). The version of rtl_433 is "21.05-105-g4cf34682 branch master at 202110112341"

I am trying to receive and process several sensors, of which most are of this model: 
    https://www.ebay.de/itm/373768320944?ViewItem=&item=373768320944

I also have 2 weather sensors of different types (can't find links for them anymore) and a motion sensor. On the same raspberry Pi, I have a 433Mhz receiver and transmitter, both on GPIO, to control 4 remote electrical plugs.

The RTL receiver I use is one of these cheap blue ones which are pretty common ([example](https://www.amazon.de/DollaTek-Digitale-Fernsehtuner-Empf%C3%A4nger-Unterst%C3%BCtzung/dp/B07DJT5NHD/ref=pd_sbs_2/258-7073153-2644455?pd_rd_w=FYerf&pf_rd_p=b1c388c3-48c2-4960-8532-fa8f1477aee9&pf_rd_r=FRRHTYB8M3C2HPPVDQ3V&pd_rd_r=6a448388-1796-4754-9cc9-d87979990c4f&pd_rd_wg=KgrDR&pd_rd_i=B07DJT5NHD&psc=1)).

I started this setup 3 weeks ago, with 6 of the Elecsa thermostats, but otherwise mostly the same (433MHz-wise - details below). For the first two weeks, it worked quite well. From the weather sensors there were some occasional message drops, but nothing consistent. **Reception from all sensors was stable**. During this interval, the RTL-SDR dongle was plugged directly into a USB port of the RPi, without any issues.

Then, about a week ago, I upgraded the setup by adding a powered USB hub, a couple of HDDs and coil antennas on the 433Mhz GPIO transmitter and receiver. I also placed a step-up DC converter for the transmitter, to increase the range a bit. And packed it all on a stacked structure. Originally, the RTL dongle was plugged in the USB hub.

Ever since, reception from the Elecsa sensors has been unstable. In random order, the setup stops receiving data from them for long periods of time. In fact, once it "loses" a sensor, it doesn't recover. I sometimes open one of them, change the channel and then it starts being received again. But a few minutes later, another one drops. And another one. And so on. Until now, I intervened whenever I had 3 sensors disconnected, by changing the channel or powering them off and on.

Initially, I thought it might be because of the distance to the sensors, even though I was losing ones that never had an issue before the upgrade. But then I lost even the one in the same room with the RPi, which is some 3 meters away in direct line of sight. Besides, when I bring a "lost" sensor right next to the RPi, it doesn't recover. Also, I don't routinely lose the same sensors, so it doesn't seem to be because of them. So far it's been a random game of whack-a-mole, where I literally lost and recovered each one of them at least once.

So now I am a bit lost, because it's such a difference from how it worked pre-upgrade. I find it hard to believe that the 3 extra sensors create so much interference as to generate such chronic disconnects. I read somewhere that it's a good idea to not plug the RTL dongle directly in the RPi (or a hub, I guess). So I connected it via an extension cable, even though that was not an issue before. No joy. I also suspected the 433MHz transmitter, so I cut power to it. Still no joy. I tried to place the antenna in various places and with various orientations (as much as the length of the cable allowed). Still no joy.

I can't think of other sources of interference (which would be different from the pre-upgrade status). I live in a detached house and neighbors are not very close. Judging by what I do receive with rtl_433 there aren't many devices around apart from mine. I don't know about the two extra HDDs and the USB hub from the setup. Could they be causing such bad interference? Even with the dongle 1 meter away?

Unfortunately, I run the RPi headless on the minimal OS, so I can't check what is going on with SDR software. Also, I have absolutely no experience with that, so I am not sure I would know what to do either way.

I am hoping there are ways to troubleshoot this without such software. I tried doing a few steps with rtl_433 itself, by running it with verbose modes, but I'm afraid I don't quite know how to proceed. I am an engineer, but in a different field. I understand some of the basics, but not much more. I will greatly appreciate any suggestions.

zuckschwerdt, 2021-11-02:
Yes, a usb cable is always a good idea, the receiver is not shielded and interference drops quickly with distance.

Changing the channel is a red herring. It's just a data bit. The battery reset might have changed the carrier frequency.
Perhaps you are right on the edge of the band?

Even headless you can check the band, rtl_433 got you covered! Use `-w myfile_433.92M_250k.cu8 -T 60` for a 60 second snapshot (adjust filename and parameters to your -f and -s setting if not default).
Then get the file and drop on https://triq.org/pdv/ to view the spectrogram.

You can the try `-f` to adjust the center frequency to better fit the band, or try `-s 1M` for a wider band (but more processing power and perhaps worse reception).

Other recommended settings are `-Y autolevel -Y minmax -Y magest -M level -M noise` to get more sensitivity, and insight on signal levels.

louisnichols, 2021-11-03:
Thank you for your support.

I figured about the channel. I just thought I would try _something_. :)

I have created a log and used the online tool to view it. But I'm not sure how to use that for the next steps. Like, when you say "better fit the band", how do I decide that from spectrogram? I can see on it the various messages, but I am not sure which ones are mine. In any case, very few of them are on the same frequency. They are scattered across.

I realize schooling noobs might not be the point of this platform. :) Is there like a tutorial I could follow?

zuckschwerdt, 2021-11-03:
We are a little light on documentation and tutorials :/

You want to investigate if some signals are close to the edges (top/bottom in the spectrogram) then change the center frequency to cover all signals better. Maybe use `-s 1M` when getting the spectrum snapshot, wider band, more too see.

peteakalad, 2021-11-03:
When headless running rtl_tcp is also option, then connect to that from a remote graphical SDR app to see the spectrum/waterfall.  This will give you a live view which can help confirm centre freq for your device by stimulating them somehow e.g. removing/replacing battery.

louisnichols, 2021-11-03:
OK, thank you. I shifted the center frequency a bit lower, so let's see. None of the signals were right on the edge, but there were more of them in the bottom half, so perhaps this helps.

The problem I have right now is that it doesn't seem to publish to mosquitto from the command line. It shows it is connected to the broker and that it is publishing to the correct topic, but there is nothing at the broker. In fact, in this config, I don't even see the messages in the command line anymore. Am I missing something?

zuckschwerdt, 2021-11-03:
You can use multiple outputs, e.g. `-F kv -F mqtt` to see what's going out by MQTT.

louisnichols, 2021-11-03:
OK, thanks. I might try that later, but for the moment I changed the docker container to add the -f option to it. Unfortunately, it doesn't seem to have done the trick.

@peteakalad Thank you for the tip about TCP. Honestly, that particular option sounds quite amazing to me. :) I'll try that. Can you recommend a piece of software which supports that capability?

peteakalad, 2021-11-03:
For windows sdr# (aka sdrsharp) can connect to remote rtl_tcp.  Google ‘rtl_tcp sdr#’ should be some instructions on how to set it up. Also may need to open up firewall port or expose ports using docker to get access to whatever port you use for rtl_tcp.

louisnichols, 2021-11-04:
I managed to get sdr# to work. I am looking at the colorful throwback to analog TVs :) But I don't know how to make it work for me. 

For example, I can see that the "lost sensors" show weaker signals, but what do I do next? Their frequencies are within range, so it's not the frequency. Should I try to increase gain? Is there a good way to decide from sdr# by how much?

I'm still bummed that they stopped working all of a sudden. I checked the database from 2 weeks back and there isn't anything of the sort. Today, I again don't have reception from the sensor which is in the same room (among others).

louisnichols, 2021-11-04:
BTW, publishing to MQTT does seem to have an issue in my case.

I started rtl_433 with
`rtl_433 -F "mqtt://localhost:1883,devices=RTL_433" -F kv`

and it shows
```
.
.
Publishing MQTT data to localhost port 1883
Publishing device info to MQTT topic "RTL_433".
.
.
MQTT Connected...
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
MQTT Connection established.
```
in the command line I am able to see the received messages, but there is nothing at the broker. The topic is correct, as shown by NodeRed
[image]

The server is also the correct one.
`mosquitto_pub -h localhost -p 1883 -t "RTL_433" -m "test"`
shows the message in NodeRed.

peteakalad, 2021-11-04:
Your topic probably needs to include a # wildcard e.g. rtl_433/#

Run 
`mosquitto_sub -h localhost -p 1883 -t 'rtl_433/#' -v -F "%I %t %p"`
To see all the messages from rtl_433 along with the topic its been published on, you'll see the hostname/device etc is involved in the topics.

On sdr# you want to try and check the centre of the sensor signals are the same frequency as what you have rtl_433 set to, or try it set to that freq to see if it improves.

louisnichols, 2021-11-04:
I had tried shifting the center frequency earlier, but no joy.

Anything else I should focus on?

==========================
Yes, it was something like that @ MQTT. I was confused by the syntax.

`rtl_433 -F "mqtt://localhost:1883,`**events**`=RTL_433" -F kv`

returns

`Publishing events info to MQTT topic "RTL_433".`

and it publishes as expected. I just need the basic messages in a single topic. But

`rtl_433 -F "mqtt://localhost:1883,`**devices**`=RTL_433" -F kv`

returns

`Publishing device info to MQTT topic "RTL_433"`

and... I think I understand what it does, but it wasn't what I needed. :)

louisnichols, 2021-11-04:
So the thing is, I could see in sdr# that my "lost" signals are, some of them, closer to the center frequency than others, which are received. So I think it's not the frequency. But they show weaker, I think.

louisnichols, 2021-11-04:
I have one question about https://triq.org/pdv/ . I don't quite understand the axes.

The Y is the frequency, but in my case it shows a range from -500 to +500 "mHz".

The X is time which shows values from 0 to 14.8 "Ms".

These are for a file created with the command:

`rtl_433 -w myfile_20211104_1.cu8 -T 60`

What does that mean?

I am trying to interpret a file where, for the same sensor, one message was received and another (earlier) one wasn't. I recorded the timestamps for both, but the X axis confuses me.

BTW: One thing I noticed is that, when I trigger the sensor via its TX button, the rate of reception is much better. Like, almost all such messages are received. Does that tell us anything useful?

zuckschwerdt, 2021-11-05:
There is no metadata in the files to pick axis ranges. You need to have the labels on the filename, e.g. `myfile_868.3M_1000k.cu8` (obviously with the values you actually use.)

Did you try to add `-Y autolevel -Y minmax -Y magest -M level -M noise` to get more sensitivity and show the reception levels?

louisnichols, 2021-11-05:
> There is no metadata in the files to pick axis ranges. You need to have the labels on the filename, e.g. `myfile_868.3M_1000k.cu8` (obviously with the values you actually use.)

And does the tool read those from the filename? I thought they would be just for my reference. 
> 
> Did you try to add `-Y autolevel -Y minmax -Y magest -M level -M noise` to get more sensitivity and show the reception levels?

Yes. I spent several hours last night, using those options and a few others in all kinds of combinations, in addition to all kinds of placements of the dongle and the antenna. None of them worked. When some settings would resurrect one sensor, they would lose another. :(

louisnichols, 2021-11-11:
One thing I noticed is that some command line options cause some sensors to be received, but not all. Other options receive the other sensors.

So I had the idea to stream the raw data via rtl_tcp and execute two instances of rtl_433 in different terminal sessions, each with different settings.

No idea if this has any chance of working, but I ran out of ideas.

However, it seems rtl_tcp refuses the second connection. Is there anything I can do to work around this?

zuckschwerdt, 2021-11-11:
There are some tools to split or repeat a source, you need to specificly look for that.
What are the sets of options to receive some groups of sensors?
A crude workaround might be to use `-T 30` and loop over starting rtl_433 with different options.

Maybe take a snapshot and post the .cu8 as zip here. Also try to read the snapshot back with various options to fine tune maybe?

Closing as there seems nothing to change in rtl_433, but please keep updating on your findings.

closed 2021-11-11 by zuckschwerdt