merbanan/rtl_433 #1846: Cannot find a specific device (V22 thermostat)

retrodans, 2021-11-04:
Hello, I have only just installed the rtl_433 script, and it appears to run nicely, I can find some of my neighbours devices, along with my own doorbell, which was the main reason I wanted to get this working.

But looking at my wireless thermostats is confusing me,  and it may be a basic misconception on my part.

In the manual for my v22 thermostat (https://flexel.co.uk/shop/controls-accessories/wireless-programmable-room-thermostat/) it says the frequency is 868 MHz, but when I listen on that frequency (rtl_433 -f 868M) and tweak some of the settings, it doesn't appear to hear the signal.

How is an ideal way to go about finding the frequency, or could it be something else completely (such as that rtl_433 is unable to hear it due to something I need to learn)?

Thanks for any time/help you are able to give,
Dan

zuckschwerdt, 2021-11-04:
It should work. At least `-A` should show you the packets. Not sure if your sensor is already supported.
Sometimes at better center frequency to pick is 868.3M. Try to use a SDR UI (CubicSDR, Gqrx, SigDigger, SDRangel) to observe the band for transmissions. Otherwise record for a short time (-w filename_868M_1000k.cu8 -T 60) and view on https://triq.org/pdv/

zuckschwerdt, 2021-11-04:
If you remove the receiver antenna and place one sender at 5-10cm that should get rid of other signals.

retrodans, 2021-11-04:
Thanks, handy tip, will give that a go later too

zuckschwerdt, 2021-11-11:
Reopen with a sample recording (post a zip) and some description of the thermostat if you need further help.

closed 2021-11-11 by zuckschwerdt