merbanan/rtl_433 #1849: Unable to decode Acurite 592TXR

thenerdie, 2021-11-06:
Hello, all!

I'm having difficulty decoding signals from my outdoor 592TXR sensor using HackRF + SoapySDR. I compiled Soapy and rtl_433 from source, with no difficulty. Running `SoapySDRUtil --probe` results in:

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] Opening HackRF One #0 88869dc2d264b1b...

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=HackRF
  hardware=HackRF One
  clock source=internal
  part id=a000cb3c006a4f6b
  serial=0000000000000000088869dc2d264b1b
  version=2018.01.1
```

When I try to run `rtl_433` by itself, I get:

```
rtl_433 version 21.05-99-g481270e8 branch master at 202110071805 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/kisperal/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 166 out of 195 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-195 ]
No supported devices found.
```

I can force it to use my HackRF by using `-d driver=hackrf` (also specifying the frequency using `-f 433.975MHz`), which results in:

```
rtl_433 version 21.05-99-g481270e8 branch master at 202110071805 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/kisperal/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 166 out of 195 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-195 ]
[INFO] Opening HackRF One #0 88869dc2d264b1b...
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM_cs16: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

And nothing else after this. If I go into gqrx and scan the 433MHz band, I can see where my 592TXR is transmitting.

[image]

To make sure that this was indeed my 592TXR, I brought it inside and waited. Sure enough, the signal was much stronger. I'm not entirely sure what's going on here; I've debugged this for hours to no avail. Any help would be appreciated.

zuckschwerdt, 2021-11-06:
The HackRF uses CS16, and usually the signal level won't be full-scale. It's recommended to use
`-Y autolevel -Y minmax -Y magest -M level -M noise`
(actually `-Y magest` is already the default for CS16 and `-Y minmax` might not be needed for 433 MHz).

thenerdie, 2021-11-06:
Hey! I tried running rtl_433 with the flags you posted, but it still won't decode. I can read the raw pulses with `-A`:

```
rtl_433 version 21.05-99-g481270e8 branch master at 202110071805 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/kisperal/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 166 out of 195 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-195 ]
[INFO] Opening HackRF One #0 88869dc2d264b1b...
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.978MHz.
Estimated noise level is -17.3 dB, adjusting minimum detection level to -14.3 dB
baseband_demod_FM_cs16: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Estimated noise level is -19.3 dB, adjusting minimum detection level to -16.3 dB
Estimated noise level is -21.0 dB, adjusting minimum detection level to -18.0 dB
Estimated noise level is -22.4 dB, adjusting minimum detection level to -19.4 dB
Estimated noise level is -23.7 dB, adjusting minimum detection level to -20.7 dB
Estimated noise level is -24.9 dB, adjusting minimum detection level to -21.9 dB
Estimated noise level is -26.7 dB, adjusting minimum detection level to -23.7 dB
Estimated noise level is -28.2 dB, adjusting minimum detection level to -25.2 dB
Estimated noise level is -29.2 dB, adjusting minimum detection level to -26.2 dB
Estimated noise level is -30.4 dB, adjusting minimum detection level to -27.4 dB
Estimated noise level is -31.6 dB, adjusting minimum detection level to -28.6 dB
Estimated noise level is -32.6 dB, adjusting minimum detection level to -29.6 dB
Detected OOK package	2021-11-06 08:24:09
Analyzing pulses...
Total count:  132,  width: 123.15 ms		(30787 S)
Pulse width distribution:
 [ 0] count:   46,  width:  972 us [736;1180]	( 243 S)
 [ 1] count:   63,  width:  480 us [388;572]	( 120 S)
 [ 2] count:    7,  width: 1676 us [1636;1780]	( 419 S)
 [ 3] count:   15,  width:  368 us [356;392]	(  92 S)
 [ 4] count:    1,  width: 2228 us [2228;2228]	( 557 S)
Gap width distribution:
 [ 0] count:   12,  width:  404 us [344;476]	( 101 S)
 [ 1] count:   18,  width:   56 us [48;68]	(  14 S)
 [ 2] count:   37,  width:  168 us [136;208]	(  42 S)
 [ 3] count:   42,  width:  248 us [212;280]	(  62 S)
 [ 4] count:   11,  width:  124 us [120;140]	(  31 S)
 [ 5] count:    4,  width:   44 us [44;44]	(  11 S)
 [ 6] count:    5,  width:   72 us [72;76]	(  18 S)
 [ 7] count:    2,  width: 2044 us [2028;2060]	( 511 S)
Pulse period distribution:
 [ 0] count:   44,  width: 1224 us [1208;1256]	( 306 S)
 [ 1] count:   77,  width:  612 us [600;624]	( 153 S)
 [ 2] count:    7,  width: 1840 us [1840;1848]	( 460 S)
 [ 3] count:    3,  width: 2840 us [2456;3040]	( 710 S)
Pulse timing distribution:
 [ 0] count:   46,  width:  972 us [736;1180]	( 243 S)
 [ 1] count:   70,  width:  472 us [388;572]	( 118 S)
 [ 2] count:    9,  width: 1756 us [1636;2060]	( 439 S)
 [ 3] count:   20,  width:  368 us [344;392]	(  92 S)
 [ 4] count:    1,  width: 2228 us [2228;2228]	( 557 S)
 [ 5] count:   18,  width:   56 us [48;68]	(  14 S)
 [ 6] count:   37,  width:  168 us [136;208]	(  42 S)
 [ 7] count:   42,  width:  248 us [212;280]	(  62 S)
 [ 8] count:   11,  width:  124 us [120;140]	(  31 S)
 [ 9] count:    4,  width:   44 us [44;44]	(  11 S)
 [10] count:    5,  width:   72 us [72;76]	(  18 S)
 [11] count:    1,  width: 22284 us [22284;22284]	(5571 S)
Level estimates [high, low]:   2074,    393
RSSI: -18.0 dB SNR: 14.4 dB Noise: -32.4 dB
Frequency offsets [F1, F2]:    -251,      0	(-1.0 kHz, +0.0 kHz)
Guessing modulation: No clue...
```
I'm not sure why it won't read it properly.

zuckschwerdt, 2021-11-06:
Those pulses look valid, but not like Acurite 592TXR. It should be 200/400 µs pulses.
Grab a sample (-S unknown) and upload here as zip.

thenerdie, 2021-11-06:
[g001_433.978M_250k.cs16.zip](https://github.com/merbanan/rtl_433/files/7490813/g001_433.978M_250k.cs16.zip)
Here ya go. I'm fairly certain I have a 592TXR, unless the first two zeros represent an entirely different model number..
[image]
.

zuckschwerdt, 2021-11-06:
I see, the DC spike interferes with the pulse detection. It's always better not to tune dead center, the default of 433.92M should give more stable results. I don't think there is a DC-spike suppresion option for the HackRF (otherwise you can apply a setting with `-t param=val`).

thenerdie, 2021-11-06:
Alright, thanks! It seems to work when I leave it at 433.92M with the flags you provided above. ❤️

[image]

closed 2021-11-06 by thenerdie