merbanan/rtl_433 #1854: bitbuffer_add_bit: Warning: row count limit (50 rows) reached

emilburzo, 2021-11-09:
# Context

I've been running rtl_433 for about a day to see if I can consistently receive my Ambientweather-F007TH sensors.

I'm using docker for this:

```bash
docker run \
 -d \
 --restart=always \
 --name rtl_433 \
 --device /dev/bus/usb/001/006 \
 hertzg/rtl_433:debian-latest-master \
 -F "mqtt://ip:1883,user=***,pass=****,retain=0,devices=rtl_433[/model][/id]"

```

# Problem

All was working well in the "test day", but now I have started seeing the following:

```
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
```

When this happens, I don't see anything from the sensors for a while, but it seems to recover at some point. 

Thinking about it, I'll enable the `kv` output now to see if they really correlate

# Things I've tried

- enabling only this device decoding protocol (`-R 20`)
- examining verbose output, but didn't see anything that stands out (probably too noob to notice what's wrong)

# Question

Do the warning messages have any real effect or is it just a coincidence that I'm not receiving anything?

zuckschwerdt, 2021-11-09:
Coincidence. When you get to 50 rows you'll very likely received just noise.
You can use `-Y autolevel -Y minmax -Y magest -M level -M noise -M time:usec` to get better reception and more informational output.

emilburzo, 2021-11-10:
It seems you are correct, when the noise level rises sharply, I see the `bitbuffer_add_bit` warning and the data stops

```
Current noise level -31.2 dB, estimated noise -33.3 dB
Estimated noise level is -32.9 dB, adjusting minimum detection level to -29.9 dB
Current noise level -34.2 dB, estimated noise -33.3 dB
Estimated noise level is -33.9 dB, adjusting minimum detection level to -30.9 dB
Current noise level -35.0 dB, estimated noise -34.4 dB
Current noise level -34.2 dB, estimated noise -34.1 dB
Current noise level -34.0 dB, estimated noise -34.2 dB
Current noise level -33.9 dB, estimated noise -33.8 dB
Current signal level -17.6 dB, estimated noise -34.0 dB
Current noise level -35.2 dB, estimated noise -34.6 dB
Current noise level -33.5 dB, estimated noise -33.4 dB
Current noise level -33.7 dB, estimated noise -33.8 dB
Current noise level -34.4 dB, estimated noise -34.5 dB
---------------> data output stops around here <-------------------------
Current signal level -2.8 dB, estimated noise -21.4 dB
Current signal level -1.6 dB, estimated noise -12.4 dB
Current signal level -1.6 dB, estimated noise -7.5 dB
Current noise level -1.8 dB, estimated noise -4.3 dB
Current noise level -1.4 dB, estimated noise -1.8 dB
Current noise level -2.1 dB, estimated noise -1.9 dB
Current noise level -2.4 dB, estimated noise -2.0 dB
Current noise level -1.5 dB, estimated noise -1.7 dB
Current noise level -1.4 dB, estimated noise -1.6 dB
Current noise level -2.3 dB, estimated noise -2.1 dB
Current noise level -1.4 dB, estimated noise -1.5 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -3.7 dB, estimated noise -2.9 dB
Current noise level -1.7 dB, estimated noise -2.5 dB
Current noise level -1.9 dB, estimated noise -2.2 dB
Current noise level -4.5 dB, estimated noise -3.8 dB
Current noise level -5.1 dB, estimated noise -3.1 dB
Current noise level -2.9 dB, estimated noise -2.8 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -2.0 dB, estimated noise -2.0 dB
Current noise level -2.5 dB, estimated noise -2.3 dB
Current noise level -4.5 dB, estimated noise -2.8 dB
Current noise level -1.8 dB, estimated noise -2.2 dB
Current noise level -1.5 dB, estimated noise -1.8 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -2.3 dB, estimated noise -3.2 dB
Current noise level -1.6 dB, estimated noise -2.0 dB
Current noise level -1.7 dB, estimated noise -1.6 dB
Current noise level -1.7 dB, estimated noise -1.7 dB
Current noise level -1.7 dB, estimated noise -1.8 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -1.9 dB, estimated noise -1.9 dB
Current noise level -4.0 dB, estimated noise -2.4 dB
Current noise level -2.5 dB, estimated noise -4.0 dB
Current noise level -2.2 dB, estimated noise -2.8 dB
Current noise level -1.6 dB, estimated noise -3.3 dB
Current noise level -6.2 dB, estimated noise -5.5 dB
Current noise level -4.3 dB, estimated noise -4.1 dB
Current noise level -2.1 dB, estimated noise -3.0 dB
Current noise level -1.7 dB, estimated noise -1.9 dB
Current noise level -4.6 dB, estimated noise -2.6 dB
Current noise level -5.5 dB, estimated noise -5.7 dB
Current noise level -1.6 dB, estimated noise -2.8 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -1.5 dB, estimated noise -1.6 dB
Current noise level -1.4 dB, estimated noise -1.6 dB
Current noise level -1.3 dB, estimated noise -1.4 dB
Current noise level -1.4 dB, estimated noise -1.4 dB
Current noise level -1.3 dB, estimated noise -1.3 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -1.3 dB, estimated noise -1.3 dB
Current noise level -1.4 dB, estimated noise -1.4 dB
Current noise level -1.4 dB, estimated noise -1.4 dB
Current noise level -1.6 dB, estimated noise -1.5 dB
Current noise level -1.4 dB, estimated noise -1.5 dB
Current noise level -1.4 dB, estimated noise -1.4 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -2.5 dB, estimated noise -1.8 dB
```

I guess nothing that can be done on the `rtl_433` side.

Thank you, that was very informative!

closed 2021-11-10 by emilburzo

zuckschwerdt, 2021-11-10:
To put that into perspective: if you are using an 8-bit Receiver (RTL-SDR) the dead-silent level is about -36 dB, usually we expect to see -20 to -28 dB.
A noise level above -15 dB surely means something is wrong. At -12 dB our demod will stop working.
With what you see there is serious interference on the band. Take a look with any of CubicSDR, Gqrx, SigDigger, SDRangel, SDR# or grab a snapshot with `-w myfile_433.92M_250k.cu8 -T 60` (adjust for -f and -s values), then view on https://triq.org/pdv/

emilburzo, 2021-11-11:
I was finally able to catch it when it happens

# no interference

```
Estimated noise level is -31.3 dB, adjusting minimum detection level to -28.3 dB
Current signal level -13.1 dB, estimated noise -31.1 dB
Estimated noise level is -32.4 dB, adjusting minimum detection level to -29.4 dB
Current noise level -33.4 dB, estimated noise -32.7 dB
Estimated noise level is -31.3 dB, adjusting minimum detection level to -28.3 dB
Current noise level -30.6 dB, estimated noise -31.0 dB
Estimated noise level is -30.3 dB, adjusting minimum detection level to -27.3 dB
Current noise level -31.3 dB, estimated noise -30.6 dB
Estimated noise level is -31.3 dB, adjusting minimum detection level to -28.3 dB
Current noise level -32.6 dB, estimated noise -32.2 dB
Current noise level -32.3 dB, estimated noise -31.7 dB
Current noise level -32.3 dB, estimated noise -31.3 dB
```

# interference

```
Current noise level -1.9 dB, estimated noise -1.6 dB
Current noise level -1.4 dB, estimated noise -1.6 dB
Current noise level -1.5 dB, estimated noise -1.5 dB
Current noise level -1.5 dB, estimated noise -1.5 dB
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Current noise level -1.5 dB, estimated noise -1.5 dB
Current noise level -2.5 dB, estimated noise -2.4 dB
Current noise level -1.8 dB, estimated noise -1.8 dB
Current noise level -2.0 dB, estimated noise -2.6 dB
```

# spectrogram

[image]

very good tip with the spectogram, it shows that there is something on ~434.03MHz causing trouble (interestingly, it starts around ~8AM daily, so it must be someone coming in to work bringing something / turning something on)

but is it expected that this affects things on 433.920MHz? I realize they are pretty close, but still surprising to me (as a noob) that it totally drowns everything else

zuckschwerdt, 2021-11-11:
Our demod picks up "all channels", an overall amplitude for OOK.
You can get around the interference by tuning away, 50 kHz down should work.

bbrendon, 2023-03-26:
I was getting this error and I wasn't collecting any data from my sensors. A day later I remember I updated the firmware on a home assistant tagreader. After powering off the tagreader, my sensors all came back!

Offending device:  https://github.com/adonno/tagreader

emilburzo, 2023-04-02:
Some closure also from my side: I could not fix it by changing frequencies, since the whatever-it-is that occasionally floods the airwaves also changes frequency slightly.

Anyway, what fixed it for me was to manually lower the gain (`-g`) from the default max until the noise level is way lower than what was measured before, e.g.:

```
Auto Level: Estimated noise level is -41.1 dB, adjusting minimum detection level to -38.1 dB
```

Since then I've had rock solid reception from all of my 433MHz sensors.