merbanan/rtl_433 #1855: La Crosse LTV-W1 wind sensor reports as R3 Rain Gauge

BarryMar, 2021-11-10:
Info on the sensor is found here: https://www.lacrossetechnology.com/products/ltv-w1

LTV-W1 a pretty simple sensor. Just wind speed (and probably a low battery indicator, but I don't have any dead C cells so far).

When rtl_433 branch master at 202110291420 receives a signal from this unit it mistakenly identifies it as a R3 Rain Gauge and spits out garbage.

Sample recording is available in https://github.com/merbanan/rtl_433_tests

zuckschwerdt, 2021-11-10:
The data shows the wind speed as (dezi)km/h plainly. Strangely that seems the only value transmitted.
See [this BitBench](https://triq.net/bitbench?c=d2aa2dd40fb220%20055555550002aabc8000000000000000%5B0%20km%20Good%20battery%5D&c=d2aa2dd40fb220%200caaaaaa05dad49a00000000000000%5B9%20km%20Good%20battery%5D&c=d2aa2dd40fb220%200eaaaaaa07faaafe00000000000000%5B13%20km%20Good%20battery%5D&c=d2aa2dd40fb220%2000aaaaaa074aaac800000000000000%5B12%20km%20Good%20battery%5D&c=d2aa2dd40fb220%2002aaaaaa0bfaaaad00000000000000%5B19%20km%20Good%20battery%5D&c=d2aa2dd40fb220%200255555500d6aa85000000000000000%5B5%20km%20Good%20battery%5D&c=d2aa2dd40fb220%2008aaaaaa011aaa3900000000000000%5B4%20km%20Good%20battery%5D&c=d2aa2dd40fb220%200aaaaaaa000aaaf200000000000000%5B2%20km%20Good%20battery%5D&c=d2aa2dd40fb220%2006aaaaaa000aaada00000000000000%5B0%20km%20Good%20battery%5D&c=d2aa2dd40fb220%200eaaaaaa000aaa0500000000000000%5B0%20km%5D&c=d2aa2dd40fb220%200b555554001555e400000000000000%5B0%20km%5D&c=d2aa2dd40fb220%2006aaaaaa000aaada00000000000000%5B0%20km%5D&c=d2aa2dd40fb220%200eaaaaaa000aaa0500000000000000%5B0%20km%5D&c=d2aa2dd40fb220%200aaaaaaa000aaaf200000000000000%5B0%20km%5D&f=ID8h8h8h%208h%208h%208h%208h%20SPD%3F12d%2012h%208h%20CHK%3F%208h%20TRAILER%208h8h8h8h8h8h8h8h&a=Preamble&m=d2aa2dd4&cl=56&cw=8).
The codes with `555` feel like a bit is missing. I'll check the checksum next.

zuckschwerdt, 2021-11-10:
I guess it's really the same protocol as the LTV-R3, but with `aa aa aa WWW aaa` (literal `a`, W=wind speed) instead of 6 bytes rain value. We could check on that magic `aa` and switch to show W1 values.

closed 2021-11-10 by zuckschwerdt

zuckschwerdt, 2021-11-10:
Try to get low battery code or sample if you can. Known flat batteries or a variable voltage power supply, if you have one.

BarryMar, 2021-11-10:
As luck would have it, I remembered that I found an old flashlight the other day that had some C cells that were at least 10 years old.  I still had to burn them for a bit in order to make them weak enough to trip the sensor.

In the sequence below, I first put fresh batteries in.  During this time the sensor beeps 4 times and sends 5 * 915 MHz signals out approximately every 10 seconds, of which I only got 3 for some reason.

I then replaced the batteries with the weak ones.  Same 4 beeps and 5 * 915 MHz signals every 10 seconds.  The La Crosse CA80550 display then showed a Low Battery symbol on the Wind Speed sensor.

rtl_433 -f 915M -s 1M -R 0 -X 'n=LTV-W1,m=FSK_PCM,s=100,l=100,r=9000,preamble=d2aa2dd40fb220'

rtl_433 version 21.05-122-g4bb75b4c branch master at 202110291420 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/barrymarshall/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Disabling all device decoders.
Registered 1 out of 200 device decoding protocols [ ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 915.000MHz.
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us

Wind Speed: 0 km, Good Battery  (all samples)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:29:20
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 120          data      : 42aaaaaa000aaa7300000000000000
codes     : {120}42aaaaaa000aaa7300000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:29:32
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 120          data      : 44aaaaaa000aaa6700000000000000
codes     : {120}44aaaaaa000aaa6700000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:29:52
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 119          data      : 0aaaaaaa000aaaf200000000000000
codes     : {119}0aaaaaaa000aaaf200000000000000

Wind Speed: 0 km, Weak Battery  (all samples)

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:30:43
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 120          data      : c2aaaaaa000aaacf00000000000000
codes     : {120}c2aaaaaa000aaacf00000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:30:55
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 120          data      : c4aaaaaa000aaadb00000000000000
codes     : {120}c4aaaaaa000aaadb00000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:31:02
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 116          data      : c6aaaaaa000aaa380000000000000
codes     : {116}c6aaaaaa000aaa380000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:31:08
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 120          data      : c8aaaaaa000aaaf300000000000000
codes     : {120}c8aaaaaa000aaaf300000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-10 14:31:15
model     : LTV-W1       count     : 1             num_rows  : 1
rows      : 
len       : 119          data      : 8aaaaaaa000aaa4e00000000000000
codes     : {119}8aaaaaaa000aaa4e00000000000000
^CSignal caught, exiting!

zuckschwerdt, 2021-11-10:
That worked out well! As you can see in [this BitBench](https://triq.net/bitbench?c=d2aa2dd40fb220%2042aaaaaa000aaa7300000000000000%20%5Bgood%5D&c=d2aa2dd40fb220%2044aaaaaa000aaa6700000000000000%20%5Bgood%5D&c=d2aa2dd40fb220%200aaaaaaa000aaaf200000000000000%20%5Bgood%5D&c=d2aa2dd40fb220%20c2aaaaaa000aaacf00000000000000%20%5Bweak%5D&c=d2aa2dd40fb220%20c4aaaaaa000aaadb00000000000000%20%5Bweak%5D&c=d2aa2dd40fb220%20c6aaaaaa000aaa380000000000000%20%5Bweak%5D&c=d2aa2dd40fb220%20c8aaaaaa000aaaf300000000000000%20%5Bweak%5D&c=d2aa2dd40fb220%208aaaaaaa000aaa4e00000000000000%20%5Bweak%5D&f=ID8h8h8h%20BATTLOW1b%20STARTUP1b%202b%20CH3h%201b%208h%208h%208h%20SPD%3F12d%2012h%208h%20CHK%3F%208h%20TRAILER%208h8h8h8h8h8h8h8h&a=Preamble&m=d2aa2dd4&cl=56&cw=8), the Batter Low flag is first, then, I guess, a "Startup" bit. I'll include that in the decoder.

BarryMar, 2021-11-11:
I tried compiling the latest sources and it looks good.  I'll admit compiling stuff isn't my forte, so this is what I tried:

git clone https://github.com/merbanan/rtl_433.git
cd build     
cmake ../
make
sudo make install 
cd /usr/local/bin
 ./rtl_433 -f915M

 ./rtl_433 -f915M
rtl_433 version 21.05-129-gc21d1e13 branch master at 202111111643 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/barrymarshall/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 169 out of 200 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 915.000MHz.
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-11 14:57:58
model     : LaCrosse-TH3 Sensor ID : 13efb7
Sequence  : 5            unknown   : 0             Temperature: 11.5 C
Humidity  : 59 %         Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-11 14:57:59
model     : LaCrosse-W1  Sensor ID : 0fb220
Battery level: 1         Sequence  : 5             Wind Speed: 4.1 km/h
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-11 14:59:34
model     : LaCrosse-W1  Sensor ID : 0fb220
Battery level: 1         Sequence  : 0             Wind Speed: 4.7 km/h
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-11 15:00:37
model     : LaCrosse-W1  Sensor ID : 0fb220
Battery level: 1         Sequence  : 2             Wind Speed: 4.5 km/h
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-11 15:01:18
model     : LaCrosse-TH3 Sensor ID : 13efb7
Sequence  : 1            unknown   : 0             Temperature: 11.5 C
Humidity  : 59 %         Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-11 15:02:43
model     : LaCrosse-W1  Sensor ID : 0fb220
Battery level: 1         Sequence  : 6             Wind Speed: 4.5 km/h
Integrity : CRC

commit 69ba4a1d (2021-11-10): Add support for LaCrosse LTV-W1 wind sensor (closes #1855)