merbanan/rtl_433 #1856: "Unknown" subtype for Interlogix smoke detector

atkulp, 2021-11-10:
I have the GE Interlogix Wireless Combo Heat/Smoke Detector (TX-6010-01-1) but it shows up as unknown for device subtype.  I'm not sure what I can provide, but here are some payloads with no alarm and cover closed:

`64012c`
`e401bc`
`0400a4`
`840034`

I'd be happy to add the device type ID to the code, but I'm not completely sure how to get it.  I'm also not completely sure of which switch to check for a detection event.  I've been watching switch1, but it didn't change when we burned something the other day.

zuckschwerdt, 2021-11-11:
The device subtype is the first nibble (character) of the ID, not the payload (raw message).
But there is no known mapping for most subtypes, we are considering to just remove subtype.
Do none of the messages show a flip in one of the five switches?

atkulp, 2021-11-11:
Please don't remove it!  It works great other than smoke.  My device ID is `21db11`, so `0x2` seems to be smoke subtype.

zuckschwerdt, 2021-11-11:
Do all your smoke detectors start with 0x2? @Sunoo reported a smoke detector in #1708, also 0x2.
I distincly remember a discussion that the ID can't be trusted though?

atkulp, 2021-11-11:
I only have one smoke detector so I can't be sure.  Across all my other devices the id/subtypes are 100% consistent though.

As for the payload, it's always OPEN|CLOSED|CLOSED|CLOSED|CLOSED.  I'll try a few more tests.  I only had that one event so I'll try to force another so I can see if switch1 really never closes.  I assumed tamper would show opening it too.  I'll report back on that after a bit more testing.

Sunoo, 2021-11-11:
For what it’s worth, I never tracked down the other 0x2 device, and I only see it occasionally. Perhaps a neighbor has an Interlogix smoke detector just on the edge of radio range or something.

zuckschwerdt, 2021-11-11:
There is test spray to trigger smoke detectors. E.g. "LOGILINK RP0011" worked for me.

atkulp, 2021-11-11:
Now you tell me!  😭
[image]

zuckschwerdt, 2021-11-12:
Where is the "This Is Fine" dog with his coffee? ;)

gdt, 2023-10-05:
Is this still an issue with git master?  Is anyone working on this?