merbanan/rtl_433 #1857: Linking Error on Ubuntu 20.04

patriciogonzalezvivo, 2021-11-11:
Hi, I'm having troubles compiling from source in Ubuntu 20.04 this is the message

```
 93%] Linking C executable rtl_433
/usr/bin/ld: libr_433.a(sdr.c.o): in function `sdr_open':
sdr.c:(.text+0xcd8): undefined reference to `SoapySDR_free'
/usr/bin/ld: sdr.c:(.text+0xeb8): undefined reference to `SoapySDR_free'
/usr/bin/ld: sdr.c:(.text+0x1235): undefined reference to `SoapySDR_free'
/usr/bin/ld: sdr.c:(.text+0x12f3): undefined reference to `SoapySDR_free'
/usr/bin/ld: sdr.c:(.text+0x13a5): undefined reference to `SoapySDR_free'
/usr/bin/ld: libr_433.a(sdr.c.o):sdr.c:(.text+0x14b7): more undefined references to `SoapySDR_free' follow
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/rtl_433.dir/build.make:90: src/rtl_433] Error 1
make[1]: *** [CMakeFiles/Makefile2:1012: src/CMakeFiles/rtl_433.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Any idea what could be?

zuckschwerdt, 2021-11-11:
S. #1853
Is that a stock Ubuntu 20.04 libSoapySDR?
I would like to known where this SoapySDR 0.8 lib without SoapySDR_free() originates from?

closed 2021-11-11 by zuckschwerdt

patriciogonzalezvivo, 2021-11-11:
Good point! compiled SoapySDR from source (0.8 https://github.com/pothosware/SoapySDR/wiki/BuildGuide ) instead of the Ubuntu packages (0.7) and the issue was solve.

Thank you for your prompt response

zuckschwerdt, 2021-11-11:
So the Ubuntu packages are 0.7 (no SoapySDR_free) but you had header files from 0.8 (indicating SoapySDR_free)?
Did you mix 0.7 and 0.8?

patriciogonzalezvivo, 2021-11-11:
exactly. Currently I have both