merbanan/rtl_433 #1858: schraeder.c is not in sync with the rtl_433_tests repository

obones, 2021-11-11:
Hello,

I have made changes to rtl_433 that I want to validate before submitting them as a pull request. To do this, I use the https://github.com/merbanan/rtl_433_tests repository to check most decoders.
All tests ran fine except for the 3 related to schraeder.c which gives me that kind of error:

```
 Expected: {'model': 'Schrader-SMD3MA4', 'type': 'TPMS', 'id': '3F0650', 'flags': 7, 'pressure_PSI': 32.8, 'mic': 'PARITY'}
  But got: {'model': 'Schrader-SMD3MA4', 'type': 'TPMS', 'flags': 7, 'id': '3F0650', 'pressure_PSI': 32.8}
```
I see that on October 29th, in commit https://github.com/merbanan/rtl_433_tests/commit/0f085f6cbb2a9c4d376c8794f47bc11a28e909ab  there was an update to the expected file but I don't see any associated commit in this repository. 

Is this something that is in a branch waiting to be merged in `master` ?

zuckschwerdt, 2021-11-11:
https://github.com/merbanan/rtl_433/issues/1734#issuecomment-860038359
got stuck on the different pressure scalings.

obones, 2021-11-11:
No worries, no emergency, just wanted to be sure it was not missed in the middle of other things