merbanan/rtl_433 #1861: Techem radio 4

Luceoria, 2021-11-12:
Hi,

I happen to have a "Techem radio 4" water meter sensors at home which I played with recently. It seems generic m-bus headers are supported - Mode, Manufacturer, ID, Type (0x62 / 0x72), etc are decoded properly. But not the custom Techem payload where the measured values reside.

After some research, I can conclude that meters send two types of protocols: C and T. Mode T seems to be more straightforward, few subsequent meter reads reveal only few bits are changed (water or time flow I guess). For mode C, more than half of the payload is changed in subsequent frames, looks like it is encrypted or at least coded.

I can upload the frame data in hex, or a raw frame recording, if needed.

Thanks

zuckschwerdt, 2021-11-14:
@merbanan and maybe @jedi7 can help with m-bus.

merbanan, 2021-11-14:
I would suggest the https://github.com/weetmuts/wmbusmeters project for help with this. You can use rtl_433 as a "driver" for wmbusmeters. What you describe looks to be supported in that project. I am closing this ticket for now.

closed 2021-11-14 by merbanan