merbanan/rtl_433 #1864: Periodic noise metadata output is broken and conflicting with "level"

drws, 2021-11-12:
I'm using manually-built `rtl_433` (latest release 21.05) on a Raspberry Pi running latest Arch ARM. Setting the periodic noise estimation output appears to not work. No noise reports are printed, even if running only `rtl_433 -M noise` with no additional settings. Setting the period higher or lower than the default (10s) doesn't change anything.

Furthermore, setting the `-M noise` parameter appears to conflict with `-M level` in the following way:

- `rtl_433 -M level` output is as expected, `rtl_433 -M noise` is not (as per the first paragraph)
- `rtl_433 -M noise -M level` outputs _level_, but again no _noise_
- `rtl_433 -M level -M noise` doesn't output either of the two!

Even the order of both arguments changes behaviour so there might be two or even more underlying issues involved. I've also confirmed these issues on an x64 PC, but please confirm, I might be holding it wrong(TM) :)

zuckschwerdt, 2021-11-12:
The `noise` stats where introduced after release 21.05. Can you retry with the master branch?

drws, 2021-11-12:
Well, it appears I was holding it wrong after all. Building from current master resolves all of the issues. Version 21.05 shouldn't accept `-M noise` if noise estimation wasn't properly supported at the time, but it is resolved now.

closed 2021-11-12 by drws