merbanan/rtl_433 #1865: Ability to report signal modulation and data rate

drws, 2021-11-12:
By setting the `-M level` parameter, `rtl_433` outputs modulation, frequency, RSSI, SNR, and noise metadata. A welcome addition would be ability to output the detected signal's modulation and/or data rate from the decoder. Additional metadata keyword to support this output would probably be preferred to extending `level` for compatibility reasons.

zuckschwerdt, 2021-11-12:
Those are fixed values. You can look them up in the decoder source or click around in https://triq.org/explorer/
We might add an API to get decoder parameters for the http output, but likely not for other outputs.
If you want all the details, try `-F http` and set "Modes" to "Raw (all)" ;)

zuckschwerdt, 2021-11-12:
On the command line try `-A` for detailed output if want to get a feel what's on the band.

gdt, 2023-10-02:
I read this as not only maintainers won't spend time, but don't think it's a good idea.  Will let it sit for the 30-day timer before closing.

closed 2024-06-23 by gdt