merbanan/rtl_433 #1867: CSV header line written to file even if device not detected

drws, 2021-11-12:
`rtl_433` currently writes CSV header data line before trying to find the input device. This results in unnecessary write cycles and if it is automatically restarted (running as a persistent systemd service for example), appending the output file with header lines might produce huge files over time or even fill up the partition. I propose the log file is rather initialized after successful input source detection.

zuckschwerdt, 2021-11-12:
Good idea. Difficult to implement though. There are multiple sources for input (files, test args, ...) and the outputs start before all of those options. We would need to delay the start and put it into each individual case...

gdt, 2023-10-05:
I'm stopping short of closing this, but Ive added wontfix since it seems like a fair bit of code ick to make it work (and I agree it would be nice behavior).

closed 2024-06-05 by gdt

drws, 2024-06-05:
@gdt: Is it really fixed now or just archived?

zuckschwerdt, 2024-06-05:
It's won't-fix for now. And there might be users that just want to quickly see the header without waiting for some decoder to work. E.g. `rtl_433 -F csv -T 1` is useful to just check things.

drws, 2024-06-05:
The header can still be printed while the file output is delayed after HW initialization. Does that really need so much rework? Would you please elaborate on it a bit?