merbanan/rtl_433 #1870: LaCrosse TX29UDTH-IT sensor not detected

varinbox, 2021-11-14:
I have a TX29UDTH-IT operating at 915MHz but the decoder doesn’t seem to pick it up. It seems like it should be supported since TX29U is listed. Any ideas?

zuckschwerdt, 2021-11-14:
# Trouble-shooting a sensor

## Overview of the band

Get on overview of the band. Check if the transmission is visible and in the expected frequency range.
Use CubicSDR, Gqrx, SigDigger, SDR#, or SDRangel.
If you have the SDR reciever on a headless machine try `rtl_tcp` to transport data to a GUI.

A poormans substitute is to record a sample `-w file_433.92M_250k.cu -T 60` (adjust for the actual frequency and sample rate).
Then drop that .cu8 file on https://triq.org/pdv/ to view.

## Center frequency

Pick a good center frequency. Don't pick the transmission frequency, use a value 10k-50k aside.
The band covered is equal to the sample rate.
At the default 433.92M and 250k sample rate that's 433.67 MHz to 434.17 MHz.
For the 868M default sample rate of 1024k that's 867.5 MHz to 868.5 MHz.
For the 868M it's like good to pick 868.3M for a band of 867.8 MHz to 868.8 MHz.

## Grabbing a sample

You can grab samples of just the transmissions with `-S all` or `-S unknown`.

If you remove the receiver antenna and place a sender at 5-10cm that should get rid of other signals.

Analyze the sample or live data with `-A`.

## More options

Be sure to also try with higher sensitivity: `-Y autolevel -Y magest -M noise -M level`

Change the sample rate, for 433M try `-s 1024k`, for 868M try `-s 250k` or  `-s 2048k`.

Use a different demod, try `-Y minmax` for 433M, for 868M try `-Y classic`.

closed 2021-11-14 by zuckschwerdt

zuckschwerdt, 2021-11-14:
S.a. https://triq.org/rtl_433/ANALYZE.html