merbanan/rtl_433 #1871: Exploratory work to build under Visual Studio 2019 with CMake

obones, 2021-11-15:
This issue is more like a "reminder" as to what needs to be done to get Visual Studio 2019 to build `rtl_433` by using CMake.
Once everything is sorted out, there might be a need to create/update documentation, change configuration files and so on, all to be placed into pull requests.

- Create submodule for `vcpkg`
```
[submodule "vcpkg"]
	path = vcpkg
	url = https://github.com/microsoft/vcpkg
	branch = master
``` 
- Init/Update the submodule
- Open a command line in the `vckpkg` folder and call `bootstrap-vcpkg`
- Install openSSL: `vcpkg --triplet=x64-windows install openssl`
- Install libusb: `vcpkg --triplet=x64-windows install libusb`
- Start Visual Studio 2019
- Choose "Open a folder" and indicate the `rtl_433` folder
- Use the `CMakeSettings.json` panel to specify the following value for the "CMake toolchain file" option: `${projectDir}\vcpkg\scripts\buildsystems\vcpkg.cmake`

The CMake cache will start to build and you will get the following output:

```
1> [CMake] -- The C compiler identification is MSVC 19.28.29336.0
1> [CMake] -- Detecting C compiler ABI info
1> [CMake] -- Detecting C compiler ABI info - done
1> [CMake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
1> [CMake] -- Detecting C compile features
1> [CMake] -- Detecting C compile features - done
1> [CMake] -- Found Git version: refs/heads/master commit 0097048a2670a469b4f6991a73c697f1e6aece74 from 2021-11-14T15:42:53+0100
1> [CMake] -- Using Git version tag: 21.05-137-g0097048a on master at 202111141542
1> [CMake] -- Found OpenSSL: C:/my/path/rtl_433/vcpkg/installed/x64-windows/debug/lib/libcrypto.lib (found version "1.1.1l")  
1> [CMake] -- OpenSSL TLS support will be compiled. Found version 1.1.1l
1> [CMake] -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> [CMake] -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> [CMake] -- Could NOT find LibRTLSDR (missing: LibRTLSDR_LIBRARY LibRTLSDR_INCLUDE_DIR) 
1> [CMake] CMake Error at C:\my\path\rtl_433\CMakeLists.txt:181 (message):
1> [CMake]   RTL-SDR development files not found.
1> [CMake] 
1> [CMake] 
1> [CMake] -- Configuring incomplete, errors occurred!
```

First of all, it appears `PKGCONFIG` should not be required at all to begin with. At least that's what I figured out looking for details about that specific error, `vcpkg` is to be used instead.

The error about `librtlsdr` is because we did not install it from `vcpkg`. But if you try it, you will get an error telling you that "no ports could be found". Sure enough, `librtlsdr` is not in the `vcpkg\ports` folder so one has to actually create a new port which appears to be quite "difficult" to do because it does not seem to have proper documentation. Here is what I have found so far:

https://github.com/microsoft/vcpkg/blob/master/docs/examples/packaging-zipfiles.md
https://github.com/microsoft/vcpkg/pull/9430
https://github.com/microsoft/vcpkg/tree/master/scripts/templates
https://github.com/microsoft/vcpkg/issues/12823
https://github.com/microsoft/vcpkg/discussions/17903

But the first step for a valid port is to identify the source Github repository for a libRTLSDR that builds under windows x64.
The original appears to be here, but seems very old: https://github.com/steve-m/librtlsdr
The, there is the one I used with VC++ here: https://github.com/winterrace/librtlsdr
But the most promising one seems to be this one: https://github.com/librtlsdr/librtlsdr

I would suggest using the later one, especially because it has properly defined tags with prebuilt DLLs.

zuckschwerdt, 2021-11-15:
The canonical "original" lib rtlsdr is https://github.com/osmocom/rtl-sdr/
("librtlsdr" is a improved variant https://github.com/librtlsdr/librtlsdr )
You can get prebuilt libs from https://downloads.myriadrf.org/builds/PothosSDR/

voltagex, 2022-08-07:
@obones how did you get vcpkg to work?

```
c:\git\rtl_433\build>cmake -DCMAKE_TOOLCHAIN_FILE=C:/git/vcpkg/scripts/buildsystems/vcpkg.cmake -G "Visual Studio 17" ..
CMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- Found Git version: refs/heads/master commit eee45de11f600631c3dd793ffaefe77f0d3c0048 from 2022-08-07T11:13:16+0200
-- Using Git version tag: 21.12-133-geee45de1 on master at 202208071113
-- IPv6 support enabled.
-- Threads support will be compiled.
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR)
-- OpenSSL development files not found, TLS won't be possible.
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find LibRTLSDR (missing: LibRTLSDR_LIBRARY LibRTLSDR_INCLUDE_DIR)
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find LibUSB (missing: LibUSB_LIBRARY LibUSB_INCLUDE_DIR)
CMake Error at CMakeLists.txt:223 (message):
  RTL-SDR development files not found.

-- Configuring incomplete, errors occurred!
See also "C:/git/rtl_433/build/CMakeFiles/CMakeOutput.log".
See also "C:/git/rtl_433/build/CMakeFiles/CMakeError.log".

c:\git\rtl_433\build>
```

```
c:\git\vcpkg\vcpkg.exe list
libusb:x64-windows                                 1.0.26           a cross-platform library to access USB devices
openssl:x64-windows                                3.0.5#3          OpenSSL is an open source project that provides ...
pthreads:x64-windows                               3.0.0#11         Meta-package that provides PThreads4W on Windows...
rtlsdr:x64-windows                                 2020-04-16#3     rtl-sdr is a library that turns your Realtek RTL...
vcpkg-cmake-config:x64-windows                     2022-02-06#1
vcpkg-cmake-get-vars:x64-windows                   2022-05-10#1
vcpkg-cmake:x64-windows                            2022-07-18
```

obones, 2022-08-08:
I'm not calling it manually, I'm using it via Visual Studio itself, with the indicated configuration

gdt, 2023-10-02:
Where are we on this?  I'd like to move to something in tree (in some readme), and not have long-term open issues.

voltagex, 2023-10-03:
Not something I've worked on recently, sorry.

obones, 2023-10-03:
I haven't looked at since then as I did not have much time for it, nor was I able to get "debugging" right with VS2019.
I should try again with VS2022

gdt, 2023-10-03:
Do you want to keep this open?  I guess if you manage to create a PR, even if draft and you are not fully happy with it, then we could close this.   There are VS2017 instructions now, and I don't know windows culture enough to know if that's no longer of interest, but I'm guessing that just updating to VSrecent is in order.