merbanan/rtl_433 #1879: const qualifier mismatch warnings

obones, 2021-11-16:
This is a spin off from #1866 as I believe it requires a bit of discussion.

Compiling with VS2019, I get those three warnings:

File|Line|Code|Message
---|---|---|---
data_tag.c | 234 | C4090 | 'function': different 'const' qualifiers
r_api.c | 923 | C4090 | 'function': different 'const' qualifiers
rtl_433.c | 1431 | C4090 | 'function': different 'const' qualifiers

They all come from a line that calls the `free` method to free memory allocated earlier. But the argument given to `free` is always a `char const **` when the method expects a non `const` pointer.

Could it be that the `const` keyword is wrongly placed in the variable/return type declaration? I must admit that I'm always a bit confused by the possible positions of the `const` keyword and their respective meanings.

Any suggestion is most welcome

zuckschwerdt, 2021-11-16:
I follow the handy rule to read types right-to-left, so a pointer to a pointer to const-chars. These instances are arrays of `char const *`. The contents must never be touched or free'd but the array itself is short-lived heap memory.

gdt, 2023-10-02:
Is this still a problem with current code?

obones, 2023-10-03:
This appears to have been fixed by d67ee968e1efb36ca4195a27c597b9f78aebb074

closed 2023-10-03 by obones