merbanan/rtl_433 #1880: Efergy E2 classic 2.0 Current Limit?

brimacd, 2021-11-16:
rtl_433 and my Efergy E2 work well up to approximately 32 amps, then it hangs until the current reduces back down below this limit.
I have saved some of the data to .cu8 files and to .ook files to view on https://triq.org/pdv/ web site, but have not had any luck yet with determining what is happening. My E2 CT transmitts at 10/15/20 seconds though.

I am using an Efergy E2 (2012) model with 2 CT's, and a new RLT2832U & R820T2 based SDR.  Installed rtl_433_win_2019-08-19 on Windows 10 in early October 2021. Woking well with my Oregon Scientific weather sensors.

Any ideas on what is hppening?

zuckschwerdt, 2021-11-17:
The "Efergy e2 classic" is protocol `36`. Start that protocol verbose: ` rtl_433 -R 36:vv` then you should get codes, e.g.
`[00] {65} 0b 34 ea 40 4f 5b 00 13 80`

Let us known the codes below and above 32 A.

brimacd, 2021-11-20:
Sent this to Christian W. Zuckschwerdt <notifications@github.com> on Nov 18th:

[00] {65} 07 c0 92 40 00 00 00 99 80     -response at 0 amps (CT's disconnected) for base line test.

[00] {65} 07 c0 92 40 7a 5a 05 72 80    -response at 30.59 amps

No response at approximately 34 amps though!

I repeated the 30.x and 34 amp tests with the -S all option, to get the cu8 files (g049_433.55M_250k.cu8 and g052_433.55M_250k.cu8
[g049_g052.zip](https://github.com/merbanan/rtl_433/files/7575522/g049_g052.zip)
)

brimacd, 2021-11-20:
Did not mean to close this issue!
Brian

zuckschwerdt, 2021-11-21:
Both signals are looking good and mostly the same (try dropping on https://triq.org/pdv/ )
And both report ok:
```
g049_433.55M_250k.cu8:
time      : @0.176992s
model     : Efergy-e2CT  Transmitter ID: 37568
Battery   : 1            Current   : 30.35 A       Interval  : 6s
Learning  : NO           Integrity : CHECKSUM

g052_433.55M_250k.cu8:
time      : @0.176972s
model     : Efergy-e2CT  Transmitter ID: 37568
Battery   : 1            Current   : 33.35 A       Interval  : 6s
Learning  : NO           Integrity : CHECKSUM
```

zuckschwerdt, 2021-11-21:
When capturing the files, did you get that report on screen? And if you run `rtl_433 thefilename.cu8` do you see as above?

brimacd, 2021-11-21:
Nothing from the g052 file, but the g049 output is as above.

gdt, 2023-10-14:
This seems like a question that has been mostly answered and also timed out.

closed 2023-10-14 by gdt