merbanan/rtl_433 #1885: Implement a scanning mode

merbanan, 2021-11-21:
In this scanning mode iterate over a parameter range and after a time duration output the "best" setting based on some kind of metric.

The initial thought is to just do a 100kHz step of the frequency parameter and then report the amount of received messages.

This would help out to find a reasonable default for users that might work as well as looking a spectrograms etc.

zuckschwerdt, 2021-11-21:
Quick and dirty: maybe just run a frequency hopping then sort through the output.

merbanan, 2021-11-21:
Yes, that is my first thought also. I would just like for it to be extendable. And I would like to have the feature of selecting the "best" parameters after the scanning is done also. And a delayed start of the scanning to warm up the components etc.

zuckschwerdt, 2021-11-21:
This might be a good case for a separate binary. We could polish the "API" along the way.