merbanan/rtl_433 #1892: Little problems with a new Bresser 5-in-1 with coloured display

DL4AAS, 2021-11-30:
Hello,
my project(*) is to build a FSK transmitter that simulates a Bresser 5-in-1 sensor for the display. rtl_433 decodes the original sensor as Bresser 6-in-1 and the source code of rtl_433 was very helpful to construct the packet for the display.

The packet with temperature/humidity was not working for the display, because it needs a 0xf0 in the last byte before checksum (instead of 0x01 for rain). Found by comparing the data of the two packet types...

And the battery bit was not working. It is not bit 3 in byte 7, but bit 1 in byte 14 (low nibble that is not used by temperature). Found by simulating a low bat.

Perhaps someone can use this information.

Jens

(*) the complete project uses a new hardware in the sensor to transmit the data as a LoraWAN packet. And the new FSK transmitter gets it's data from TheThingsNetwork using MQTT. FSK transmitter is the Lora-Chip SX1276.

zuckschwerdt, 2021-12-01:
Thanks for the battery bit info! It's an odd place, I need to look up the original thread, I guess nobody really checked properly for battery low bit.

gdt, 2023-09-26:
It seems someone(tm) needs to read the code with the details in this PR, and decide to change the code or not, and close the PR>

gdt, 2023-10-15:
@DL4AAS Are you still working on this?   It would be helpful if you could analyze the current driver and see what is missing and submit a PR to improve it.