merbanan/rtl_433 #1893: Help Decoding Chinese TPMS for Truck

m4ri01, 2021-12-01:
Hi mate, i came from indonesia and from last week to this week i tried to decode the chinese TPMS for truck like this  
https://id.aliexpress.com/item/1005001763517202.html?spm=a2g0o.productlist.0.0.729d44c8PTjvOC&algo_pvid=e0b83ee3-71e8-49d2-92ec-fe1c00e9db5b&algo_exp_id=e0b83ee3-71e8-49d2-92ec-fe1c00e9db5b-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000021201006904%22%7D  

I already have .cu8. i attach the .cu8  in the file below  
and this is the spectogram of my .cu8 file  
[image]  

thank you, mate.  
PS: i don't have any idea to continue from .cu8, because i still stuck at this point.  
[cu8 signal.zip](https://github.com/merbanan/rtl_433/files/7631972/cu8.signal.zip)

zuckschwerdt, 2021-12-01:
Try `rtl_433 -A yourfile.cu8`
There will be a triq.org/pdv link to view the data. But switch "Slicer PWM" to "MC".

It looks like the usual TPMS: 50 µs (half-bit) Manchester code.
Use `rtl_433 -R 0 -X 'n=mytpms,m=FSK_MC_ZEROBIT,s=52,l=52,r=150'` to get the raw data.
Now try different pressures and temperatures and write a list of codes like this:
```
{197}7ffffffffffffffffffffffffffffa0581b2810230001e9bf8 [23.0 kPa 33 C]
{197}7ffffffffffffffffffffffffffffac594ae140530001cc2f8 [42.0 kPa 99 C]
```

m4ri01, 2021-12-01:
this the output from  `rtl_433 -A g013_433.92M_250k.cu8 `  
  
  

```sh
Test mode active. Reading samples from file: g013_433.92M_250k.cu8
Detected FSK package	@0.152388s
Analyzing pulses...
Total count:  178,  width: 20.86 ms		( 5215 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:  153,  width:   48 us [40;68]	(  12 S)
 [ 2] count:    4,  width:   60 us [60;64]	(  15 S)
 [ 3] count:   19,  width:   96 us [92;104]	(  24 S)
 [ 4] count:    1,  width:  468 us [468;468]	( 117 S)
Gap width distribution:
 [ 0] count:  157,  width:   52 us [44;64]	(  13 S)
 [ 1] count:    2,  width:   40 us [40;40]	(  10 S)
 [ 2] count:   17,  width:  108 us [104;112]	(  27 S)
 [ 3] count:    1,  width:  212 us [212;212]	(  53 S)
Pulse period distribution:
 [ 0] count:    1,  width:   52 us [52;52]	(  13 S)
 [ 1] count:  151,  width:  104 us [92;116]	(  26 S)
 [ 2] count:   13,  width:  152 us [148;164]	(  38 S)
 [ 3] count:   11,  width:  204 us [204;212]	(  51 S)
 [ 4] count:    1,  width:  316 us [316;316]	(  79 S)
Level estimates [high, low]:  15957,     29
RSSI: -0.1 dB SNR: 27.4 dB Noise: -27.5 dB
Frequency offsets [F1, F2]:    1181,  -9282	(+4.5 kHz, -35.4 kHz)
Guessing modulation: No clue...

Detected FSK package	@0.175736s
Analyzing pulses...
Total count:  179,  width: 20.86 ms		( 5216 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:  154,  width:   48 us [40;64]	(  12 S)
 [ 2] count:    4,  width:   60 us [60;60]	(  15 S)
 [ 3] count:   19,  width:  100 us [92;108]	(  25 S)
 [ 4] count:    1,  width:  472 us [472;472]	( 118 S)
Gap width distribution:
 [ 0] count:  160,  width:   52 us [44;64]	(  13 S)
 [ 1] count:   18,  width:  104 us [100;112]	(  26 S)
Pulse period distribution:
 [ 0] count:    1,  width:   56 us [56;56]	(  14 S)
 [ 1] count:  151,  width:  104 us [92;116]	(  26 S)
 [ 2] count:   15,  width:  156 us [152;164]	(  39 S)
 [ 3] count:   11,  width:  204 us [204;212]	(  51 S)
Level estimates [high, low]:  15981,     37
RSSI: -0.1 dB SNR: 26.4 dB Noise: -26.5 dB
Frequency offsets [F1, F2]:    4705,  -8631	(+17.9 kHz, -32.9 kHz)
Guessing modulation: No clue...
```

m4ri01, 2021-12-01:
and this is the output from  
`rtl_433 -R 0 -X 'n=mytpms,m=FSK_MC_ZEROBIT,s=52,l=52,r=150'`  
  
  
```sh
ime      : 2021-12-01 15:59:53
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 4
codes     : {2}4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:01
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 4
codes     : {2}4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:06
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:11
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 4            data      : 2
codes     : {4}2
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:18
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:18
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 4
codes     : {2}4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:22
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:27
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 4
codes     : {2}4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:28
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 197          data      : 7ffffffffffffffffffffffffffffa0581b2810230001d98f8
codes     : {197}7ffffffffffffffffffffffffffffa0581b2810230001d98f8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:28
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 197          data      : 7ffffffffffffffffffffffffffffa0581b2810230001d98f8
codes     : {197}7ffffffffffffffffffffffffffffa0581b2810230001d98f8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:38
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:41
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 4
codes     : {2}4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:46
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:00:47
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 0
codes     : {1}0
[... cut ...]
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-01 16:01:00
model     : mytpms       count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 4
codes     : {2}4

```

zuckschwerdt, 2021-12-01:
Yes, make a list of codes and the value you expect, then we can figure out the encoding.

Add `,bits>=100` to the `-X` to get rid of the short codes.

Is there something like "05809ee2" printed on the sensors? That would be a good start. I don't think wheel order is in the data.

Use some cheap "pressure sprayer" to pressurize the sensors if needed. Like e.g. https://www.amazon.com/dp/B07TDS7GLX/ (check that the sensor fits in the opening!)

m4ri01, 2021-12-01:
i got this data 

```sh
7ffffffffffffffffffffffffffffac594ae140530001bc5f8 = 0.0 27
7ffffffffffffffffffffffffffffa05809ee20130001bd3f8 = 0.0 27
7ffffffffffffffffffffffffffffac4b5a5710330001b8df8 = 0.0 27
7ffffffffffffffffffffffffffffa076cd3050430001b92f8 = 0.0 27
7ffffffffffffffffffffffffffffa0581b2810230001c99f8 = 0.0 28
7ffffffffffffffffffffffffffffac594ae140530001bc5f8 = 0.0 27
7ffffffffffffffffffffffffffffa05809ee20130001bd3f8 = 0.0 27
7ffffffffffffffffffffffffffffac4b5a5710330001b8df8 = 0.0 27
```

the first data after '=' are the pressure and the second data is temperature.

zuckschwerdt, 2021-12-01:
Are that 5 different sensors? See [this BitBench](https://triq.net/bitbench#c=7ffffffffffffffffffffffffffffac594ae140530001bc5f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa05809ee20130001bd3f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffac4b5a5710330001b8df8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa076cd3050430001b92f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa0581b2810230001c99f8%20%5B0.0%2028%5D&f=8h8h%208h8h8h8h%20FLAGS%3F8h8h%20PRES%3F8h%20TEMP%3F8d%20CRC%3F8h%20TRAILER%3F8h%20&a=Sync&m=fffa&cw=4).

m4ri01, 2021-12-02:
yes, that are 5 different sensors, I saw your bitbench and it can be used to decode the TPMS data. anyway thank you for your help

zuckschwerdt, 2021-12-02:
The checksum is an XOR over the 64 bit before the checksum. BitBench link updated.
If you can find the value for the pressure (kPa?) and what the flags mean (init, spinning, pressure loss...) then we can add a decoder.

m4ri01, 2021-12-02:
for the pressure in Bar, I need to convert from hex to decimal and divide it by 100.  
for the flag, the first hex is wheel order example 0530. the value 05 is the 6th wheel order if I see from the tpms LCD. and the other one like 0310 is the 4th wheel order. count in 0 

but the second hex value, i still don't get it.  
[updated bitbench](https://triq.net/bitbench#c=7ffffffffffffffffffffffffffffac594ae140530001bc5f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa05809ee20130001bd3f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffac4b5a5710330001b8df8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa076cd3050430001b92f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa0581b2810230001c99f8%20%5B0.0%2028%5D&f=ID%3F8h8h8h8h%20FLAGS%3F8h8h%20PRES%3F8d%20TEMP%3A8d%20CHK%3A8h%20TRAILER%3F8h%20&a=Preamble&m=fffa&cf=XOR&cl=72&cw=8)

zuckschwerdt, 2021-12-02:
If you get bar by div 100, then the value is kPa. One hex byte only allows 255 kPa (2.55 bar) which seems not enough. Perhaps the hex `0530??` is not `??` kPa but `0??` kPa. You need to test with >2.6 bar for that.
The `3` could be "all okay". It would change with low battery and with rapid pressure loss. Those are the common flags for TPMS.

m4ri01, 2021-12-02:
but I already test with some values, I get decimal value 216 then my monitor says 2.2. and then I got 082 value from tpms, then my monitor show 0.8

zuckschwerdt, 2021-12-02:
The maximum you can get is `255`. This will not be enough for car tires, right? A common regular pressure is already 2.5 bar (=250).
Is there a spec on the device (e.g. Temperature -20 to +70, Pressure up to 450 kPa)?

m4ri01, 2021-12-02:
the specification from the devices are like this :  
Monitoring temperature range: -40 ℃ 130 ℃  
Monitoring air pressure range: 0.1bar 12.0bar

merbanan, 2021-12-02:
When the registered pressure is above 2.5 bar you will notice that another byte will start to increment also.

zuckschwerdt, 2021-12-02:
12 bar == 1200 kPa, at least 11 bit, so it's likely 12 bit (3 nibbles), like suggested above ;)
For "temperature range: -40 ℃ 130 ℃" the 8 bit (byte) value is perhaps signed (sign+7bit), that would give -128 to +127, which is close enough.

m4ri01, 2021-12-03:
oh i see, yes you are right @zuckschwerdt , this is the new [bitbench](https://triq.net/bitbench#c=7ffffffffffffffffffffffffffffac594ae140530001bc5f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa05809ee20130001bd3f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffac4b5a5710330001b8df8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa076cd3050430001b92f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa0581b2810230001c99f8%20%5B0.0%2028%5D&c=7ffffffffffffffffffffffffffffac594ae140530001bc5f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa05809ee20130001bd3f8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffac4b5a5710330001b8df8%20%5B0.0%2027%5D&c=7ffffffffffffffffffffffffffffa05809ee20130d81f0ff8%20%5B2.2%2031%5D&c=7ffffffffffffffffffffffffffffa05809ee20130d82030f8%20%5B2.2%2032%5D&c=7ffffffffffffffffffffffffffffac5937a290030001f2af8%20%5B0.0%2032%5D&c=7ffffffffffffffffffffffffffffac4b5a5710331481ec1f8%20%5B3.3%2030%5D&c=7ffffffffffffffffffffffffffffac5937a2900305a1f70f8%20%5B0.9%2031%5D&c=7ffffffffffffffffffffffffffffa05809ee20130d82030f8%20%5B2.2%2032%5D&c=7ffffffffffffffffffffffffffffac594ae140530001ec0f8%20%5B0.0%2030%5D&c=7ffffffffffffffffffffffffffffac594ae140530701fb1f8%20%5B1.1%2031%5D&c=7ffffffffffffffffffffffffffffa05809ee20130d81f0ff8%20%5B2.2%2031%5D&c=7ffffffffffffffffffffffffffffac5937a2900305a1f70f8%20%5B0.9%2031%5D&c=7ffffffffffffffffffffffffffffac5937a290030541f7ef8%20%5B0.8%2031%5D&f=ID%3F8h8h8h8h%20FLAGS%3F8h4h%20PRES%3F12d%20TEMP%3A8d%20CHK%3A8h%20TRAILER%3F8h%20&a=Preamble&m=fffa&cf=XOR&cl=72&cw=4)  from the last data that i got last night. The pressure was 3 nibbles then convert to decimal.

zuckschwerdt, 2021-12-03:
Very good! A format of `ID:32h POS:8h FLAGS?4h KPA:12d TEMP:8d CHK:8h TRAILER?8h` seems to cover it.
(No good way to test the battery low flag, except opening/destroying a sensor...)

Now we need to find a name (is there any name written on the plastic?) and I'll add this decoder to rtl_433.

m4ri01, 2021-12-03:
nice! but I'm not sure about the name of this TPMS. Because from the box I can't get any information of this TPMS. it only says "TPMS" without any other names 😢. like this one  
[image]
 
I bought these tpms from this [link](https://www.aliexpress.com/item/1005002399356761.html?spm=a2g0o.productlist.0.0.69fd16d30i1WGR&algo_pvid=71192462-0cc1-4a5d-8a95-6f3b17ca4cee&algo_exp_id=71192462-0cc1-4a5d-8a95-6f3b17ca4cee-13&pdp_ext_f=%7B%22sku_id%22%3A%2212000025029129963%22%7D) and I also didn't get any information about the name :)

zuckschwerdt, 2021-12-03:
The decoder has been added:
```
model     : Truck        type      : TPMS          id        : c594ae14
wheel     : 5            Pressure  : 0 kPa         Temperature: 28 C
State?    : a            Flags?    : 3             Integrity : CHECKSUM
```

If you can grab a code with negative temperature please let me know!

closed 2021-12-03 by zuckschwerdt

m4ri01, 2021-12-03:
thanks @zuckschwerdt! oke if I got the negative temperature I will send the data to this issue.

commit 2f77d390 (2021-12-03): Add support for Truck TPMS (closes #1893)