merbanan/rtl_433 #1894: Bresser 5 in 1 not detected

grinco, 2021-12-03:
I got a Bresser 5 in 1 weather station (model 7002511) which doesn't work with existing decoders. There was a similar report [here](https://github.com/merbanan/rtl_433/issues/1126#issuecomment-751372083) however I haven't seen a resolution. Symbol rate is set to 250K, tried both the new and classic FSK pulse detector mode. Please let me know if there is anything I can capture to get it supported.

zuckschwerdt, 2021-12-03:
Did you verify a good reception of the signal? https://triq.org/rtl_433/ANALYZE.html has some basic info.
If you have a sample that should be working but doesn't decode, please upload here as zip and we'll take a look.

grinco, 2021-12-04:
The transmitter is just 2 meters away from the receiver. Also the main unit is able to read the signal.
I've captured a few samples ([zip in google drive](https://drive.google.com/file/d/1_p2nVcaBZetdSL0yBcYyvT1vUkSbXLJK/view?usp=sharing)). There seems to be some other equipment using this frequency - not sure how to filter out irrelevant data. I've created a separate folder for every reading that I was monitoring, and documented in readings.txt the displayed values on the main unit. Please let me know if there is anything else I need to provide.

zuckschwerdt, 2021-12-04:
There is serious interference present. Looks like pulses at 868.30M with a regular 10ms interval. That won't work well.
Is that a signal from your own equipment? Or it could be LoRa? Your device seems to be at 868.33M, perhaps tuning that to the edge could help, e.g. `-f 868.44M`.

Or you could try to set a fixed gain which excludes the interference but captures the wanted source. At 2m distance you could also try to remove the antenna, could work.

This would be a great example for needing a custom [band-stop filter](https://en.wikipedia.org/wiki/Band-stop_filter). I.e. an option to selectivly block 868.3M would be neat. Not terribly difficult to implement, but we don't have that currently.
Maybe a simple but steep (=cpu intensive) high-pass to block DC. I.e. then tune to 868.3M, enable the high-pass and receive everything but 868.3M.

grinco, 2021-12-04:
Thanks for all the help. I'm living in a block of flats and people have all sorts of stuff - including car keys, remote controls, and probably lora hardware. There was a base station within 1Km range so the interference can be coming from it.... The following settings seem to work:
Sample rate: 250000 S/s.
Tuner gain: Auto.
Tuned to 868.425MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us

Not all signals are decoded correctly, but I'll keep playing with frequency selection and gain control - at least I know where to dig.

closed 2021-12-04 by grinco