merbanan/rtl_433 #1898: ESIC EMT7110 power meter (used by display EMR 7370) not working

petterreinholdtsen, 2021-12-05:
Something seem to have gone wrong with the esic_emt7110 driver (see #1132).  When I have it connected and close to my reciever which do receive the power data, rtl_433 report no power information  What can I do to debug this?

I tried dropping 6b09d633200de9d7ccdd5904d9eb6df03a45148d, but it did not help.

I notice #1351 mention the messages are quite close to the ones for Fine Offset WH65B.  I do not have any of those neary, but I do have several reported as the model Fineoffset-WH2.

merbanan, 2021-12-05:
What does the following command give?

rtl_433 -s1024k -X 'n=EMT7110,m=FSK_PCM,s=104,l=104,r=10700'

petterreinholdtsen, 2021-12-05:
[Benjamin Larsson]
> What does the following command give?
>
> rtl_433 -s1024k -X 'n=EMT7110,m=FSK_PCM,s=104,l=104,r=10700'

A test run gave me this:

$ ./src/rtl_433 -s1024k -X 'n=EMT7110,m=FSK_PCM,s=104,l=104,r=10700'
rtl_433 version 21.05-154-g7fc96b45 branch master at 202112051124 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pere/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 171 out of 201 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1024000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 102400 Hz, 9.8 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:39
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 789          data      : 796abdd9f7ee974577aa4d6e3beec9b6b6f66bd7ffdb25cbeab74f7f7b31cef0bb933535efaaaebf0ddabf6f7e9fdc954c74f5fd9381da4dbfa05ff76f03c38ec91acd854513c6bf4ae54b915588581a7b659513c96d7cd2c920143c06f337b71ada38
codes     : {789}796abdd9f7ee974577aa4d6e3beec9b6b6f66bd7ffdb25cbeab74f7f7b31cef0bb933535efaaaebf0ddabf6f7e9fdc954c74f5fd9381da4dbfa05ff76f03c38ec91acd854513c6bf4ae54b915588581a7b659513c96d7cd2c920143c06f337b71ada38
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:40
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 841          data      : b4a690cfe61d795475c33a5bfc6913ba99a57fbb8b45f6f87fb76bd9ad4b92c1f2bec564c3b3b05084d1610add0bfe89f5e2bb355f9766c773b5fb7a9dffa6adf3af76ea92aeaef1f43b81aa35f5d7be1e5ee5ffc9167389af6eddab592ddc9fdbe7abaf9dafdee7d78
codes     : {841}b4a690cfe61d795475c33a5bfc6913ba99a57fbb8b45f6f87fb76bd9ad4b92c1f2bec564c3b3b05084d1610add0bfe89f5e2bb355f9766c773b5fb7a9dffa6adf3af76ea92aeaef1f43b81aa35f5d7be1e5ee5ffc9167389af6eddab592ddc9fdbe7abaf9dafdee7d78
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:40
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 48           data      : 80546c05501b
codes     : {48}80546c05501b
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:43
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 44           data      : 460708b001e
codes     : {44}460708b001e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:43
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 37           data      : 64fb8cf060
codes     : {37}64fb8cf060
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:43
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 48           data      : 562c9b9d87b0
codes     : {48}562c9b9d87b0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:43
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 13           data      : 1f50
codes     : {13}1f50
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:43
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 25           data      : fc458c8
[... 68 lines cut ...]
time      : 2021-12-05 19:46:45
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 41           data      : 4aa85021d68
codes     : {41}4aa85021d68
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:45
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 46           data      : f02084f6d200
codes     : {46}f02084f6d200
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-05 19:46:45
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 23           data      : f4c502
codes     : {23}f4c502

C^CSignal caught, exiting!
Reattached kernel driver
%

-- 
Happy hacking
Petter Reinholdtsen

gdt, 2023-10-14:
@petterreinholdtsen Can you retest with current git master?

petterreinholdtsen, 2023-10-14:
[Greg Troxel]
> @petterreinholdtsen Can you retest with current git master?

I rebuild git master and ran "src/rtl_433 -s1024k -X
'n=EMT7110,m=FSK_PCM,s=104,l=104,r=10700'", but got no information from
the power meter.  The display on the reciver was updated, so I know the
radio messages were transmitted.

-- 
Happy hacking
Petter Reinholdtsen

gdt, 2023-10-14:
Please read all the files in docs/ and you can begin trying to figure it out like it's a new decoder, among other hints.

petterreinholdtsen, 2023-10-14:
[Greg Troxel]
> Please read all the files in docs/ and you can begin trying to figure
> it out like it's a new decoder, among other hints.

I assumed something relevant had changed to make such test useful,
perhaps someone tried to implement it and wanted me to verify it was
working.  The comment above give the impression that there is nothing
new to test and a new decode need to be written.  I suspect I
misunderstood something.  Why did you ask me to test the latest master?

-- 
Happy hacking
Petter Reinholdtsen

gdt, 2023-10-14:
Because reports of anything against old code are not really useful; a lot has changed.  I really don't know what has changed and am going through the entire bug backlog to try to clean out issues that are not useful or actionable and freshen the ones that are possibly useful.  In general, there is no group working on requests, and it's more trying to help people help themselves..

gdt, 2024-08-04:
Status?   @petterreinholdtsen If you want to try to bisect to see if older code works, or start debugging anew, that seems useful, but as It is I don't see how anybody can do anything useful with this issue, so I'll close it in a week if you don't comment.  I don't mean to demand that you do something substantive in a week, but if you aren't going to dig in, I don't see that it helps the project to have this open, and we have too many open issues.

petterreinholdtsen, 2024-08-04:
[Greg Troxel]
> Status?   @petterreinholdtsen If you want to try to bisect to see if
> older code works, or start debugging anew, that seems useful, but as
> It is I don't see how anybody can do anything useful with [...]

The status is that I do not know what to do.  I would be happy to run
more tests, provide debug output or whatever I can do with my current
skill set and knowledge.  I do not know when 'older code work'.  I still
got the power meter and SDR units.

-- 
Happy hacking
Petter Reinholdtsen

gdt, 2024-08-04:
I am afraid that in order to make progress you will have to learn enough to debug this.  It is very likely nobody else has both the hardware and the motivation.

What I meant is that first you need to be able to build rtl_433 from git sources.    Then, you can build older versions from git and see if an of them decode your device.  Or you can debug from scratch by changing the code and adding prints or whatever.   You said "gone wrong with" but you were unclear about the last working state.

If you need help with building or git please ask on the list or open a discussion.

zuckschwerdt, 2024-08-04:
The data you got (those two very long codes) are not what we expect. It looks like random noise. You should see a preamble of `aaaa` (or `5555` if misaligned) and then a sync word of `2dd4` (might need to be bit-shifted).
S.a. the description here https://github.com/merbanan/rtl_433/blob/master/src/devices/esic_emt7110.c
Maybe grab some samples from the sender to analyze further. https://triq.org/rtl_433/ANALYZE.html