merbanan/rtl_433 #1899: Rtl_433 usb_open error -4

FinestDice17, 2021-12-06:
root@f46e57ff9091:/sdr/rtl_433# rtl_433
rtl_433 version 21.05-154-g7fc96b45 branch master at 202112051124 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 170 out of 201 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201 ]
usb_open error -4

zuckschwerdt, 2021-12-06:
> usb_open error -4

LIBUSB_ERROR_NO_DEVICE
No such device (it may have been disconnected).

Try `lsusb -v` and then check access rights.

closed 2021-12-06 by zuckschwerdt

FinestDice17, 2021-12-06:
This is what I get when I run lsusb -v , im noticing though its saying that on all usb devices though.

Bus 001 Device 002: ID 0bda:2838 Realtek RTL2838UHIDIR
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda 
  idProduct          0x2838 
  bcdDevice            1.00
  iManufacturer           1 Realtek
  iProduct                2 RTL2838UHIDIR
  iSerial                 3 00000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 
      bInterfaceSubClass    255 
      bInterfaceProtocol    255 
      iInterface              5 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 
      bInterfaceSubClass    255 
      bInterfaceProtocol    255 
      iInterface              5

zuckschwerdt, 2021-12-06:
> Couldn't open device, some information will be missing

Access permission problems? Run with `sudo`, compare. Check that USB is available to the user you want to run apps with.

FinestDice17, 2021-12-06:
Based on what I've been reading, I have Debian running in a container on my NAS and even though it can see the device, its not able to fully utilize the device because I need to allow the container to use host devices. I think thats my issue but not sure on how to get that working.

zuckschwerdt, 2021-12-06:
Or the host/container has drivers loaded that block access, search for "rtl-sdr blacklist dvb"