merbanan/rtl_433 #1900: dsc: WS4975 door contact not decoding open events

roger-, 2021-12-07:
My WS4975W gets closed events decodes correctly (x4):

[image]

But open events don't get decoded at all. Here are two recordings of the unknown signal (might be other signals mixed in):

[g002_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/7670283/g002_433.92M_250k.zip)
[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/7670284/g001_433.92M_250k.zip)

Tried shifting the frequency and raising the sample rate, but still not decoding.

Appreciate any help!

zuckschwerdt, 2021-12-07:
The signal doesn't look particularly bad. But the pulse detector is missing a few bits and that stops the decoding.
It will need some experimentation to see if we can improve something.

GhettoDubs, 2022-04-04:
Second this.  I have the same issue with two different types of DSC door sensors.   Works for the heartbeat and close events, but its not picking up open.  @roger- did you find a fix for this?

roger-, 2022-04-04:
@GhettoDubs not yet unfortunately.

GhettoDubs, 2022-04-20:
i was wondering if this could be due to reset_limit=6000 us.

In your sample you have low signal for 6888us.  Could this be causing it to fall over?

Not a programmer so hard for me to make sense of this!

GhettoDubs, 2022-04-20:
yes, i think thats it.

when i use code below I can consistently get correct signal decoding:
`decoder n=door sensor,m=OOK_RZ,s=536,l=1072,r=7000`

GhettoDubs, 2022-04-20:
@zuckschwerdt is it possible to creat pull request for this?  sorry not sure how to work github

I think the change is:
`r_device dsc_security_ws4945 = {
        .name        = "DSC Security Contact (WS4945)",
        .modulation  = OOK_PULSE_RZ,
        .short_width = 536,  // Pulse length, 536 µs
        .long_width  = 1072, // Bit period, 1072 µs
        .reset_limit = 6000, // Max gap,
        .decode_fn   = &dsc_callback,
        .fields      = output_fields,
};`

to

`r_device dsc_security_ws4945 = {
        .name        = "DSC Security Contact (WS4945)",
        .modulation  = OOK_PULSE_RZ,
        .short_width = 536,  // Pulse length, 536 µs
        .long_width  = 1072, // Bit period, 1072 µs
        .reset_limit = 7000, // Max gap,
        .decode_fn   = &dsc_callback,
        .fields      = output_fields,
};`

zuckschwerdt, 2022-04-20:
@GhettoDubs very good. Yes that solves the problem. There is a 6800µs gap near the start. According to the docs it should not be there though? Needs some explanation/ documentation for that observed new behaviour.

>There is a sync/start 1 bit in between every 8 bits. A zero byte would be 8 x 500 uS of no signal (plus the 250 uS of  silence for the first half of the next 1 bit) for a maximum total  of 4,250 uS (4.25 mS) of silence."

zuckschwerdt, 2022-04-20:
Oh wait, I get it now. the docs are for the 250/500µs original DSC. the WS4945 has 500/1000µs length. So the maximum expected (worst case) gap would be 8 x 1000 µs + 500 µs = 8500 µs, right?

GhettoDubs, 2022-04-20:
Yes thats right. Unfortunate number of consecutive zero bits in this case.

closed 2022-04-20 by zuckschwerdt