merbanan/rtl_433 #1901: Support for Byron DBY-23510 doorbell

alanbowman, 2021-12-07:
https://www.chbyron.eu/byron-dby-23510-wireless-bell-push-button-dby510-dby--23510

Currently detected as Smoke-GS558 when the button is pushed, but with varying data in the fields.  I've captured the signal, 
[g010_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/7671035/g010_433.92M_250k.cu8.zip), but am honestly not sure how to get from that to something more meaningful.  Any pointers how to get useful values to put into a decoder would be appreciated.

zuckschwerdt, 2021-12-07:
If it's detected as Smoke-GS558 then it's very likely a EV1527 encoder. Just 24 fixed pulses and a trailing sync.
See the examples folder for some decoders to adapt. Basically tune the timing to fit and filter (`match`) by your ID.

alanbowman, 2021-12-07:
I've managed to find timings that give me 25 repeated PWM-based words of 25 bits each at the start of each transmission.  This repeated word seems to vary with each push of the bell.  Then there's some trailing bits that seem to use different timings - does that sound like the trailing sync you referred to?

zuckschwerdt, 2021-12-08:
Oh, I meant **[conf examples](https://github.com/merbanan/rtl_433/tree/master/conf)** (not examples folder), sorry. E.g. the [Silverline doorbell](https://github.com/merbanan/rtl_433/blob/master/conf/silverline_doorbell.conf) has some notes.

I see what you mean. There is acutally two different encondings/timings in the transmission. Drop your .cu8 on https://triq.org/pdv/ -- you'll notice the last few packets are slower/longer.

Inspect [the RfRaw link](https://triq.org/pdv/#AAB027062A014403CC01FC05F00CA41BB48181909081819090908181909090908190818190908190819455+AAB0260601014403CC01FC05F00CA41BB481819090818190909081819090909081908181909081908355+AAB00F0601014403CC01FC05F00CA41BB4A555+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB0110601014403CC01FC05F00CA41BB4B2B2A355+AAB00F0602014403CC01FC05F00CA41BB4A355+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB0130601014403CC01FC05F00CA41BB4B2B2B2B2A355+AAB00F0601014403CC01FC05F00CA41BB4A355+AAB0110601014403CC01FC05F00CA41BB4B2B2A355+AAB00F0603014403CC01FC05F00CA41BB4A355+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB0120601014403CC01FC05F00CA41BB4B2B2B2A355+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB00F0601014403CC01FC05F00CA41BB4A355+AAB0110601014403CC01FC05F00CA41BB4B2B2A555+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB0110601014403CC01FC05F00CA41BB4B2B2A355+AAB00F0602014403CC01FC05F00CA41BB4A355+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB0130601014403CC01FC05F00CA41BB4B2B2B2B2A355+AAB00F0601014403CC01FC05F00CA41BB4A355+AAB0110601014403CC01FC05F00CA41BB4B2B2A355+AAB00F0603014403CC01FC05F00CA41BB4A355+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB0120601014403CC01FC05F00CA41BB4B2B2B2A355+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB00F0601014403CC01FC05F00CA41BB4A355+AAB0110601014403CC01FC05F00CA41BB4B2B2A555+AAB0100601014403CC01FC05F00CA41BB4B2A355+AAB0110601014403CC01FC05F00CA41BB4B2B2A355+AAB00F0601014403CC01FC05F00CA41BB4A355) from `-A` output, "PWM, 320, 970" will work for the first part, then "PWM, 500, 1500" for the second.
Since PWM only needs to decide between short and long, a careful setting will decode both timings.

zuckschwerdt, 2021-12-08:
E.g. `rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=900,g=2000,r=5000' g010_433.92M_250k.cu8`
gives 43 codes of `{25}cc61650` which is actually `cc6165` ("bit" 25 is sync),
and 3 of the slower codes as `{33}4e867a2c8` which is compatibility for another receiver I guess.

alanbowman, 2021-12-08:
OK - I understand.  Thanks for recommending those timings, I can see where the bits come from now.

Looking across several presses of the doorbell, it looks like the values change each time (for both of the timing sequences) given no other differences.  Also, I know that it supports 16(?) different chimes, but it's tricky to switch between them.  I'll try to see if I can figure out the differences into something useful though.

zuckschwerdt, 2021-12-08:
The codes are consistent in one transmission, but different on each press? How would the receiver pick up on that? It might be some new Rolling Code scheme, interesting!

Can you post the codes from different presses? One line of 25 + 33 code per press, check that the codes are valid, i.e. match their repeats.

alanbowman, 2021-12-08:
I just rang the doorbell three times, roughly 5 seconds apart.  The first one had the same code as previously posted 25x`{25}cc61650` then several `{33}4e867a2c8`.  The second ring gives 25x`{25}ecacb10` then `{33}0c792a6c8`.  The third ring gives 25x`{25}e55cb90` then `{33}4e867a2c8`.

As I write this, it looks like the {25} bit blocks vary more than the {33} (and some pulses that follow).  I wonder if it's the later section that is more useful to identify the doorbell.  I'll see if I can make more sense of the scheme.
[captures.zip](https://github.com/merbanan/rtl_433/files/7675691/captures.zip)

zuckschwerdt, 2021-12-08:
Strange, can't really say what that is. If you put the [codes in a BitBench](https://triq.net/bitbench#c=%7B25%7Dcc61650%20%7B33%7D4e867a2c8&c=%7B25%7Decacb10%20%7B33%7D0c792a6c8&c=%7B25%7De55cb90%20%7B33%7D4e867a2c8&f=25b%20%7C%2033b) you can maybe find a pattern.

gdt, 2023-10-05:
What's the status?  Anyone working on this?

ryancdotorg, 2025-07-28:
This seems to find it for me:

```
rtl_433 -f 433.92M -s 250k -X 'name=Byron_DBY-23510,modulation=OOK_PWM,short=500,long=1500,reset=7000,gap=1500,tolerance=300,sync=0,bits=33'
```

However, the doorbell has two different bit sequences used for alternate presses. I have two of these doorbells with the same behavior...

zuckschwerdt, 2025-07-28:
Thanks for the feedback. Closing this issues as the key points about this device are known now.
PR a conf file with some notes and a link to this issue if you like!

closed 2025-07-28 by zuckschwerdt