merbanan/rtl_433 #1902: What's the expected release schedule for new tags?

deviantintegral, 2021-12-08:
I've been maintaining an [addon for Home Assistant](https://github.com/pbkhrv/rtl_433-hass-addons/issues/31) to integrate rtl_433. It looks like there's interest in recently added protocols, as seen at https://github.com/pbkhrv/rtl_433-hass-addons/issues/31 and https://github.com/pbkhrv/rtl_433-hass-addons/issues/33.

Currently, we're building the addon against the `21.05` tag. That's helpful for us to support since it means that all addon users are using the same rtl_433 release. Looking at [the CHANGELOG](https://triq.org/rtl_433/CHANGELOG.html), I see releases are not really on a fixed schedule, but averaging every 6-8 months.

As an upstream maintainer, would you prefer for us to stick to tags, or would you rather we track master? If tags, is there anything we can do to help get releases on a more consistent 3 or 6 months schedule?

merbanan, 2021-12-09:
I have nothing to add before a release.

sandstormkeshav, 2022-05-24:
Just wanted to add on to this, there have been quite a few pushes in master since the last release, can we expect a new release to come sometime soon? Thank you!

gdt, 2023-09-30:
There has been release since this issue....

closed 2023-09-30 by gdt