merbanan/rtl_433 #1905: Proflame 2 support

robloh, 2021-12-13:
Spec: https://github.com/johnellinwood/smartfire

Tried to convert the information on that page in to something I can try to decode with rtl_433, but unfortunately this is over my head. I was unable to match the RfCat modulation (MOD_ASK_OOK) to one of the modulation_types in this project, same with how baud rate relates to rtl_433 parameters:

"The command bursts are transmitted at 314,973 KHz using the On-Off Keying (OOK) variant of Amplitude Shift Keying (ASK) which is a type of Amplitude Modulation (AM), and are sent at a transmission rate of 2400 baud. Inside the command burst are the command packets. The same command packet is transmitted 5 times in each command burst, and the packet repetitions are separated by 12 low amplitude bits (zeros).

The command packets are encoded with a variant of Thomas Manchester encoding. In this variant, 0 is represented by 01, a 1 by 10, zero padding (Z) by 00, and synchronization words (S) as 11. The encoded command packet is 182 bits, and the decoded packet is 91 bits. If each data part of the packet were manchester encoded separately, and then separated with the bit pattern 11, then the encoding of individual command packets could be considered standard Thomas Manchester encoding."

https://github.com/merbanan/rtl_433/files/7700285/proflame2_samples.zip

zuckschwerdt, 2021-12-13:
Interesting.

- OOK alone is not enough information, you need to know that Manchester is a special PCM, thus `OOK_PCM` ;)
- 2400 Baud is 2400 symbols per second. Reciprocal is seconds per symbol, i.e. µs per symbol is `1000000/2400 = 417`

This gets you the raw bits: `rtl_433 -X 'n=proflame,m=OOK_PCM,s=417,l=417,g=1000,r=6000'
The Encoding (MC variant) needs some custom code before you can use the data.

robloh, 2021-12-13:
Thank you so much! I was able to decode the data with some simple code now, awesome. I might take a stab at writing a decoder for rtl_433

zuckschwerdt, 2021-12-13:
Hold off on the decoder, I might look into it today ;)

zuckschwerdt, 2021-12-13:
The checksum / error detection is not as described -- it's noted to maybe include the serial.
Can you collect as many outputs as possible, and post? E.g known so far:
```
Id: 257a02  Cmd1: 51  Cmd2: e2  Err1: 04  Err2: 6b
Id: 47eb02  Cmd1: 00  Cmd2: 36  Err1: 61  Err2: 32
Id: 47eb02  Cmd1: 01  Cmd2: 36  Err1: 40  Err2: 32
```

closed 2021-12-13 by zuckschwerdt

robloh, 2021-12-19:
Attached is a file with as many changes as I could get thru from the included fireplace remote. Last entry is a neighbors fireplace.

[proflame2_sample_output.txt](https://github.com/merbanan/rtl_433/files/7741784/proflame2_sample_output.txt)

BTW, the name (as per the original github link) is "SIT Proflame 2" or "SIT Group Proflame 2", also see: https://www.sitgroup.it/wp-content/uploads/2021/03/584-IFC-9955116_02_Proflame_2_Basic_Standard_Complete_03-03-2016.locked.pdf

It's integrated in to a lot of fire places, including Regency for example which is what I have.

zuckschwerdt, 2021-12-20:
Great! Is it possible to produce these two codes, so we can compare the checksum to the other IDs?

Id: 47eb02  Cmd1: 51  Cmd2: e2  Err1: ?  Err2: ?
- pilot: `0`
- light: `5`
- thermostat: `0`
- power: `1`
- front: `1`
- fan: `6`
- aux: `0`
- flame: `2`

Id: 47eb02  Cmd1: 02  Cmd2: 3e  Err1: ?  Err2: ?
- pilot: `0`
- light: `0`
- thermostat: `1`
- power: `0`
- front: `0`
- fan: `3`
- aux: `1`
- flame: `6`

Also any more codes from the neighbor, if you get those by chance, will help.

zuckschwerdt, 2021-12-21:
It seems like an LFSR digest, i.e. bits are independant (so no CRC) and the mask for each bit rotates to the next bit. E.g. for cmd1/err1 we get a pattern 

|Cmd1-Bit|Err1-Mask|
|--|--|
|6| `42` (`01000010`)|
|7| `21` (`00100001`)|

Looking at the cmd2/err2, you got more distinct codes there, we get

|Cmd2-Bit|Err2-Mask|
|--|--|

|2| `62` (`01100010`)|
|3| `31` (`00110001`)|
|4| `08` (`00001000`)|
|5| `84` (`10000100`)|
|6| `42` (`01000010`)|
|7| `21` (`00100001`)

Observe 3->4 and 4->5 we neither see a consitent left or right shift with a generator though.
Could be that the mask is wider than 8 bit or the key/mask-stream is only LFSR-like.
Interestingly the masks seem identical for cmd1 and cmd2.
We are missing the mask for bit 1, so we can't toggle the 2+4 bit on cmd1 to get an equivalent `51` code :/
A guess for the bit-1 mask would be `c4` (      `11000100`)

robloh, 2021-12-30:
@zuckschwerdt just wanted to let you know that I've not forgotten about this, but we unfortunately had a fire (unrelated!) at the place (everyone okay and such) so it'll be a while before I can capture more stuff again

zuckschwerdt, 2021-12-30:
@robloh thanks for the update. No problem at all. Hope rebuilding goes well!

paulirish, 2022-01-17:
I'm a lurker but also happy to have proflame support here.  

> Great! Is it possible to produce these two codes, so we can compare the checksum to the other IDs?

I tried to collect this data.. my setup:
```sh
$ rtl_433 -f 315M -R 207    # rtl_433 version 21.12 branch
```

I just iterated a bit with the remote to try to get the values you wanted @zuckschwerdt  .. If I invoked the tool wrong, I'm happy to try again.

```
time      : 2022-01-17 14:51:38
model     : Proflame2-Remote                       Id        : 21dd02
Cmd1      : 51           Cmd2      : e2            Err1      : a9            Err2      : e7            Pilot     : 0
Light     : 5            Thermostat: 0             Power     : 1             Front     : 1             Fan       : 6
Aux       : 0            Flame     : 2             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-17 14:56:05
model     : Proflame2-Remote                       Id        : 21dd02
Cmd1      : 02           Cmd2      : 3e            Err1      : 3f            Err2      : 16            Pilot     : 0
Light     : 0            Thermostat: 1             Power     : 0             Front     : 0             Fan       : 3
Aux       : 1            Flame     : 6             Integrity : CHECKSUM
```

zuckschwerdt, 2022-01-18:
@paulirish Thanks! Exactly the right codes. Since the codes differ for every `Id` a full set of all outputs would be even better ;)

And if there are more proflame user stumbling upon this: get all the codes you can get like above and post as .txt or send to me by mail (for privacy).

jbruening, 2022-01-23:
I was attempting to get my own sending of the codes, and yes, the ECC method described in the original repository is not correct. Here is some filtering of recordings I've captured (to help with this mask calculation)

```
time      : 2022-01-23 21:53:01
model     : Proflame2-Remote                       Id        : bf3902
Cmd1      : 01           Cmd2      : 06            Err1      : fc            Err2      : da            Pilot     : 0             Light     : 0
Thermostat: 0            Power     : 1             Front     : 0             Fan       : 0             Aux       : 0             Flame     : 6
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-23 21:54:07
model     : Proflame2-Remote                       Id        : bf3902
Cmd1      : 01           Cmd2      : 05            Err1      : fc            Err2      : b9            Pilot     : 0             Light     : 0
Thermostat: 0            Power     : 1             Front     : 0             Fan       : 0             Aux       : 0             Flame     : 5
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-23 21:54:19
model     : Proflame2-Remote                       Id        : bf3902
Cmd1      : 01           Cmd2      : 04            Err1      : fc            Err2      : 98            Pilot     : 0             Light     : 0
Thermostat: 0            Power     : 1             Front     : 0             Fan       : 0             Aux       : 0             Flame     : 4
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-23 21:54:24
model     : Proflame2-Remote                       Id        : bf3902
Cmd1      : 01           Cmd2      : 03            Err1      : fc            Err2      : 7f            Pilot     : 0             Light     : 0
Thermostat: 0            Power     : 1             Front     : 0             Fan       : 0             Aux       : 0             Flame     : 3
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-23 22:56:05
model     : Proflame2-Remote                       Id        : bf3902
Cmd1      : 01           Cmd2      : 02            Err1      : fc            Err2      : 5e            Pilot     : 0             Light     : 0
Thermostat: 0            Power     : 1             Front     : 0             Fan       : 0             Aux       : 0             Flame     : 2
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-23 22:56:11
model     : Proflame2-Remote                       Id        : bf3902
Cmd1      : 01           Cmd2      : 01            Err1      : fc            Err2      : 3d            Pilot     : 0             Light     : 0
Thermostat: 0            Power     : 1             Front     : 0             Fan       : 0             Aux       : 0             Flame     : 1
Integrity : CHECKSUM
```

jbruening, 2022-01-24:
as an update, I some quick c# interactive scripts, and brute forced the 'C' and 'D' variables referred to in the original repository (https://github.com/johnellinwood/smartfire#error-detection-words). They stay consistent for different inputs + results, at least with the data I have:

```
var X = new Func<byte, byte, byte>((cd, cmd) => {
                var a = cmd >> 4;
                var b = cmd & 0b0001111;
                return (byte)(((cd >> 4) ^ (a << 1) ^ a ^ (b << 1)) & 0xF);
            });
            var Y = new Func<byte, byte, byte>((cd, cmd) => {
                var a = cmd >> 4;
                var b = cmd & 0b0001111;
                return (byte)((cd & 0b0001111) ^ a ^ b);
            });

            var ecc = new Func<byte, byte, byte>((cd, cmd) =>
            {
                var x = X(cd, cmd);
                var y = Y(cd, cmd);
                return (byte)((x << 4) | y);
            });

            var findCandD = new Action<byte, byte>((cmd, r) =>
            {
                for (byte i = 0; i < 0b11111111; i++)
                {
                    var ec = ecc(i, cmd);
                    if (ec == r)
                        Console.WriteLine($"{i:X2}");
                }
            });
```

```
cmd 1....
> findCandD(0x01, 0xfc);
DD

cmd 2...
> findCandD(0x06, 0xda);
1C
> findCandD(0x05, 0xb9);
1C
> findCandD(0x04, 0x98);
1C
> findCandD(0x03, 0x7f);
1C
> findCandD(0x02, 0x5e);
1C
> findCandD(0x01, 0x3d);
1C

zuckschwerdt, 2022-01-24:
Interesting. That would be just to compensate for the ID, the keys for the bits in cmd are the same, I guess.
E.g. with your data
```
00000010  01011110 +
00000011  01111111 =
          00100001

00000100  10011000 +
00000101  10111001 =
          00100001
```
and
```
00000100  10011000 +
00000110  11011010 =
          01000010
```
which matches the mask stream guessed above. I suspect the ID is processed in a similar fashion.

If we work this backwards for all cmd bits, then we get the checksum just after ID and can work out how those bits generate a value.

jbruening, 2022-01-24:
some additional samples for light, fan, and thermostat. I'm unsure of how to get the pilot, aux, or front.

[proflame2_samples_lightNfanNthermo.txt](https://github.com/merbanan/rtl_433/files/7922862/proflame2_samples_lightNfanNthermo.txt)

zuckschwerdt, 2022-01-24:
Your data confirms that the bit-1 mask is be c4 (`11000100`). The mask for bit-0 is unknown, a guess would be `88`.

This lets us roll back the checksum on cmd1 and cmd2 and we should get a fixed initial value per ID. Then we could see how an ID leads to that value.

|Cmd-Bit|Err-Mask|
|--|--|
|1| `c4` (      `11000100`)|
|2| `62` (`     01100010`)|
|3| `31` (`    00110001`)|
|4| `08` (`   00001000`)|
|5| `84` (`  10000100`)|
|6| `42` (` 01000010`)|
|7| `21` (`00100001`)|

zuckschwerdt, 2022-01-24:
Ah yes, that worked. Using
```
Id: 21dd02  Cmd1: 51  Cmd2: e2  Err1: a9  Err2: e7
Id: 21dd02  Cmd1: 02  Cmd2: 3e  Err1: 3f  Err2: 16
```
we can transform `3e` to `62` (and err2 along) which is `e2` without bit-0.
```
11100010  11100111 (e2 e7)
00111110  00010110 (3e 16)
->
01100010  01101111 (62 6f)
and thus
10000000  10001000 (80 88)
```
So the bit-0 mask is indeed `88`.

zuckschwerdt, 2022-01-24:
Or as a closed expression, the upper nibble goes to three positions, the lower nibble to two positions:
`err = init ^ ((cmd&0xf0) << 1) ^ (cmd&0xf0) ^ ((cmd&0xf0) >> 4) ^ (cmd&0x0f) ^ ((cmd&0x0f) << 5)`
Similar to what you have.

lukrative, 2024-10-25:
I've tried one silver SIT remote and the third byte for the serial number was still 0x02 which leads me to believe this field is fixed.

kaechele, 2025-06-05:
I tried ChatGPT to solve this puzzle about a year ago and it failed miserably.

Today it delivered a fully working solution:
```c
/*
 * proflame2_gen.c – stand‑alone packet generator for Proflame 2 fireplaces
 *
 * Build & run:
 *   cc -std=c99 -Wall -O2 proflame2_gen.c -o pf2gen
 *   ./pf2gen
 *
 * The demo in main() prints one valid 7‑byte packet as hex.
 */

#include <stdio.h>
#include <stdint.h>
#include <string.h>

/*———————————————————————————————— XOR‑parity helpers —————————————————————————*/

/* rtl_433 helper – parity of one byte (returns 0 for even, 1 for odd) */
static inline int parity8(uint8_t byte)
{
    byte ^= byte >> 4;
    byte &= 0xf;
    return (0x6996 >> byte) & 1;
}

static const uint64_t pf2_mask[16] = {
    0x800000C000ULL, 0x1080006200ULL, 0x7000003100ULL, 0xC080001000ULL,
    0x2000008000ULL, 0xC080004000ULL, 0x9000002200ULL, 0x6000001100ULL,
    0x90800000C4ULL, 0x0000000062ULL, 0xD000000031ULL, 0x1000000010ULL,
    0x2000000088ULL, 0x2080000044ULL, 0x9080000022ULL, 0xE000000011ULL,
};

static inline uint8_t parity64(uint64_t v)
{
    v ^= v >> 32;
    v ^= v >> 16;
    v ^= v >> 8;              /* parity now only depends on low byte */
    return parity8((uint8_t)v);
}

static inline uint16_t proflame2_calc_parity(const uint8_t *d)
{
    /* Build 40‑bit value: serial_hi serial_lo version CMD1 CMD2 */
    uint64_t data = ((uint64_t)d[0] << 32) | ((uint64_t)d[1] << 24) |
                    ((uint64_t)d[2] << 16) | ((uint64_t)d[3] <<  8) | d[4];

    uint16_t ecc = 0;
    for (int i = 0; i < 16; i++) {
        ecc = (ecc << 1) | parity64(data & pf2_mask[i]);
    }
    return ecc;               /* MS‑bit first, matches on‑air order */
}

/*—————————————————————————— state → packet builder ———————————————————————*/

typedef struct {
    uint16_t serial;       /* handset serial number */
    uint8_t  version;      /* usually 0x02 */

    /* CMD‑1 fields */
    uint8_t  power;        /* 0/1 */
    uint8_t  thermostat;   /* 0‑2 */
    uint8_t  night_light;  /* 0‑6 */
    uint8_t  pilot;        /* 0/1 */

    /* CMD‑2 fields */
    uint8_t  flame_height; /* 0‑6 */
    uint8_t  blower_speed; /* 0‑6 */
    uint8_t  aux;          /* 0/1 */
    uint8_t  split_flow;   /* 0/1 */
} pf2_state_t;

/* Fill 7‑byte buffer with a valid packet */
void pf2_build_packet(const pf2_state_t *st, uint8_t pkt[7])
{
    /* serial + version */
    pkt[0] = st->serial >> 8;
    pkt[1] = st->serial & 0xFF;
    pkt[2] = st->version;

    /* CMD‑1 */
    pkt[3] = (st->pilot & 1)            << 7 |
             (st->night_light & 7)      << 4 |
             (st->thermostat & 3)       << 1 |
             (st->power & 1);

    /* CMD‑2 */
    pkt[4] = (st->split_flow & 1)       << 7 |
             (st->blower_speed & 7)     << 4 |
             (st->aux & 1)              << 3 |
             (st->flame_height & 7);

    /* parity */
    uint16_t ecc = proflame2_calc_parity(pkt);
    pkt[5] = ecc >> 8;
    pkt[6] = ecc & 0xFF;
}

/*——————————————————————————— demo / test harness ———————————————————————*/

static void print_packet(const uint8_t pkt[7])
{
    for (int i = 0; i < 7; i++)
        printf("%02X ", pkt[i]);
    putchar('\n');
}

int main(void)
{
    pf2_state_t s = {
        .serial        = 0xa4ed, /* example handset */
        .version       = 0x02,
        .power         = 1,
        .thermostat    = 1,
        .night_light   = 3,
        .pilot         = 1,
        .flame_height  = 4,
        .blower_speed  = 2,
        .aux           = 0,
        .split_flow    = 1,
    };

    uint8_t pkt[7];
    pf2_build_packet(&s, pkt);

    printf("Generated Proflame 2 packet: ");
    print_packet(pkt);    /* should look like: A4 ED 02 B3 A4 53 46 (example) */

    return 0;
}
```

zuckschwerdt, 2025-06-05:
Interesting results from a LLM. It just tabled the suspected LFSR though -- we still don't know how the suspected LFSR is generated, I think.

kaechele, 2025-06-06:
You're right. I got a little excited there because I personally wasn't able to figure out the code required to generate packets from the information contained in this issue. The code is reasonably small, even with the mask table, that I think it'll work for my use case.

From what I have learned so far, I don't know if we'll ever get the description for the LFSR (if there is one) figured out, because the set of possible values each remote can generate is very limited (given there are only so many states each component of the fireplace can be in). So we can't use something like the Berlekamp–Massey algorithm to reverse engineer it, as we can't arbitrarily generate long messages through the original LFSR.

Thanks for taking a look anyway :)

dblessing, 2025-10-20:
You all mention that the ECC calculation in https://github.com/johnellinwood/smartfire is not correct. Is it possible there are different ECC calculations for different serials?

  1. The serials from this thread (0x47EB02, 0xA4ED02, 0xBF3902, 0x21DD02) successfully pair with my fireplace using the 16-mask parity ECC with NO XOR
  2. My cloned remote serial (0x4B5202) works with 16-mask parity ECC but requires XOR 0x0902
  3. The smartfire repo serial (0x257A02) does NOT pair with my fireplace using either method. But the ECC calculation in that repo is entirely different. Maybe that serial is just a test value and isn't valid? 

Randomly generated serials consistently fail to pair regardless of whether I apply XOR or not. This suggests there's some validation on the serial number itself (perhaps a checksum relationship between the hi and lo bytes?) beyond just ECC correctness.

 Has anyone successfully paired with a randomly generated serial number? Or is cloning from an existing remote the only reliable approach?

Aetf, 2026-08-20:
Small note on where the packet dumps from this thread ended up, in case anyone
minds.

I've published a Python library for this protocol
([`proflame`](https://github.com/Aetf/proflame), MIT/Apache-2.0), and its
tests use the aggregated `cmd.csv` posted here — 220 packets from five
remotes. I'd downloaded it from this thread a while back for an earlier
prototype of my own and had lost track of where it came from; I traced it back
recently and the repository now credits this thread. If anyone who captured
those dumps would rather they weren't redistributed, say so and I'll drop
them.

One data point in return, since I have a handset to measure. My captures
quantize cleanly at a **450 µs** symbol period — pulse widths cluster at
450/900 µs at 2 Msps, 349 recorded frames agree, and a frame comes out around
81 ms rather than the ~75 ms you'd expect at 417 µs. Both clocks evidently
decode in practice, since slicing rounds and resyncs per edge, so this isn't a
bug report. But 8% is more than my measurement noise, so I suspect a hardware
variation. If anyone's captures land nearer 417 µs, comparing the two would
settle it — happy to share raw timings.

kaechele, 2026-08-21:
Glad you found it useful, @Aetf 
I have no issue with the `cmd.csv` I posted here being used.

Took a look at your repo, and here are a few notes for you so that Claude doesn't need to go down some of the rabbit holes it is mentioning:
1. The IFC cannot echo back commands as it doesn't have a transmitter. There's a relatively easy way to confirm this: There is no FCC ID for the IFC, as it doesn't transmit andthe schematic of the remote also only lists a transmitter (so at least the remote couldn't hear any echoed back commands, even if it wanted to).
2. The IFC *does* have state. It is available on the UART port. I go into a bit of detail in the documentation of my `bonaparte` library, which is for Napoleon fireplaces that come with the Bluetooth module: https://bonaparte.readthedocs.io/en/latest/system_information.html
The Napoleon module basically passes through what we identified here as the `cmd1` and `cmd2` structures to the IFCs UART. But I haven't done any reverse engineering on the actual SIT UART debug protocol, so I don't fully know how it actually works.

Aetf, 2026-08-21:
Thanks for the tips, and for the OK on using the cmd.csv, @kaechele!

I recall reading somewhere that there's an echo/ack response, but I can't confirm this without proper equipment. I write code but have basically no RF background, so I've been relying on Claude quite a bit — surprisingly helpful, honestly :P. It flagged some odd frames that it thinks could possibly be two sources transmitting simultaneously, but the lack of an FCC ID is definitely stronger evidence on its own. It would still be nice to know the actual state of the fireplace for safety reasons, but for now I'm satisfied with my periodic re-transmit approach, basically mirroring what the remote does.

That ESPHome project looks really cool — I already have a bunch of ESP32s lying around the house, so I would have used it in a heartbeat if I'd known about it earlier. But now that I have a working setup, I probably won't be tinkering with it much more. LOL

commit 21d7fade (2021-12-13): Add support for SmartFire Proflame 2 remote control (closes #1905)