merbanan/rtl_433 #1906: RTL stops receiving data with HA on RPi 4

donoghdb, 2021-12-14:
I have tried a number of addon's for HA but the issue still occurs where rtl_433 stops receiving data after a number of hours. Im using protocol 43. I have an automation to restart the addon after a time period and sometimes that works, other times i need to unplug the SDR. How can i see what is happening and why it is not receiving data?
I am using this integration https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433

zuckschwerdt, 2021-12-15:
Likely the band you are watching is jammed or the signal drifted out of range. See https://triq.org/rtl_433/ANALYZE.html

donoghdb, 2021-12-15:
looking at the data, most of the time i get data every 15 minutes then sometimes nothing. No sign of it drifting (not that i can see anyway from the charts). I switched to a mac so i can watch the data come in. I wonder if anyone else has seen the issue of the Oil watchman randomly stopping sending data. There is nothing around that is changing to suddenly interfere with the signal.

zuckschwerdt, 2021-12-17:
The `-S unknown` mode can help to find undecoded transmissions. It will take up some disk space though. Or perhaps try the HTTP output and follow signals there, use Modes->Raw to visualize but not save the signals.

bkenobi, 2021-12-23:
I have a similar issue but wasn't sure if it was the same so opened another issue.  If it turns out these are related they can be combined.

https://github.com/merbanan/rtl_433/issues/1912

donoghdb, 2021-12-23:
I moved my antenna, reinstalled HA and RTL and I am no longer getting the issue. The data I'm getting on the terminal is the same as MQTT. I do have an automation checking the time between messages and if it's above 30 minutes it restarts the addin. But since I have done the above, it has not triggered.

bkenobi, 2021-12-23:
The data output to MQTT is the same as decoded. In my case one or more devices stop being decoded after a while. I currently have 5 total (1 blueline, 4 accurite). The blueline has been going offline recently. Was off yesterday, i restarted rtl_433 and it was back. It went down again last night but havent checked it yet.

danielgoepp, 2022-06-23:
It looks like this thread is largely dead, but I will say I have the same problem. It's frustrating. Even restarting sometimes doesn't fix it. Seems if I turn it off for a while, then it will kick back in. Also, strangely, if I play with the antenna it sometimes comes back. If I run directly off a computer, no problem, but on a Pi, it will die every couple days. I'd definitely take any advice on perhaps how to tune the radio settings a bit more? I'm running pretty much default, which works, until it doesn't.

bkenobi, 2022-06-23:
I should have updated but in my case the issue turned out to be a faulty sensor. In my experience the Blueline device is very poor quality. My 1st unit lasted a few years before water got in combined with battery corrosion. The 2nd (this one) lasted a couple months. The 3rd was doa. The 4th unit has been perfect since installed (1/2022 ish).

I have also tested a lot more acurite sensors. I have a 5in1 weather station and ~8 temp/humidity sensors that can all work together (though the do start to interfere).  During all of this I've had minimal issue with the radio or rtl_433.  In the other thread i provide a solution in case the unit doesn't work the 1st time due to a bad auto-gain setting.

danielgoepp, 2022-06-23:
@bkenobi thanks for the follow up. I just started reading that other thread now...that will take a while :) I have an Acurite Atlas and 3 different radios, they all seem to do the same (2xNooelec NESDR and one that I got with the ADS-B PiAware project).

gdt, 2023-09-30:
This feels like several reports that all amounted to flaky hardware.   If you have a situation where you think there's a bug in rtl_433 please feel free to open a fresh issue, tested against up-to-date git master.

closed 2023-09-30 by gdt