merbanan/rtl_433 #1912: rtl_433 stops recieving data after days or weeks

bkenobi, 2021-12-23:
I use rtl_433 (20.11-85) with an RPi 3 and SDR to receive data from several Acurite sensors and a BlueLine power meter reader.  In order to get everything receiving correctly on startup, I have to launch rtl_433 several times until it gets the gain set correctly.  Once data starts flowing correctly, I can generally let everything work unattended for at least hours if not days to weeks.  At some point, one of the protocols stops for some unknown reason.  Last week I noticed that my 5in1 weather station was not updating so I restarted the RPi and everything was good.  I added a watchdog to my Node-Red flows to email me when data has not been seen for a few hours.  This evening I was informed that the Blueline power meter was not being received.

I am planning on updating to the latest version, but I thought it might be better to report this with my existing version first so I can provide any helpful debug data.  I can repeat with the same with the new version if necessary.

My first thought was I could create a NR flow or Bash script to restart the service should the data fall off but it seems like raising the issue first would be preferable all around in case others are having similar issues.  Might as well fix the problem rather than kludge a fix after all.

zuckschwerdt, 2021-12-23:
That's not normal. "[restart to] the gain set correctly", do you mean manual gain? AGC shouldn't get "stuck".
Try `-Y autolevel` with the new version and also use `-M level` to track signal strength and `-M noise` to track if the band is clean.

bkenobi, 2021-12-23:
I worked through the startup problem earlier this year:
https://github.com/merbanan/rtl_433/issues/1669

The result of that thread was to create a script that would run rtl_433 several times before launching for real. It has something to do with how rtl_433 initializes my SDR. I agree fixing that would be better but at the time i was the only one who reported the issue. I see that a few others have now reported same experience but no fix.

Maybe this is related to that issue.  I will update and try your added switches to test.

bkenobi, 2021-12-23:
I updated to the latest version of rtl_433 and launched the basic command to monitor traffic.  The Accurite sensors came up correctly but not the Blueline.  I replaced the batteries from the Blueline meter since rtl_433 doesn't seem to work with the battery bit and it came back online.  The old ones were dead which explains why it was dropping.  However, the Accurite is showing no warning for batteries either with the battery bit or with the Accurite display and yet it has dropped recently too.

I will continue to monitor to see if the system is more stable with the new version.  Hopefully the update will improve things.

bkenobi, 2021-12-23:
This is what the data from the weather station looks like (temperature is representative of holes).  It does not appear to be based on light level.

zuckschwerdt, 2021-12-23:
To help with debugging there is now https://github.com/merbanan/rtl_433/pull/1915
Run rtl_433 with added `-F rtl_tcp:0.0.0.0`, then run Gqrx any time you want to check what radio data the receiver is actually delivering.

bkenobi, 2021-12-24:
I updated my service so I can try that.  Is there a guide on how to configure gqrx to see the data/gain/etc?  I looked around and see some info on how to configure it to listen to radio/Ham stations abroad but nothing about displaying details on data (other than seeing the spectrum plot of course).

bkenobi, 2021-12-24:
I tried running the option you recommended `-F rtl_tcp:0.0.0.0` but I got an error indicating that the rtl_tcp command was not available.

`pi@Raspi3:~ $ rtl_433 -F rtl_tcp
rtl_433 version 21.12-6-g8008da78 branch master at 202112232346 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Invalid output format rtl_tcp
Generic RF data receiver and decoder for ISM band devices using RTL-SDR and SoapySDR.
`

zuckschwerdt, 2021-12-24:
Clone, then switch to the feature branch before compiling: `git checkout feat-rtsrv`
The feature "lives" on that branch, and a clone will get you all branches :)

bkenobi, 2021-12-27:
I rebuilt rtl_433 with the feature checked out and am able to launch with both rtl_tcp and mqtt running.  I installed SDRsharp on my Win10 machine but couldn't figure out how to visualize anything (it's been a while since using it last).  I installed SDR Touch on my phone and was able to set the frequency to 433 and saw the waterfall but was not clear how to see any other details.  Is there a guide that shows how to determine if what the SDR is receiving is valid?  I guess I was envisioning something like IQ Spectrograph, but anything that shows that there is something would be ok I suppose.

zuckschwerdt, 2021-12-27:
Yes it should look like the IQ Spectrogram (but sideways, a waterfall). You should see every transmission as a short intense streak. It's the same view you would get without rtl_433 from Gqrx or any other SDR software. rtl_433 will just give the unmodified data through -- that way you can run rtl_433 and an SDR software (needs rtl_tcp support) at the same time.

bkenobi, 2021-12-30:
Ok, got it. Next time the devices disapear ill check the spectrum analyzer to confirm that i have any traffic on 433 or if the gain is all out of range again. I'll leave this open until i have a chance to look at the signal at that time.

bkenobi, 2022-01-22:
Well I had an interesting development this morning.  I noticed that my power had been reported as 2.7kW constant for a while.  Here's the data that makes sense for Jan1-Jan18.
[image]
Then this happened:
[image]
So when I saw that the usage wasn't changing I tried resetting rtl_433 in case it was just being interpreted wrong.  It didn't seem to change anything.  I then pulled the batteries from the device and pressed the reset button.  The output changed...
[image]
I have a transformer box on my property with 2 200A panels so I can pull 400A or 96kW technically.  But, I don't have anything that would do that and it's constant anyway.  If I were pulling 1600kW I'm pretty sure the cops would show up looking for my grow operation!

This device is around a year old (I purchased it when the code was added to rtl_433 to replace a dead unit).  Everything looks good inside (no water intrusion or corrosion).  Has anyone seen this kind of an issue?

bkenobi, 2022-01-22:
I guess I should mention what rtl_433 is outputting for completeness.  This is what is being output to MQTT:

```rtl_433/Blueline-PowerCost/time 2022-01-22 03:18:45
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/flags 31
rtl_433/Blueline-PowerCost/battery_ok 1
rtl_433/Blueline-PowerCost/temperature_C 2.776
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 03:18:45
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/flags 31
rtl_433/Blueline-PowerCost/battery_ok 1
rtl_433/Blueline-PowerCost/temperature_C 2.776
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 03:18:45
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
```

bkenobi, 2022-01-22:
In looking more at the data, it seems more and more like the Blueline device must be faulty somehow.  This is the last 7 days:
[image]
And here is the last 12 hours including the period after I restarted the device.
[image]
It seems pretty clear that the device seems to be reaching the maximum value for impulses which is why there is a saw tooth pattern.  Just to make sure I'm not running an unknown grow lab on a neighbor's property, I watched the wheel and recorded the gap manually for a few minutes.
[image]
Based on the time and calculated kW, the hand calcs seem accurate and the Blueline appears to be garbage.  Booo!

bkenobi, 2022-01-22:
My kh is 24 so using the formula:
```
kW=3600/gap*kh
gap=3600*kh/kW
```
If a reasonable usage is around 10kW, then I should see a gap of 8640.  The range I could technically ever see reported would have to be between 90 (96kW theoretical maximum) and 65533 (1.3kW minimum for my kh).  
```
rtl_433/Blueline-PowerCost/time 2022-01-22 14:22:24
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/impulses 28899
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:22:24
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/impulses 28899
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:22:24
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:22:55
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:22:55
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:22:56
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:23:27
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:23:27
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:23:27
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:23:59
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:23:59
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:23:59
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:24:31
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:24:31
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:24:31
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:25:03
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/flags 31
[... 201 lines cut ...]
rtl_433/Blueline-PowerCost/time 2022-01-22 14:34:04
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:34:04
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
rtl_433/Blueline-PowerCost/time 2022-01-22 14:34:05
rtl_433/Blueline-PowerCost/id 55280
rtl_433/Blueline-PowerCost/gap 53
rtl_433/Blueline-PowerCost/mic CRC
```
The rtl_433 code is sending valid temperatures to MQTT but the gap is a constant 53.  The impulses value should slowly increase but in my case, it starts at 28899 and ends at 39531 in a ~12 minute recording.  That's a delta of 10632.
```
kWh = 0.001 * (accumulated pulses) * Kh
```
So this is suggesting that I've used 255kWh in 12 minutes!

This is what the current 2 day log of data looks like in Grafana
[image]

Looking at all of this, it appears that the unit is sending valid packets as the CRC check must be completed ok for the code to output anything.  The temperature matches my weather station so it is working but the impulses and gap are WAY off.

The only 2 things I can think to check are moving the sensor away from the wheel and seeing if it still sends data.  If so, the unit is clearly faulty for whatever reason.  Second, I noticed there was water trapped between the reader arm and the glass bell.  Since they are so close, this must be a common occurance but I can dry it off and maybe blow air on the sensor to clean it up.  I don't see how that would cause it to detect rediculously fast, but it's worth checking.

zuckschwerdt, 2022-01-22:
So, the sensor seems to be working but the input data to the sensor is "flashing too fast". Perhaps something with the water is causing a lens-like effect with some "glitter" due to refraction? Like a fresnel lens?

bkenobi, 2022-01-22:
That's what I was thinking, too.  It's still dark here so I'll take a look when the sun comes up.  Looking at the rtl_433 standard output format, the results are the same which was expected.  The only other thing I could do to confirm it's the meter is dig out the receiver that came with the unit and configure it.  Since the CRC is good though, that seems pointless.  I'll update when I check those 2 things.

bkenobi, 2022-01-22:
[image]
I cleaned the probe tip but it was dry this morning.  I moved it below the wheel so it can't see anything and yet it's clearly happily trotting away at a very high rate.  I think I'll remove it from the meter so there's no way it could see anything.  Worst case it still doesn't work. Best case, it bumps something lose I guess.

zuckschwerdt, 2022-01-22:
Good luck! (I have/had an IR-optical reader on a meter, then it suddenly degraded and failed within 2-3 days. No way to get it working again. Then I got a hint of "oh, that brand of meter, they have trouble with the IR LEDs, failing after half a year". Looked at the calendar: ~6 months...)

bkenobi, 2022-02-08:
I received my replacement meter reader last week and have been watching the data to see if all is working.  The interesting thing is that it is identical to my last unit (Black and Decker EM100B) but the reported power usage is double what my previous reader was reporting and the output is not nearly as smooth.  I haven't changed anything in code in Node Red and my code is very simple (direct from the documentation).

```
//kW = (3600/gap) * Kh
var Kh = 24.0;
msg.topic = "rtl_433/Blueline_PowerCost_Monitor/kW"
msg.payload = (3600.0 / msg.payload)*Kh;
return msg;
```

Here's an example of a couple dates at random:
[image]
[image]

I stated double the cost because I have been recording the numeric reading from the meter manually for years and know that the usage should be on the order of 2 units per day which works out to around 80kWh but the the code suggests I went from 282 to 469 kWh accumulated or a difference of 187kWh.  I can't explain either the double reporting nor the noisy signal.  This is a new unit so I suppose it could be faulty but considering these aren't on a shelf anymore, it makes it difficult to find a new perfect unit without guess and check.

zuckschwerdt, 2022-02-08:
Yes, that new kW curve looks too noisy. Old one looks plausible, like e.g. some base load, periodic a motor, and then sometimes lamps on top.
Is a phase feeding this circuit accessible? A contactless current probe, e.g. Uni-T UT210E, might be a way to at least get a feel (measures just A, integrate by multiplying and adding real fast ;) what the kW should be.

merbanan, 2022-02-08:
I think the IR sensor needs an adjustment. Or add some duct tape to get rid of reflection or something.

bkenobi, 2022-02-08:
One other thing I just thought of looking at is the impulses.  If the device is seeing extraneous flashes, the gap will be shorter than in reality and the power usage will be higher.  The impulses should support this by showing many more wheel spins than is actually happening.

```
time	impulses	delta	hand measurements
8:57:05	31827		
9:00:17	31855	28	
9:03:28	31875	20	
9:06:40	31899	24	10
9:09:51	31927	28	9
9:13:02	31935	8	3
9:16:14	31939	4	
```
Formatting may make that hard to read so here's an image.

[image]

I didn't record enough manual data to prove anything here since the first and last numbers above didn't encompass the entire time range (3:12 between impulse packets).  The middle point is valid but it shows that the sensor is reporting around 3x too many impulses (10 versus 24).

bkenobi, 2022-02-08:
I just noticed something a bit odd in the data.  The protocol for the Blueline meter is to output data every ~30 seconds in a burst of 3 repeated packets that contain the same data.  There are different types of packets, but when the data is sent, all 3 are identical.  I'm looking at my data and seeing some strange trends.

The data is sent 3x for redundancy in case it doesn't make it through.  So I don't expect to see all of the data from all packets.  However, any time I see 3 at the same time, they should produce the same data.  Not always the case:
```
time	blueline
8:56:34	5849
8:57:06	3957
8:57:37	3797
8:57:37	3797
8:57:37	3797
8:58:09	6705
8:58:09	6705
8:58:09	6705
8:58:41	6713
8:58:41	6713
8:58:41	6713
8:59:13	9933
8:59:13	9933
8:59:13	10145
8:59:45	6977
9:00:17	7145
9:00:49	17941
9:00:49	17941
9:00:49	18149
9:01:21	18857
9:01:21	18857
9:01:21	18857
9:01:52	8881
9:01:53	8881
9:01:53	8881
9:02:24	8609
9:02:24	8609
9:02:25	8609
9:02:56	8777
9:04:00	8677
9:04:00	8677
9:04:00	8677
9:04:32	8901
9:04:32	8901
9:05:04	13181
9:05:04	13181
9:05:04	13389
9:05:36	12961
9:05:36	12961
9:05:36	13169
9:06:08	7041
9:06:40	2593
9:07:12	1721
9:07:12	1721
9:07:12	1929
9:07:43	1333
9:07:44	1333
9:07:44	1333
9:08:15	5913
9:08:15	5913
9:08:16	5913
9:08:47	11377
9:08:47	11377
9:08:47	11585
9:09:19	10693
9:09:51	6641
9:10:23	1521
9:10:23	1521
9:10:23	1729
9:10:55	33513
9:10:55	33513
9:10:55	33713
9:11:27	15041
9:11:27	15041
9:11:27	15041
9:11:58	44645
9:11:59	44645
9:11:59	44853
9:12:31	15201
9:13:02	47189
9:13:34	61869
9:13:34	61869
9:13:34	61869
9:14:06	46233
9:14:06	46233
9:14:06	46233
9:14:38	19929
9:14:38	19929
9:14:38	20137
9:15:10	46365
9:15:10	46365
9:15:10	46365
9:15:42	22665
9:16:14	54661
9:16:46	65521
9:16:46	65521
9:16:46	65521
9:17:17	5269
9:17:18	5269
9:17:18	5269
9:17:49	16913
9:17:49	16913
```
Looking at times:
8:59:13
9:00:49
9:05:04
9:05:36
9:07:12
9:08:47
9:10:23
9:10:55
9:11:59
9:14:38

All of these have the same value for the gap for the first 2 but a different one for the 3rd.  There are many that output the correct data for all 3.  I'm not sure what would cause this but it's certainly odd.

I've put painters tape over the sensor to see if shielding the IR helps.  I can also try adding a gap (right now the sensor is touching the glass).  If I do that, I'll also pull it off and see if there's some kind of residue/object blocking that I didn't see when installing initially.

bkenobi, 2022-02-08:
Still seems pretty noisy.
[image]
I put tape on the meter around 10:28 and the usage is jumping from 5-10kW without anything significant running in the house.  I could put a cardboard box over it but otherwise it's gotta be the sensor.

bkenobi, 2022-02-08:
I put a box over the meter around 30 minutes ago and the output is still garbage.  Either it's the IR LED or it's junk.  Awesome!  The temperature is perfectly fine so it's not the unit or the transmission.

bkenobi, 2022-02-08:
Well that certainly proves it.  I pulled the unit to check the transmit/receive LED's and both looked fine.  I guess I could have tried alcohol on them, but they were clean.  I installed new batteries though the old ones were only a few days old.  The meter is still reporting bizarre things.  Yes, that's going from 5kW to 47kW down to 3kW in a span of 5 minutes.  The heat pump was on at some point, but it doesn't draw 47kW!

[image]

I pulled the cover on my 2-200A panels to measure the current with my clamp meter.  I found with the furnace off and everything quiet in the house 9.7A draw (1.1kW) around 14:30 and a little while later something pulled it up to 11.7A (1.4kW).  If I do the math, the gap should be around that 65536 mark and yet when I checked it was all over the place as small as ~4000ms.  This thing is broken or garbage!

If I send it back and try again, I'll be on my 4th meter in a little over a year.  Hmmm, maybe there's a reason Blueline discontinued these!

zuckschwerdt, 2022-02-09:
When my readings failed, I tried to diagnose the IR-LED: I hooked an IR-LED as receiver (loaded with a 10-100k resistor) up to a scope to see the readings (the output is digital data, not just flashes). The readings where mostly ok, but really faint. I then used a proper IR transistor from a CNY 70 hooked to first a Saleae knock-off logic analyzer, then just to a Raspberry Pi Zero. Which all worked well. I guess a ESP8266 (NodeMCU) will also work (I have some Tasmota code to read flashes which I should PR someday). Maybe you need to build your own Wifi-enabled reader, one you can trust? Counting flashes should be as easy as 10 lines Python to read edge trigger from a raw pin.

bkenobi, 2022-02-09:
The only reason I don't is the weatherproof enclosure.  I wasn't impressed by the Blueline design as it allowed water in.  However, the B&D version seems to have improved the design and it is still water tight.  If I could repurpose the enclosure that might be a great option.  At this point I have 2 broken ones and the IR LED from the original one too.  ESP devices are fairly small but the battery life is likely to be an issue since it needs to transmit every time the wheel comes around which could be as little as 5 seconds or as long as over 1 minute.  I'm not adverse to writing the code, it's really just the weather proofing and battery life that keeps me from rolling my own.

bkenobi, 2022-02-16:
My 4th power meter reader arrived this morning.  I only have had it installed for the last 30 minutes or so, but at least it is making the right motions.  The data is smooth and around 5kW with heat pump on and drops back to 1.3kW (minimum for my setup) when it turns off.  I'll continue to monitor but seems like these sensors work great when the work but may not be that good in the long term.  I don't think I'll buy another at this point even though they are fairly cheap on secondary market.

I found some links to projects where others have rolled their own.  Since I have all of these enclosures that are designed to mount on a power meter, if I can find a way to power an ESP-type device long enough I may just do something.  The LED/read head can be easily removed and reused so it then becomes a question of finding a solution where batteries/solar can be used efficiently to power such a device.  But, that's out of scope of the rtl_433 project obviously.