merbanan/rtl_433 #1913: wind_max_km_h in rtl_433_mqtt_hass.py

thtro, 2021-12-23:
Hi,

"wind_max_km_h" should also be a key, alternative to "gust_speed_km_h".
This is what my station publishes:
`{"time" : "2021-12-23 18:45:20", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 5, "battery_ok" : 1, "temperature_C" : 0.200, "humidity" : 69, "wind_dir_deg" : 113, "wind_avg_km_h" : 3.672, "wind_max_km_h" : 6.120, "rain_mm" : 135.300, "mic" : "CRC", "mod" : "ASK", "freq" : 868.359, "rssi" : -0.104, "snr" : 17.791, "noise" : -17.895}`

zuckschwerdt, 2021-12-23:
Yes. PR welcome.

gdt, 2023-09-30:
ticket gardening hat: What's the status and path forward?  Is this still an issue with git master?

gdt, 2023-09-30:
normal hat: Either that, or the decoder should be adjusted.    I personally lean to regularizing names/semantics, and am not so concerned about compat.

merbanan, 2026-11-08:
Fixed in #1914 (3689ff1f): `rtl_433_mqtt_hass.py` maps `wind_max_km_h` alongside `gust_speed_km_h`.

closed 2026-11-08 by merbanan