merbanan/rtl_433 #1916: Windows Building Instruction Problems

MaxFun, 2021-12-25:
When following build instructions from https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md had problems.
Download the source code for librtlsdr from https://github.com/librtlsdr/librtlsdr, not https://github.com/steve-m/librtlsdr per instructions. This solved the problem of no FindLibUSB.cmake file.
When building the librtlsdr library: mingw32-make, errors were produced, see attached file "build the librtlsdr library 12-25-21.docx"
Error;  "cannot find -lusb-1.0", but lusb-1.0 is in "C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/include/ from downloaded  libusb-1.0 per instructions.

Build instruction did not indicate if errors were acceptable or not, so I continued blindly following instructions.

When clicking configure in cmake-gui for rtl_433, the GUI opened a window asking "Specify the generator for this project". Visual Studio 17 2022 was in the window for selection. What generator for rtl_433 should be selected?

[build the librtlsdr library 12-25-21.docx](https://github.com/merbanan/rtl_433/files/7775977/build.the.librtlsdr.library.12-25-21.docx)

zuckschwerdt, 2022-04-24:
There are new build instructions for Visual Studio 2017: https://triq.org/rtl_433/BUILDING.html#visual-studio-2017
Should be really easy now, just a few clicks.

closed 2022-04-24 by zuckschwerdt

MaxFun, 2022-04-26:
Tried following the Visual Studio 2017 instructions. Was bale to install PothosSDR, but had to manually add PothosSDR to the system PATH for the current user, error was "path to long"
Did not install Chocolatery, in the past my Windows 10 has problems with mqtts.
Opened Visual Studio 2017, cloned https://github.com/merbanan/rtl_433.git, It seemed to build correctly with output "Build succeed"
Selected rtl_433.c and started debug which produced multiple errors
'rtl_433.exe' (Win32): Loaded 'C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\build\x64-Debug (default)\src\rtl_433.exe'. Symbols loaded.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. Cannot find or open the PDB file.
'rtl_433.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. Cannot find or open the PDB file.
The thread 0x3dbc has exited with code -1073741515 (0xc0000135).
The thread 0x31bc has exited with code -1073741515 (0xc0000135).
The program '[1172] rtl_433.exe' has exited with code -1073741515 (0xc0000135) 'A dependent dll was not found'.
System error indicated the dll was rtlsdr.dll .  Did find rtlsdr.dll in C:\Windows\System32, modified date was 8/12/19

Is Chocolatery required?

zuckschwerdt, 2022-04-26:
"chocolatey" is just one option to install OpenSSL on Windows. On Windows 10 you can also use "WinGet". But you don't need OpenSSL at all (unless you use mqtts) -- thus no need for chocolatey.

A plain PothosSDR install with all defaults, just selecting "Add PothosSDR to the system PATH for the current user" should just work. It will install the needed libs.
On manual install you would need to make the libs available in some way. Worst case: copy the libs alongside the rtl_433.exe.

MaxFun, 2022-04-26:
Uninstalled PothosSDR, deleted C:\Users\MySelf\source\repos\merbanan
Restarted computer , installed PothosSDR, it found a copy of PothosSDR and wanted to remove the old and replace with the new, but could not find it to uninstall?
Deleted all files and folders which had “PothosSDR” in its name. Power down and restarted computer.
Same problem; installed PothosSDR, it found a copy of PothosSDR and wanted to remove the old and replace with the new, but could not find it to uninstall?
Tried not removing old and PothosSDR appear to have installed

Started Visual Studio 2017,  “Clone” and enter https://github.com/merbanan/rtl_433.git
1> Command line: C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2017\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe  -G "Ninja" -DCMAKE_INSTALL_PREFIX:PATH="C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\install\x64-Debug (default)"  -DCMAKE_CXX_COMPILER="C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe"  -DCMAKE_C_COMPILER="C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe"  -DCMAKE_BUILD_TYPE="Debug" -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2017\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\Users\MySelf\Source\Repos\merbanan\rtl_433"
1> Working directory: C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\build\x64-Debug (default)
1> -- Found Git version: refs/heads/master commit bb1be7f186ac0fdb7dc5d77693847d96fb95281e from 2022-04-25T11:00:23+0200
1> -- Using Git version tag: 21.12-100-gbb1be7f1 on master at 202204251100
1> -- IPv6 support enabled.
1> -- Threads support will be compiled.
1> -- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR) 
1> -- OpenSSL development files not found, TLS won't be possible.
1> -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> -- RTL-SDR device input will be compiled. Found version 
1> -- libusb-1.0 error messages are available. Found version 
1> -- SoapySDR device input will be compiled. Found version 0.8.1-PothosSDR-2021.07.25-vc16-x64
1> -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
1> -- Configuring done
1> -- Generating done
1> -- Build files have been written to: C:/Users/MySelf/CMakeBuilds/4f482441-8b72-4531-987a-ff95273c27bd/build/x64-Debug (default)
1> Starting CMake target info extraction ...
1> CMake server connection made.
1> Extracted includes paths.
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted global settings.
1> Extracted code model.
1> Extracted CTest info.
1> Collating data ...
1> Target info extraction done.
It looks like a lot of stuff is missing
Tried to run rtl_433.c in debug and got the following error
rtl_433.exe – Entry Points Not Found
The procedure entry point SoapySDR_free could not be located in the dynamic link library

MaxFun, 2022-04-26:
I'm running 
Edition	Windows 10 Pro
Version	21H1
Installed on	‎8/‎14/‎2020
OS build	19043.1645
Experience	Windows Feature Experience Pack 120.2212.4170.0

What version of windows did you use?

zuckschwerdt, 2022-04-26:
That looks perfect. `PkgConfig` and `Doxygen` are *not expected* to be there. All good.
Can't say anything about debugger integration. If you run "Build All", open a Command Prompt and go to `C:/Users/MySelf/CMakeBuilds/4f482441-8b72-4531-987a-ff95273c27bd/build/x64-Debug (default)/src` then run `rtl_433.exe` it should work, right?

zuckschwerdt, 2022-04-26:
Oh, if it's always exactly "SoapySDR_free" that's missing then your PothosSDR version is not recent enough.
Early SoapySDR 0.8 versions (before Jul 2020) are broken I guess. The recommended `PothosSDR-2021.07.25-vc16-x64.exe` will work.

For debugging (if you really want that) I guess you need to set the Startup Item to `rtl_433.exe (src/rtl_433.exe)`, verify the active configuration is "Debug", set some breakpoint, e.g. in `rtl_433.c` somewhere in `main()`, then just "Start".
I don't think you can just run/debug rtl_433.c alone.

MaxFun, 2022-04-26:
I download the latest, PothosSDR-2021.07.25-vc16-x64.exe and that what giving the error

zuckschwerdt, 2022-04-26:
Not sure I can help then. Likely some old installs or custom paths settings. It just works for me (In Win 8.1, 10, 11 with VS2017, 2019, 2022).

But if you are not using Soapy inputs you can just go to the CMakeSettings.json (change the build config to trigger VS to generate that file) and add to the `cmakeCommandArgs` the string `-DENABLE_SOAPYSDR=OFF`

MaxFun, 2022-04-26:
Cannot find CMakeSettings.json in the VS project or anywhere on my computer.

zuckschwerdt, 2022-04-26:
Go to "CMake" > "Change CMake Settings" > "rtl433", select "x64-Release" and it's generated and opened to edit.

zuckschwerdt, 2022-04-26:
> "cmakeCommandArgs": "","-DENABLE_SOAPYSDR=OFF",

that needs to be (I should have said that more carefully, not *adding*, but *inserting*)

> "cmakeCommandArgs": "-DENABLE_SOAPYSDR=OFF",

zuckschwerdt, 2022-04-26:
Strange. Did you try to clear the CMake cache and/or do a full rebuild? The configure should show that SoapySDR is disabled.
Also I wonder why you use a MinGW command window? Does that perhaps mess with paths? Did you try a "cmd.exe" Command Prompt?

MaxFun, 2022-04-26:
Cashe(x64-Release Only)-> Deleted Cashe Folders->rtl433
Cashe(x64-Release Only)->Generate->rtl433
Cmake -> Build All , opened mingw-w64 command window and ran rtl_433.exe
Got the same error

Opened Command Prompt -> cd C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\build\x64-Debug (default)\src
C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\build\x64-Debug (default)\src>rtl_433.exe
Got the same error
rtl_433.exe – Entry Points Not Found
The procedure entry point SoapySDR_free could not be located in the dynamic link library

If I try to look at Manage Configurations, it will not open. Cannot view configuration.
[image]

zuckschwerdt, 2022-04-26:
Yes, "Deleted Cache Folders", then "Generate" and the output will show
```
1> -- SoapySDR device input disabled.
```
But above you are using "build\x64-Debug (default)\src". That is *not* the directory where you are building.
It would be "build\x64-Release\src". Check the file data/time. You are running an old binary.
Maybe just delete the whole "C:\Users\MySelf\CMakeBuilds" then rebuild to avoid confusion.

MaxFun, 2022-04-26:
4/26/22 1146hr
Deleted all C:\Users\MySelf\CMakeBuilds
Cashe(x64-Release Only)->Generate->rtl433
1> Command line: C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2017\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe  -G "Ninja" -DCMAKE_INSTALL_PREFIX:PATH="C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\install\x64-Release"  -DCMAKE_CXX_COMPILER="C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe"  -DCMAKE_C_COMPILER="C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe" -DENABLE_SOAPYSDR=OFF -DCMAKE_BUILD_TYPE="RelWithDebInfo" -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2017\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\Users\MySelf\Source\Repos\merbanan\rtl_433"
1> Working directory: C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\build\x64-Release
1> -- The C compiler identification is MSVC 19.16.27034.0
1> -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe
1> -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe -- works
1> -- Detecting C compiler ABI info
1> -- Detecting C compiler ABI info - done
1> -- Detecting C compile features
1> -- Detecting C compile features - done
1> -- Found Git version: refs/heads/master commit bb1be7f186ac0fdb7dc5d77693847d96fb95281e from 2022-04-25T11:00:23+0200
1> -- Using Git version tag: 21.12-100-gbb1be7f1 on master at 202204251100
1> -- IPv6 support enabled.
1> -- Looking for pthread.h
1> -- Looking for pthread.h - not found
1> -- Found Threads: TRUE  
1> -- Threads support will be compiled.
1> -- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR) 
1> -- OpenSSL development files not found, TLS won't be possible.
1> -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> -- Found LibRTLSDR: C:/Program Files/PothosSDR/lib/rtlsdr.lib  
1> -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
1> -- Found LibUSB: C:/Program Files/PothosSDR/lib/libusb-1.0.lib  
1> -- RTL-SDR device input will be compiled. Found version 
1> -- libusb-1.0 error messages are available. Found version 
1> -- SoapySDR device input disabled.
1> -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
1> -- Configuring done
1> -- Generating done
1> -- Build files have been written to: C:/Users/MySelf/CMakeBuilds/4f482441-8b72-4531-987a-ff95273c27bd/build/x64-Release
1> Starting CMake target info extraction ...
1> CMake server connection made.
1> Extracted includes paths.
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted global settings.
1> Extracted code model.
1> Extracted CTest info.
1> Collating data ...
1> Target info extraction done.
Cmake -> Build All
  [250/250] cmd.exe /C "cd . && "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E vs_link_exe --intdir=tests\CMakeFiles\data-test.dir --manifests  -- C:\PROGRA~2\MICROS~2\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\link.exe /nologo tests\CMakeFiles\data-test.dir\data-test.c.obj tests\CMakeFiles\data-test.dir\__\src\output_file.c.obj tests\CMakeFiles\data-test.dir\__\src\term_ctl.c.obj  /out:tests\data-test.exe /implib:tests\data-test.lib /pdb:tests\data-test.pdb /version:0.0  /machine:x64 /debug /INCREMENTAL /subsystem:console  src\data.lib ws2_32.lib mswsock.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."

Build succeeded.

Opened Command Prompt-> cd C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\build\x64-Release\src

Ran rtl_433.exe
C:\Users\MySelf\CMakeBuilds\4f482441-8b72-4531-987a-ff95273c27bd\build\x64-Release\src>rtl_433.exe

Received the following errors
[image]
[image]
[image]

zuckschwerdt, 2022-04-26:
The configure and build looks good. And we are back to the problem at the begining. Pothos didn't install correctly or you have changed some paths so the DLL is not found anymore.
Just copy the rtlsdr.dll from Pothos to "build\x64-Release\src".

And, huh, how did you manage to link to msys-2? That's part of Mingw-w64. You want to build with either MinGW or with MSVC not both! Something is mixed up there.

zuckschwerdt, 2022-04-26:
I'm out of ideas what to check. I don't have clue how to verify or set library paths on Windows or how MSYS got into the mix. @gvanem is an active Windows developer and @winterrace brought the vs15 build support. Maybe they can help?

MaxFun, 2022-04-26:
Thanks for the help. I will contact both to see if they have an idea.
I don't understand how VS decides what compiler to use. If I create a new project, I don't get the CMake tab, just a build tab.
Again Thanks