merbanan/rtl_433 #1917: trying to decode nexa 101lc and create plugin to rtl_433 - not quite sure what to do next...

cicci44, 2021-12-26:
hi, newbie here.. im used sdr# with rtl_433 plugin, got sample file g001_433.92M_1024k.cu8 while recording and holding the test-button. (really annoying learn object...) i put sample file into..
rtl_433.exe -A g001_433.92M_1024k.cu8
ang get two suggestion...
-X 'n=name,m=OOK_PWM,s=3252,l=33252,r=77756,g=5756,t=12000,y=0'
and
-X 'n=name,m=OOK_PWM,s=3468,l=8992,r=56,g=0,t=2208,y=0' ("Detected OOK package @6.278960s" but the sample file is 2sec ish.. so i ignore this one.)
and tried with 
`rtl_433.exe -A g001_433.92M_1024k.cu8 -X 'n=name,m=OOK_PWM,s=3252,l=33252,r=77756,g=5756,t=12000,y=0'`
seems to pic up the signal... great!

.. i use univeral radio hacker 
get this 
```
1111111111010010000100001000010010010010010010010010010000100100001000010010000100001001001001000010001 [Pause: 19377 samples]                                                                                         
sample/symbol = 800microsec
```
Now i want to put this into rtl_443 lib to get it to search for this signal and fetch to mqtt server before testing next one nexa 101lc..   

I got suggestion to run 
rtl_433.exe -w test2.ook g001_433.92M_1024k.cu8
and drop .ook into https://triq.org/pdv/
but.. not sure what i can do next...

zuckschwerdt, 2021-12-26:
If you recorded at 1024k, then with older rtl_433 you need to add `-s 1024k` always (e.g. when using `-A`).
So actual timings would be ` -X 'n=nexa101lc,m=OOK_PPM,s=1437,l=2812,y=8314,r=3500'`
That would give
```
codes     : {25}3805b18
```
Now press buttons and watch that code. Each device/button should have a unique code.
The Nexa 101LC is a smoke detector, right? That codes (25 bit) fits that. An alarm would be the same code as the button.

cicci44, 2021-12-26:
yep, its a firealarm. Tried to find any version nr on the rtl_433 i used before, but you were right, something with the old version (so i took rpi, recompile rtl and take git on repo... did run)
rtl_433 -f 433.88M -s 1024k -X 'n=nexa101lc,m=OOK_PPM,s=1437,l=2812,y=8314,r=3500'
and master published:
```
time      : 2021-12-26 17:30:16
model     : nexa101lc    count     : 1             num_rows  : 1             rows      :
len       : 25           data      : 3805b18
codes     : {25}3805b18
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-26 17:30:20
model     : nexa101lc    count     : 1             num_rows  : 1             rows      :
len       : 25           data      : 3805b18
codes     : {25}3805b18

```
tried on slave and it seems to publish the same code.
```
time      : 2021-12-26 17:34:43
model     : nexa101lc    count     : 1             num_rows  : 1             rows      :
len       : 25           data      : 3805b18
codes     : {25}3805b18
_
```
What can i do now? :S

zuckschwerdt, 2021-12-26:
Look at some sample conf files in https://github.com/merbanan/rtl_433/tree/master/conf
You want to create a file with your timings and a `match=` with your code. Then the conf will output just for that device.
Create one conf (or multiple sections in one file) with all your devices and load with `-c`

cicci44, 2021-12-26:
did run 
rtl_433 -f 433.88M -s 1024k -c nexa101lc.conf
seems to work fine with master and slave test-buttons, so i guess when i pressed "learn" for like half a year ago it just send out this random value for every slave in sniffing state... 
Time for define values as "testalarm","Tx/Rx number".. possible from .conf ?

```
# Nexa 101LC
# Firealarm, this is test-button
# This is just testing, have no idea if it will work on other version of Nexa 101LC.
# 
# rtl_433 -f 433.88M -s 1024k -X 'n=nexa101lc,m=OOK_PPM,s=1437,l=2812,y=8314,r=3500'
# more info of cmd like above https://github.com/merbanan/rtl_433#running search "spec"
decoder {
    n=Nexa_101LC,
    m=OOK_PPM,
    s=1437,
    l=2812,
	y=8314,
    r=3500,
    match={25}3805b18
}
```

zuckschwerdt, 2021-12-26:
My "networked" sensors also copy the ID. I have unlearned/reset them all so I get a unique code for each ;)

If all your codes are `3805b18` then you can't have test/alarm/... from that.
Otherwise do e.g.
```
decoder {
    n=MyNexa_A,
...
    match={25}3805b18
}
decoder {
    n=MyNexa_B,
...
    match={25}abc1238,
}
...
```

cicci44, 2021-12-26:
ah, i should do that to! then i can know where signal came from! Working with reseting the capturing made both loose the connection to activate each other, however, seems like it recognize the the signal and respont but wont set the alarm and vice versa. I guess this two codes are for each of 

this is when slave is reset, master still got group-id set.
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-26 20:05:59
model     : nexa101lc    count     : 1             num_rows  : 1             rows      :
len       : 25           data      : 49c6028
codes     : {25}49c6028
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-26 20:06:03
model     : nexa101lc    count     : 1             num_rows  : 1             rows      :
len       : 24           data      : 938c05
codes     : {24}938c05

```

this is when master and slave are bot reset. (i guess)
```

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-12-26 20:12:59
model     : nexa101lc    count     : 1             num_rows  : 1             rows      :
len       : 24           data      : 938c05
codes     : {24}938c05
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-26 20:14:52
model     : nexa101lc    count     : 1             num_rows  : 1             rows      :
len       : 25           data      : 3805b18
codes     : {25}3805b18
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

```

.. i could just add both separate into the .conf as you explain.. do i need to decode this? i could just take the value and push to mqtt via rtl433 (i guess)

zuckschwerdt, 2021-12-26:
Lots of other devices (doorbell, ...) have 25-bit codes. Without `match=` you will likely get false positives. I send every code (unfiltered) to MQTT, but I filter my own codes when using the MQTT data (display, email, ...)

cicci44, 2021-12-26:
> 

i think i can work with `match=` for each of one in one .conf. For the moment i got -R 36 up and running pushing data to mqtt, i could move frequency closer to get both, that will work. Then it should push the unfiltered data to mqtt. yes. 

using -c combined with -R shouldnt affect the readings from the -R36 stream?

zuckschwerdt, 2021-12-26:
Both -c and -R together are fine. You can also put everthing in one file, look at https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

closed 2021-12-27 by cicci44