merbanan/rtl_433 #1919: Add hostname tag for InfluxDB output

pfassberg, 2021-12-28:
For MQTT output the hostname is always added in the topic.
For InfluxDB output the hostname can't be added.  The only way to distinguish between two receivers is to use a separate database/organization.

I suggest to add hostname (or other configurable information) as a tag for InfluxDB.

zuckschwerdt, 2021-12-28:
For MQTT the topic is configurable, but yes, by default the hostname is used ;)

Does the token tagging feature help for InfluxDB?
```
[-K FILE | PATH | <tag> | <key>=<tag>] Add an expanded token or fixed tag to every output line.
```

pfassberg, 2021-12-28:
Yes, it does!

However, this adds a field to InfluxDB and not a tag. Extremely much better than nothing but I think it's more correct to send this as a tag.

Thank you!

zuckschwerdt, 2021-12-28:
Yes, there is no clear distinction which tokens are actual fixed tags or variable data, that feature is used for gps data also.
Our data layer only knows "strings", not label vs. variable data. Maybe we could add that some day.

closed 2021-12-28 by pfassberg