merbanan/rtl_433 #1921: Sonoff wall switch T2EU3C-RF

tgeorgiop, 2021-12-28:
Hello, I run rtl_433 for the wall switch T2EU3C-RF. 

1. When I press the buttons with `rtl_433` I see no response.
2. When I press the buttons with `rtl-433 -G 4` shows 
[image]

I would like to publish mqtt decoded messages to home assistant broker. Is this possible?

zuckschwerdt, 2021-12-28:
https://triq.org/rtl_433/ANALYZE.html to get started. A few examples for simple push buttons are in our conf directory.
If you like, post a working and documented .conf here or by PR.

tgeorgiop, 2021-12-29:
Thanks for the answer.

- I ran `rtl_433 -S unknown` and the 3 signals were saved.
- I analyzed 1 of them with `rtl_433 -A g001_433.92M_250k.cs16` and the output was
[image]

- I ran rtl_433 with the suggested decoder `rtl_433 'n=sonoff_T2EU3C-RF,m=OOK_PWM,s=188,l=588,r=6212,g=724,t=160,y=0'` and the output was
[image]

- I created a conf file 

> decoder {
    n=sonoff_T2EU3C_RF,
    m=OOK_PWM,
    s=188,
    l=588,
    y=6212,
    g=724,
    t=160,
    y=0,
    match={25}11e2f78
}
and I got the error "Bad flex spec, missing reset limit!". So, I entered the r with value 700 (bigger than l). Is it correct? Now, I see the signals from 1 button, which seems reasonable. I will enter 2 decoders more in the config file.

Now, I run `rtl_433 -c sonoff_T2EU3C-RF.conf -F "mqtt://192.168.2.178:1883,user=user,pass=pass,events=rtl_433"` and I get the messages in the mqtt broker. 
[image]

Are they useful now? Can I decode them more? I see responses of other sensors and the format is very different.

zuckschwerdt, 2021-12-29:
Change the suggestion to:
- s=200,
- l=600,
- g=1000,
- r=7000,
- no y= or t=

Those 25 bit (actually 24 plus sync bit) codes are the very common EV1527. It's just a unique number.

tgeorgiop, 2021-12-29:
Thank you. Now, with `rtl_433 'n=sonoff_T2EU3C-RF,m=OOK_PWM,s=200,l=600,r=7000,g=1000'` the response is
[image]

zuckschwerdt, 2021-12-29:
You can add `unique` or `countonly` to get less data. Otherwise you are done!

tgeorgiop, 2021-12-29:
Thank you very much! Can this button be supported by rtl_433 natively, without running the decoder?

zuckschwerdt, 2021-12-29:
That's pretty much native support. Too much false positives if we add EV1527 decoders. The `match=` you use is essential.

closed 2021-12-29 by zuckschwerdt