merbanan/rtl_433 #1923: Question regarding pulse_data_t template - specifically what type/range of data values to load

keith721, 2021-12-30:
I'm adapting small portions of some rtl_433 source code for an ESPHome custom component, specifically for capturing and decoding data from the CoTech 36-7959 / SainLogic WS019T weather station.   I have been able to confirm data reception from the station using the Windows build of rtl_433 and a USB NESDR radio.

```
verbose
device	0
frequency	433.92M
protocol 153
output csv
```

Currently, the ESPHome C code is capturing the most recent four micros( ) microsecond counters when OOK pulses begin and end.  
Pulse length = usec[3] - usec[2] (these are more varying, but definite short pulses near 508 usec and long pulses near 996 usec.
Gap length = usec[2] - usec[1]   (these are rather consistent, near 5364 usec)

Using the pulse_demod_manchester_zerobit( ) routine, my captured pulse data does not appear to be interpreted correctly.  The CRC calculation does not match with data byte 13 from the weather station.  While I definitely don't expect anyone to mystically deduce what's not correct, I'd like to better understand what type or range of values are expected in the pulses template/structure in the arrays of gaps[ ] and pulses[ ].  Should these be microsecond deltas/lengths/measures, as shown above, or a literal/absolute microsecond counter value, increasing for each subsequent pulse?    I'm currently placing delta gap and pulse values (tried both deltas/measures and absolutes) in the arrays.

Thank you in advance for any insight you can provide.  I've looked through the docs folder, but seem to have missed this part, if it's there.

zuckschwerdt, 2021-12-30:
Manchester is both pulse and gap at half- or bit width. So 500 or 1000 µs here, 5464 is the packet gap.
Write your data to an .ook file and visualize on https://triq.org/pdv/ then compare with our cotech samples.

keith721, 2021-12-30:
Thank you for your prompt suggestion.  I was unable to find any description anywhere for an OOK file format. I do have a CSV text file of captured samples, including the following fields:  HH:MI:SS, N, MICROS[1], MICROS[2], MICROS[3], GAP, PULSE .  Dropping this CSV file onto https://triq.org/pdv/ displays a chart with no method to further expand/zoom or interpret the data. I am attaching the CSV file here.  If a different format or sequence of data in CSV format is better, please advise. 
[WS019T Pulse and Gap Data.csv](https://github.com/merbanan/rtl_433/files/7793328/WS019T.Pulse.and.Gap.Data.csv)

zuckschwerdt, 2021-12-30:
That doc is rendered at https://triq.org/rtl_433/IQ_FORMATS.html#pulse-data-formats

Best to create an .ook with rtl_433 and take that as template :)
`rtl_433 -w FILE.ook FILE.cu8`

keith721, 2021-12-31:
I'm happy to say I took a step back today, brought the edge detect / ISR routine down to the most minimal possible code, and determined my understanding and logic were incorrect, and my code was just a tiny bit sloppy with recording microsecond timestamps. After correcting both of these issues, I'm seeing more coherent data this evening.  I also figured out the two-finger mousepad motions to move/drag the bit data graphic, and also to expand it. 👍 Thank you for your assistance.  I'll let you know how it progresses.

keith721, 2022-01-02:
Unfortunately, using an SRX882 receiver and an RXB6 receiver did not produce satisfactory results, no matter how high or low I adjusted the device tolerance.  Reviewing the RF pulse/gap capture code, I opted to order an RXB12 receiver instead.  It will require a few weeks to arrive, however.

closed 2022-01-02 by keith721

zuckschwerdt, 2022-01-02:
Ah ok, those are those ultra-cheap 1-bit superheterodyne receviers. You will need strong filtering specifically tuned to the protocol you are looking for. Basically you want to low-pass filter the signal to guess if the expected OOK might be present.

The biggest problem with 1-bit receivers is the lack of signal strength report and thus framing. They will output random noise if no carrier is present.

I would suggest something like a CC1101. There is an option (async mode iirc) to get raw bit-banging with a single pin. The chip needs a setup via I2C though.

zuckschwerdt, 2022-01-02:
Also #1927 is looking for something similar.

keith721, 2022-01-02:
> The biggest problem with 1-bit receivers is the lack of signal strength report and thus framing. They will output random noise if no carrier is present.
I agree.  When I could not determine where/why there was an issue, I began reviewing those frequent receiver interrupts that were not fitting the transmission profile.  All of the gap, idle, and pulse values were returning the same constant value, which I determined was probably simply the noise floor of the inexpensive receivers.

> I would suggest something like a CC1101. There is an option (async mode iirc) to get raw bit-banging with a single pin. The chip needs a setup via I2C though.
I'll test the RXB12 receiver first, and if it generates similar disappointing results, upgrade to the CC1101.  
Thank you again for your kind assistance and very helpful and informative responses!