merbanan/rtl_433 #1927: Alternative Chip

garudaonekh, 2021-12-31:
Hi, 
I am working on small scale agriculture project using RPI zero. SDR_RTL is more than 20$ which is nearly h
the same price as RPI zero. I try to reduce budget as it is for farmers.
Can we use cheaper chip like CC1101 or others?

zuckschwerdt, 2021-12-31:
#769 -- note that at least the cc1101 already demodulates data and can output message based data
#940 -- feed ook data to `rtl_443 -r OOK:-`

garudaonekh, 2021-12-31:
@zuckschwerdt thanks for your quick response,
Before I post this issue, I saw the above two issues which mentioning CC1101 but I donot understand them

zuckschwerdt, 2021-12-31:
The CC1101 (and similar) generally only work for one specific protocol. You need to configure that protocol, but then you don't need rtl_433 (and you don't need a SoC, a MCU will do)

What radio protocols are you expecting to receive?

My advice:
- ESP32 + CC1101: cheap, low power demand, needs good programming skills
- Raspi + RTL-SDR: not-so-cheap, high power demand, existing solutions ready to use

garudaonekh, 2022-01-01:
[32]  Fine Offset Electronics WH1080 and [195] rainPoint moisture sensor

zuckschwerdt, 2022-01-01:
- RainPoint is OOK PCM at 500 µs rate (half-bit rate as it's Manchester) and should work well with a CC1101-like.
- Fine Offset WH1080 (protocol 32) is OOK PWM, short pulse 544µs, long pulse 1524µs, fixed gap 1036µs, which might work as ~500 µs PCM with a custom 3/5-to-1 decoding.

So, yeah CC1101 could work if you put the coding in. rtl_433 won't be needed.

keith721, 2022-01-02:
While waiting for a CC1101 module to arrive (a few weeks wait time, unfortunately) I happened upon a faster method for obtaining a reliable MICRF211 433 MHz receiver module: harvest one from [an AcuRite 06002 weather sensor](https://www.amazon.com/gp/product/B00T0K8NXC/). It's a 3.3VDC intelligent receiver module.  [Data can be read](http://www.techspin.info/archives/985) either using an RPi with Python code, or an ESP8266 with C code or Arduino using a C sketch/code.

zuckschwerdt, 2022-01-03:
Something mass-produced down to cost, readily available! True hacker spirit.
But from the sensor really? Is there a MICRF211 *receiver* in there?

keith721, 2022-01-03:
> Something mass-produced down to cost, readily available! True hacker spirit. But from the sensor really? Is there a MICRF211 _receiver_ in there?

:( You are right, I jumped too soon.  It's the MICR112 **transmitter** in the inexpensive sensor.  The MICR211 **receiver** is in the base station.   Oh well . . .

closed 2022-07-07 by zuckschwerdt