merbanan/rtl_433 #1931: Multiple sensors of the same type--Newbie question

garudaonekh, 2022-01-03:
It's a newbie question!

We'll need to use up to 10 to 20 soil moisture sensor of the same type for a big greenhouse(Rainpoint).
Is it possible for a single receiver to distinguish each of them thru some ID or any mean?
In the Rainpoint, there're 3 channel to switch(their LCD can only display 3 devices).

zuckschwerdt, 2022-01-03:
We don't know. There is a maybe a unique ID. Look at the "Sync" and "ID" output.
We have seen "AA81" for sync and "FE03", ""F703" for ID so far. The ID might be fixed and unique, or reset when you change batteries, or something else and not useful.
Some info is in #1781

garudaonekh, 2022-01-21:
I just ordered another 2 sensors so let wait and see when I use all three together

garudaonekh, 2022-01-28:
Ok, I got two soil moisture sensors and set them both to channel 3(It has three channel to choose). It look like each of them have a difference id but I am not sure how unique they are because, it's only 4 digit id.
Here's what I received:

```
[00] {370} 92 49 24 e4 e7 39 ce 72 49 ce 4e 4e 49 24 9c e4 9c e4 e7 39 ce 73 9c e7 39 ce 73 9c e7 39 ce 73 9c e7 39 c9 ce 72 73 9c e7 24 e7 24 92 72 00 
Exact bit width (in us) is 490.00 vs 500.00, 28 bit preamble
Exact bit width (in us) is 488.00 vs 500.00, 28 bit preamble
Exact bit width (in us) is 489.43 vs 500.00, 28 bit preamble
rainpoint_decode: {96} aa 81 03 03 b7 04 08 1c 60 00 54 15 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-28 22:33:43
model     : RainPoint-Soil                         id        : 0303
channel   : 3            Sync?     : aa81          Flags?    : b7            Status?   : 0408          Temperature: 28.0 C       Moisture  : 96 %          Integrity : CHECKSUM
pulse_demod_pcm(): RainPoint soil temperature and moisture sensor
bitbuffer:: Number of rows: 1 
[00] {2772} 55 55 55 54 cc cd 2a ad 4a ab 4a ab 53 4c b2 aa ac aa b5 2a aa d2 aa aa b3 33 33 2a ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff fd aa aa aa aa 66 66 95 56 a5 55 a5 55 a9 a6 59 55 56 55 5a 95 55 69 55 55 59 99 99 95 7f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff fe d5 55 55 55 33 33 4a ab 52 aa d2 aa d4 d3 2c aa ab 2a ad 4a aa b4 aa aa ac cc cc ca 80 
RainPoint soil temperature and moisture sensor: Exact bit width (in us) is 492.40 vs 500.00, 201 bit measured
pulse_demod_pcm(): RainPoint soil temperature and moisture sensor
bitbuffer:: Number of rows: 1 
[00] {378} 92 49 24 e4 e7 39 ce 72 49 ce 4e 4e 49 24 e7 39 27 39 39 ce 73 9c e7 39 ce 73 9c e7 39 ce 73 9c e7 39 ce 72 73 9c 9c e7 39 c9 39 39 c9 c9 ce 00 
RainPoint soil temperature and moisture sensor: Exact bit width (in us) is 492.24 vs 500.00, 201 bit measured
pulse_demod_pcm(): RainPoint soil temperature and moisture sensor
bitbuffer:: Number of rows: 1 
[00] {378} 92 49 24 e4 e7 39 ce 72 49 ce 4e 4e 49 24 e7 39 27 39 39 ce 73 9c e7 39 ce 73 9c e7 39 ce 73 9c e7 39 ce 72 73 9c 9c e7 39 c9 39 39 c9 c9 ce 00 
RainPoint soil temperature and moisture sensor: Exact bit width (in us) is 492.22 vs 500.00, 201 bit measured
pulse_demod_pcm(): RainPoint soil temperature and moisture sensor
bitbuffer:: Number of rows: 1 
[00] {378} 92 49 24 e4 e7 39 ce 72 49 ce 4e 4e 49 24 e7 39 27 39 39 ce 73 9c e7 39 ce 73 9c e7 39 ce 73 9c e7 39 ce 72 73 9c 9c e7 39 c9 39 39 c9 c9 ce 00 
Exact bit width (in us) is 490.86 vs 500.00, 28 bit preamble
Exact bit width (in us) is 490.14 vs 500.00, 28 bit preamble
Exact bit width (in us) is 492.00 vs 500.00, 28 bit preamble
Exact bit width (in us) is 490.43 vs 500.00, 28 bit preamble
Exact bit width (in us) is 490.14 vs 500.00, 28 bit preamble
Exact bit width (in us) is 490.14 vs 500.00, 28 bit preamble
rainpoint_decode: {96} aa 81 06 03 b7 04 00 1c 08 00 51 15 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-28 22:35:32
model     : RainPoint-Soil                         id        : 0603
channel   : 3            Sync?     : aa81          Flags?    : b7            Status?   : 0400          Temperature: 28.0 C       Moisture  : 8 %           Integrity : CHECKSUM
pulse_demod_pcm(): RainPoint soil temperature and moisture sensor
bitbuffer:: Number of rows: 1 
[00] {2780} 55 55 55 54 cc cd 2a ac d2 ab 4a ab 53 4c b2 aa aa aa b5 2a ac aa aa ab 2b 33 33 2a ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff da aa aa aa a6 66 69 55 66 95 5a 55 5a 9a 65 95 55 55 55 a9 55 65 55 55 59 59 99 99 57 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff fe d5 55 55 55 33 33 4a ab 34 aa d2 aa d4 d3 2c aa aa aa ad 4a ab 2a aa aa ca cc cc ca 80 
```

zuckschwerdt, 2022-01-28:
Thanks! So `Sync` is really sync or device type. 
For ID we have seen "FE03", ""F703" and now "0303", "0603". 
That could be 8 bit of ID ("FE", ""F7" and "03", "06"), but it looks more like the first 4 bit are fixed and mean something else? So actually more like a "1 digit ID".

Do you get different values when you change batteries?

garudaonekh, 2022-01-29:
> Thanks! So `Sync` is really sync or device type. For ID we have seen "FE03", ""F703" and now "0303", "0603". That could be 8 bit of ID ("FE", ""F7" and "03", "06"), but it looks more like the first [...]

not changed when I change the battery.

zuckschwerdt, 2022-05-31:
Usually the sensor are not capable of receiving data -- pairing is just learning the sensor IDs in the head unit.
The Rainpoint protocol we know isn't that complicated: https://github.com/merbanan/rtl_433/blob/master/src/devices/rainpoint.c#L25-L40

tipofthesowrd, 2022-05-31:
With rtl_433 -R 195:vv -g 37 I'm getting some values, however very far from the required # bits.
SNR doesn't seem to be the issue. I'm picking the TPMS values from my neighbours car with SNR over 12 dB

> 
> rainpoint_decode: bit_per_row 76 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] {76} 92 49 24 92 49 24 92 49 24 80
> rainpoint_decode: bit_per_row 7 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 7} 90        : 1001000
> rainpoint_decode: bit_per_row 4 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 4} 80        : 1000
> rainpoint_decode: bit_per_row 4 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 4} 80        : 1000
> rainpoint_decode: bit_per_row 4 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 4} 80        : 1000
> rainpoint_decode: bit_per_row 7 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 7} 90        : 1001000
> rainpoint_decode: bit_per_row 4 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 4} 80        : 1000
> rainpoint_decode: bit_per_row 4 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 4} 80        : 1000
> rainpoint_decode: bit_per_row 7 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 7} 90        : 1001000
> rainpoint_decode: bit_per_row 4 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] { 4} 80        : 1000
> rainpoint_decode: bit_per_row 76 out of range
> pulse_slicer_pcm(): RainPoint soil temperature and moisture sensor
> bitbuffer:: Number of rows: 1
> [00] {76} 92 49 24 92 49 24 92 49 24 80

zuckschwerdt, 2022-05-31:
If you are not sure about the transmission first try to grab a sample and verify the modulation and timings: https://triq.org/rtl_433/ANALYZE.html
If that all fits then gather codes to pick apart.

garudaonekh, 2022-06-04:
> I've a new Rainpoint sensor (labeled TCS005FRF. Which, according the info online should transmit on 433.9 MHz.
> 
> However, I'm getting no useful info from the device. Also not when inserting the batteries
> 
> Does it need to be linked to the gateway before it starts transmitting values?

> I've a new Rainpoint sensor (labeled TCS005FRF. Which, according the info online should transmit on 433.9 MHz.
> 
> However, I'm getting no useful info from the device. Also not when inserting the batteries
> 
> Does it need to be linked to the gateway before it starts transmitting values?

I do not think it is required to connect to its display panel. You may try to switch among the three channel

tipofthesowrd, 2022-06-04:
How can I switch channels?
So far I've been unable to pickup anything, also not in gqrx or sdr#.

garudaonekh, 2022-06-06:
There's a black switch near battery. it has 3 positions. Its LCD display support maximum 3 soil sensors that's why this sensor has a 3 channel switch.

tipofthesowrd, 2022-06-06:
FWIW in my unit it's not a switch but a push button. When it's pushed for a
few seconds a red light appears on the front for a few seconds

garudaonekh, 2022-06-06:
[image]

pswiatki, 2022-06-06:
Low resolution picture, undiscernable. But this topic is quite interesting for me. Let me share my experience with similar sensors (TH, as in temperature / humidity). Got an Auriol weather station (sold by LIDL in Polska, where I live). Three remote TH transmitters were included in the kit together with the base station. I placed those around the house and monitored the parameters with `rtl_433`.
Also, just to have a visual status on them - I gave base station channels meaningful names to match the locations of my sensors. All worked great. However, a year or so passed and I decided to get another such set to use in other locations of interest inside the building. This is where things got screwy. Base stations are practically useless - I never know what they show. My guess is the software only checks the channel, but not the ID of the remote sensor. Now it is **only** `rtl_433` that registers all six of the sensors and I can analyze historical data of how the temperature and humidity change here in the house. I can clearly see channel numbers repeated twice each time all sensors are picked up by the RTL receiver, but six unique ID values.

pswiatki, 2022-06-07:
This is the confusion of the base station:

[image]

garudaonekh, 2022-06-07:
> This is the confusion of the base station:
> 
> [image]
> 
> All sensors in range, naturally.

Is there a binding mechanism? For example, I have 6 sensors with 2 LCD display panel and within each other range so it should be binding to separate LCD gateway(3 each).

pswiatki, 2022-06-07:
I guess the makers did not think through the concept well enough. I read reports of people's base station picking up sensors from their neighbours (who happened to get the same model - possibly from the same source, too).

tipofthesowrd, 2022-06-08:
Mine is a pushbutton iso a switch  to select a channel.
Pressing the button does not give any indication there is any activity on  433 MHz.
Upto now no values found in RTL_433

[image]

anthyz, 2022-06-08:
I have the sensor labeled TCS005ARF which has the channel switch near the batteries and is decoded by rtl_433.  On the box it came in the RF frequency spec is, “433.92MHz ASK.”  According to the manual for the TCS005FRF model, the button by the battery is used to connect the sensor to their wifi water timer, and the RF frequency spec is “433.92MHz FSK.”  Unless the docs are wrong, rtl_433 needs a new decoder for the FSK version of the sensor.

merbanan, 2022-06-08:
Record some samples and we can probably figure out if that is needed.

tipofthesowrd, 2022-06-10:
I'm trying to record some samples, but as before I'm not seeing any unknown frames passing. Also not aftering removing & entering the batteries or pressing the buttons to report levels via de LEDs. I'll leave it running for some more time to see if anything shows up.

My box also indicated TCS005FRF (double way version) however the frequency is reported as 433.92 MHz ASK (which I doubt). What worries me is the 'double way'

```
 rtl_433 -S unknown -Y autolevel -Y magest -M noise -M level
rtl_433 version 21.12-103-g8cdb233b branch master at 202205181328 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 186 out of 218 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-218 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Estimated noise level is -17.5 dB, adjusting minimum detection level to -14.5 dB
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Estimated noise level is -19.6 dB, adjusting minimum detection level to -16.6 dB
Estimated noise level is -21.4 dB, adjusting minimum detection level to -18.4 dB
Estimated noise level is -23.0 dB, adjusting minimum detection level to -20.0 dB
Estimated noise level is -24.4 dB, adjusting minimum detection level to -21.4 dB
Estimated noise level is -25.6 dB, adjusting minimum detection level to -22.6 dB
Estimated noise level is -26.6 dB, adjusting minimum detection level to -23.6 dB
Estimated noise level is -28.4 dB, adjusting minimum detection level to -25.4 dB
Estimated noise level is -29.7 dB, adjusting minimum detection level to -26.7 dB
Estimated noise level is -30.7 dB, adjusting minimum detection level to -27.7 dB
Current noise level -34.1 dB, estimated noise -30.7 dB
Estimated noise level is -31.9 dB, adjusting minimum detection level to -28.9 dB
Estimated noise level is -32.9 dB, adjusting minimum detection level to -29.9 dB
Current noise level -34.1 dB, estimated noise -33.8 dB
Estimated noise level is -33.9 dB, adjusting minimum detection level to -30.9 dB
Current noise level -34.1 dB, estimated noise -34.1 dB
Current noise level -34.1 dB, estimated noise -34.1 dB
Current noise level -34.1 dB, estimated noise -34.1 dB
Current noise level -34.1 dB, estimated noise -34.1 dB
Current noise level -34.1 dB, estimated noise -34.1 dB
Current noise level -34.1 dB, estimated noise -34.1 dB
Current noise level -34.1 dB, estimated noise -34.1 dB
Current noise level -34.2 dB, estimated noise -34.2 dB
```

garudaonekh, 2022-06-24:
> after capturing again some transmission I do think this is the correct device. I'm only picking up this transmission when the rainpoint is directly next to the SDR.
> 
> The PDV analysis guesses it (with FSK) as PWM modulated. Any hints on how to proceed?
> 
> [g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/8965215/g001_433.92M_250k.zip) e

How far apart?

closed 2022-10-12 by garudaonekh

tcwatson, 2023-08-22:
@tipofthesowrd  did you ever get this to work? I've just purchased the same TCS005FRF you have. I have openmqttgateway running on a lilygo_rtl_433, and while I can see the mqtt messages it publishes from the 433MHz messages from my weather station, nothing has been published for the soil sensor. Not sure if I just need a newer version of rtl_433 on my board, or if there's some initialization sequence with the device, or whether it works for anyone.

tipofthesowrd, 2023-09-02:
Unfortunately, no I have never made any progress in trying to get this to
work. I believe some sort of 2way communcation is needed since I never saw
any RF messages leaving the device.

RodEzIO, 2025-04-22:
so is it possible to read it with rtl433 ? I have just bought one and not able to see it communicating....