merbanan/rtl_433 #1933: Filter temperature sensor down to ID

binkenney94, 2022-01-07:
Hello,

I love rtl_433!!

My question is I have two Acurite temp and humidity sensor units and both are registering in rtl_433. I got it filtered down to just these two sensors by using the -R 40 flag. How can I filter it down further to just the ID since I only want to view one of the two sensors? What flags do I use, or is it implemented in rtl_433 yet?

Many thanks!

merbanan, 2022-01-08:
Hi, that is not possible with rtl_433. But piping to grep might work well enough.

binkenney94, 2022-01-08:
Hi Ben, thank you for the information. Once I learn to program, I might w to this project to lake such thing possible. I don’t think I’m the only one with this request. Thanks.

Bin Kenney
The Earth Stands in 3rd Place

zuckschwerdt, 2022-01-08:
Sounds like you want a steady display of the current values. Add `-F http` to the command, then browse to http://127.0.0.1:8433/ (or even http://YOURCOMPUTERNAME:8433/ from another device in your home network).
Also look at https://triq.org/rtl_433/INTEGRATION.html

closed 2022-01-08 by zuckschwerdt

mikethebee, 2022-01-08:
Hi Bin, I use Linux and run a script from cron that saves the "rtl_433 -F json -T 360 -E quit ..." output to a file and then rtl_433 exits. I then grep the file to filter each sensor ID result and forward them to a Grafana dashboard for display. This allow sanity checking of the collected output and re-running if rtl_433 timed-out before collecting any data. An "ID=" filter within rtl_433 would be handy, but with so many sensors able to be collected it would only be of use in a few. I like that it focuses on the core job of decoding. 

I have used a pipe to grep directly on the output, but stopped when I saw a comment from the developers suggesting it might be unreliable on some OS's.

HTH, Mike

zuckschwerdt, 2022-01-08:
Mike, you can also filter and process the data in real-time, see https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_custom.py -- no intermediate file or 6 minute delay.
If forwarding to Grafana involves a database then you can perhaps write to the database more efficiently, see the other example scripts and https://triq.net/articles/rtl-433-networking

mikethebee, 2022-01-08:
Thanks zuck, good tip, I will look at that python script; I am still learning all the great features of rtl_433.
As I am sampling outside and inside weather sensors and loft temperature data using a raspberrypi I have plenty of time for other jobs. I switch the RTL-SDR to monitor 868, 433 and RadioSonde at different times (RadioSonde only fly twice a day) so it is convenient to stop and restart rtl_433. I do have to reset the USB sometimes and I do that with python3.
-Mike