merbanan/rtl_433 #1936: Sensors interfering with each other?

tannercollin, 2022-01-08:
First off, amazing project. It's rare to stumble across software that does exactly what you need at the time.

I have three identical Acurite 606TX temperature sensors with IDs 35, 62, and 231. They were all reading perfectly fine until last night:

```
{"time" : "2022-01-08 10:38:47", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.400, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:39:18", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.500, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:39:49", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.500, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:40:20", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.500, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:40:51", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.500, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 10:41:22", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.500, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:41:48", "model" : "Acurite-606TX", "id" : 231, "battery_ok" : 1, "temperature_C" : -20.800, "mic" : "CHECKSUM"}                                                      
{"time" : "2022-01-08 10:41:53", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.500, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:42:19", "model" : "Acurite-606TX", "id" : 231, "battery_ok" : 1, "temperature_C" : -20.800, "mic" : "CHECKSUM"}                                                      
{"time" : "2022-01-08 10:42:24", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.600, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:42:50", "model" : "Acurite-606TX", "id" : 231, "battery_ok" : 1, "temperature_C" : -20.800, "mic" : "CHECKSUM"}                                                      
{"time" : "2022-01-08 10:42:55", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.600, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 10:43:21", "model" : "Acurite-606TX", "id" : 231, "battery_ok" : 1, "temperature_C" : -20.800, "mic" : "CHECKSUM"}                                                      
{"time" : "2022-01-08 10:43:57", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}                                                             
{"time" : "2022-01-08 10:44:23", "model" : "Acurite-606TX", "id" : 62, "battery_ok" : 1, "temperature_C" : 17.500, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 10:44:28", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:44:54", "model" : "Acurite-606TX", "id" : 62, "battery_ok" : 1, "temperature_C" : 17.500, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:44:59", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:45:25", "model" : "Acurite-606TX", "id" : 62, "battery_ok" : 1, "temperature_C" : 17.500, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:45:30", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:45:56", "model" : "Acurite-606TX", "id" : 62, "battery_ok" : 1, "temperature_C" : 17.500, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:46:01", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:46:27", "model" : "Acurite-606TX", "id" : 62, "battery_ok" : 1, "temperature_C" : 17.500, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:46:32", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:46:58", "model" : "Acurite-606TX", "id" : 62, "battery_ok" : 1, "temperature_C" : 17.500, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 10:47:03", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}
```

Connection from 62 and 231 became spotty and eventually stopped completely:

```
{"time" : "2022-01-08 11:10:18", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 11:10:49", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 11:11:20", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 11:11:51", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 11:12:22", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 11:12:53", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 11:13:24", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2022-01-08 11:13:55", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}                                              
```

I tried moving 62 right beside the rlt-sdr this afternoon, but it still wouldn't show up.

Next I removed and reconnected 62's battery, and its ID changed to 20 and all three starting working instantly:

```
{"time" : "2022-01-08 22:11:08", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.300, "mic" : "CHECKSUM"}                                                        
{"time" : "2022-01-08 22:11:39", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.300, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 22:12:10", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.300, "mic" : "CHECKSUM"}                                                        
* battery reconnected *
{"time" : "2022-01-08 22:12:14", "model" : "Acurite-606TX", "id" : 20, "battery_ok" : 1, "temperature_C" : 19.300, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 22:12:36", "model" : "Acurite-606TX", "id" : 231, "battery_ok" : 1, "temperature_C" : -15.400, "mic" : "CHECKSUM"}                                                      
{"time" : "2022-01-08 22:12:41", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.300, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 22:12:45", "model" : "Acurite-606TX", "id" : 20, "battery_ok" : 1, "temperature_C" : 19.700, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 22:13:07", "model" : "Acurite-606TX", "id" : 231, "battery_ok" : 1, "temperature_C" : -15.400, "mic" : "CHECKSUM"}                                                     
{"time" : "2022-01-08 22:13:12", "model" : "Acurite-606TX", "id" : 35, "battery_ok" : 1, "temperature_C" : 19.200, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 22:13:16", "model" : "Acurite-606TX", "id" : 20, "battery_ok" : 1, "temperature_C" : 19.900, "mic" : "CHECKSUM"}                                                       
{"time" : "2022-01-08 22:13:38", "model" : "Acurite-606TX", "id" : 231, "battery_ok" : 1, "temperature_C" : -15.400, "mic" : "CHECKSUM"}                                                     
... repeats
```

How could power cycling 62 fix 231's reception? Were the sensors interfering with each other in some way? Is there a way to prevent this or is it inevitable with clock drift? Thanks!

zuckschwerdt, 2022-01-08:
Could be overlapping sends, but that should also drift apart again quickly (10 minutes?).
Maybe the signal went out of band (thermal drift in the frequencies).
See https://triq.org/rtl_433/ANALYZE.html

tannercollin, 2022-01-08:
Thanks I'll look into this. Also I'm getting a 404 Not Found for the https://triq.net/pdv tool linked by that page.

zuckschwerdt, 2022-01-09:
Thanks. Fixed!

zuckschwerdt, 2022-01-09:
If you are comfortable compiling a new version from a feature branch, then try #1915 and when this happens again connect Gqrx to rtl_433  to inspect the signal. (or use https://github.com/pothosware/SoapyRTLTCP with e.g. SigDigger, CubicSDR, ...)

tannercollin, 2022-01-16:
The problem occurred again and I grabbed a sample of it with `-S unknown`. I can't make heads or tails of it though.

Any tips on next steps?

[g005_433M_250k.cu8.tar.gz](https://github.com/merbanan/rtl_433/files/7876538/g005_433M_250k.cu8.tar.gz)

zuckschwerdt, 2022-01-16:
Yes, that's two sensors exactly overlapping. The frequency is almost exactly the same, barely visible. There is nothing we can do here.

[image]

tannercollin, 2022-01-16:
Wow, good eye seeing that overlap.

I'm surprised they stay synced for so long considering one is outside, and there's a 20 C temperature difference.

I guess I need to buy different sensor models instead? Or manufacturing batches?

closed 2022-01-16 by tannercollin

merbanan, 2022-01-17:
They must have gotten crystals that are too exact or unaffected by temperature. I always postulated that this could occur but this is the first time I actually hear of it happening. I just assumed that temperature differences would cause an overlap to drift out in less then 30 minutes but I never did the math on it.

I see no other way to get around this but getting sensors that do not share the same cycle timing.