merbanan/rtl_433 #1937: Problems with mqtt qos

SumDum, 2022-01-10:
Hardware: rPi 4 1Gb
Version: 21.12
Devices: Ultrasonic Oil Standard (-R 81)
             Fineoffset 5in1 weather station (-R 32)

Hi,
Thank you all contributors for this great software.

**Problem 1:**
When I run with qos=1 I get acknowledge messages to the console, I'm logging to csv and mqtt.  Each mqtt send produces:
`MQTT Message publishing acknowledged (msg_id: xx)
`
This happens running it in an interactive session from a console (rtl_433 {stuff} &) or the messages appear on the boot console when run from rc.local.  Is there any way to stop this?  Minor issue really but since I'm here...

**Problem 2:**
When I run with qos=2 I get nothing at my mqtt broker.  csv logging continues fine so it's only the mqtt push affected.
Using Wireshark I see:
qos=0 (as expected):
```
> Publish message (id)
```
qos=1 (as expected):
```
> Publish message (id)
< Publish ack (id)
```
qos=2 (incomplete transaction):
```
> Publish message (id)
< Publish received (id)
```

There should be further messages, the complete sequence (assuming no packet loss) should be:
```
> Publish message (id)
< Publish received (id)
> Publish release (id)
< Publish complete (id)
```
It looks like rtl_433 is not sending the release packet so mosquitto doesn't accept & publish the message.

Thanks
SumDum

zuckschwerdt, 2022-01-10:
Thanks for testing. rtl_433 currently is intended for qos=0. We would need a resend queue and timer/timeout logic.
Maybe something to look into.

SumDum, 2022-01-10:
So what was the point of making a pull request that introduces the command line option?
https://github.com/merbanan/rtl_433/commit/e3ec350287a79d1c9048849f9c2635186456e755#diff-15c3e6e2f6f7cb2019b5351c19b77cef45ad53a64b9452e1144ae13d3fd95a94
Your entries into the five help options really ought to mention that it does absolutely nothing.  qos=1 only works by pure fluke and when there's no lost packets.

closed 2022-01-10 by SumDum

zuckschwerdt, 2022-01-10:
Well, `qos=1` gives you the ack notice to help check things are working. We never looked into full qos or even `qos=2`. Loss isn't really expected with local connections and shouldn't be a problem for telemetry data.
Maybe somebody wants to take a look at using the Mongoose timers and a `list_t` backing.

SumDum, 2022-01-10:
Yes but qos=1 is incomplete, it only 'works' if there is no packet loss and qos=1 or 2 are all about handling packet loss.  It should not be advertised in the various help as a feature when it is completely unimplemented.